Xilinx/dma_ip_drivers #6: runtime error on ubuntu 18.04.2 LTS: ./qdma_run_stress_test.sh: line 302: echo: write error: Invalid argument

closertotheheart, 2019-04-04:
root@super-poe:~/dma_ip_drivers/QDMA/linux-kernel/scripts# ./stress_test_top.sh 02000
#######################
####  user         ####
#######################
make -C user clean;
make[1]: Entering directory '/home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/user'
make[1]: Leaving directory '/home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/user'
#######################
####  tools        ####
#######################
make -C tools clean;
make[1]: Entering directory '/home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/tools'
rm -rf *.o *.bin dma_to_device dma_from_device dma_from_device_w_udd dmautils
make[1]: Leaving directory '/home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/tools'
#######################
####  drv          ####
#######################
make[1]: Entering directory '/home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/drv'
srcdir = /home/mcoyle/dma_ip_drivers/QDMA/linux-kernel.
KSRC = .
VF = .
EXTRA_FLAGS = .
ccflags-y = -D__GENL_REG_FAMILY_OPS_FUNC__ -D__READ_ONCE_DEFINED__.
make[1]: Leaving directory '/home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/drv'
"distro=, dmajor= dminor= "
4.15.184.15.0-47-generic: GIT,,, -DGITSP -DGIT
#######################
####  user        ####
#######################
make -C user
make[1]: Entering directory '/home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/user'
cc -g -I. -I../include    -c -o cli/reg_cmd.o cli/reg_cmd.c
cc -g -I. -I../include    -c -o cli/cmd_parse.o cli/cmd_parse.c
cc -g -I. -I../include    -c -o cli/nl_user.o cli/nl_user.c
cc -g -I. -I../include    -c -o cli/main.o cli/main.c
cc cli/reg_cmd.o cli/cmd_parse.o cli/nl_user.o cli/main.o -o dmactl
make[1]: Leaving directory '/home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/user'
#######################
####  tools        ####
#######################
make -C tools
make[1]: Entering directory '/home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/tools'
cc -g -c -std=c99 -o dma_to_device.o dma_to_device.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -lrt -o dma_to_device dma_to_device.o -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -g -c -std=c99 -o dma_from_device.o dma_from_device.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -lrt -o dma_from_device dma_from_device.o -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -g -c -std=c99 -o dma_from_device_w_udd.o dma_from_device_w_udd.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -lrt -o dma_from_device_w_udd dma_from_device_w_udd.o -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -g -c -std=c99 -o dmautils.o dmautils.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -pthread -lrt -o dmautils dmautils.o -laio -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
make[1]: Leaving directory '/home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/tools'
#######################
#### PF drv          ####
#######################
make[1]: Entering directory '/home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/drv'
srcdir = /home/mcoyle/dma_ip_drivers/QDMA/linux-kernel.
KSRC = /lib/modules/4.15.0-47-generic/build.
VF = 0.
EXTRA_FLAGS = .
ccflags-y = -D__READ_ONCE_DEFINED__.
make[2]: Entering directory '/usr/src/linux-headers-4.15.0-47-generic'
srcdir = /home/mcoyle/dma_ip_drivers/QDMA/linux-kernel.
KSRC = /lib/modules/4.15.0-47-generic/build.
VF = 0.
EXTRA_FLAGS = .
ccflags-y =  -D__READ_ONCE_DEFINED__.
  CC [M]  /home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/drv/nl.o
  CC [M]  /home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/drv/cdev.o
  CC [M]  /home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/drv/qdma_mod.o
  CC [M]  /home/mcoyle/dma_ip_drivers/QDMA/linux-kernel/drv/libqdma/qdma_mbox.o
[... 313 lines cut ...]
cmp: /tmp/out_mm02003_53: No such file or directory
cmp: /tmp/out_mm02003_55: No such file or directory
cmp: /tmp/out_mm02003_57: No such file or directory
cmp: /tmp/out_mm02003_59: No such file or directory
cmp: /tmp/out_mm02003_61: No such file or directory
cmp: /tmp/out_mm02003_63: No such file or directory
cmp: /tmp/out_mm02003_65: No such file or directory
cmp: /tmp/out_mm02003_67: No such file or directory
cmp: /tmp/out_mm02003_69: No such file or directory
cmp: /tmp/out_mm02003_71: No such file or directory
cmp: /tmp/out_mm02003_73: No such file or directory
cmp: /tmp/out_mm02003_75: No such file or directory
cmp: /tmp/out_mm02003_77: No such file or directory
cmp: /tmp/out_mm02003_79: No such file or directory
cmp: /tmp/out_mm02003_81: No such file or directory
cmp: /tmp/out_mm02003_83: No such file or directory
cmp: /tmp/out_mm02003_85: No such file or directory
cmp: /tmp/out_mm02003_87: No such file or directory
cmp: /tmp/out_mm02003_89: No such file or directory
cmp: /tmp/out_mm02003_91: No such file or directory
cmp: /tmp/out_mm02003_93: No such file or directory
cmp: /tmp/out_mm02003_95: No such file or directory
cmp: /tmp/out_mm02003_97: No such file or directory
cmp: /tmp/out_mm02003_99: No such file or directory
cmp: /tmp/out_mm02003_101: No such file or directory

hmaarrfk, 2019-05-10:
what is your output of `dmesg`. For me, `dmesg` shows that the kernel module isn't loaded with Error -16 EBUSY http://www-numi.fnal.gov/offline_software/srt_public_context/WebDocs/Errors/unix_system_errors.html

I wonder if it is tested on Ubuntu 18.04. They claim that the guest OS is Ubuntu 18.04, so I assume it works with it....

sujathabanoth-xlnx, 2021-01-29:
stress_test_top.sh is no more available as a release artifact in latest driver package. It was tested in older driver package on Ubuntu 18.04 and it was working fine. Check the test environment.
Currently closing the issue

closed 2021-01-29 by sujathabanoth-xlnx