Xilinx/dma_ip_drivers #10: non-root access?

hmaarrfk, 2019-05-14:
Are you accepting PRs that would enable non-root access to the device driver?

Seems like we should be able to setup a rule with `udev` to assign the devices to belong to a certain group.

kewlcreatures, 2020-06-11:
is this for QDMA?

hmaarrfk, 2020-06-11:
Yes

kewlcreatures, 2020-06-11:
So my application starts as a non root user and I faced the same issue of not being able to access the /Dev/QDMa queues. I changed the access to chmod 777 for those /Dev/QDMa and it works for me.

hmaarrfk, 2020-06-13:
doing chmod is fine if you, the developer are using your own machine. but it is quite annoying to tell others to do it.

kewlcreatures, 2020-06-13:
Yes I agree. Its a very good suggestion to allow non-root access

dfreese, 2020-07-23:
It would be great to add it to the repository, since it appears that there used to be something like that (see https://github.com/Xilinx/dma_ip_drivers/commit/8d603ef2654827c3d640d7b3b6c7bc3bc5179962).

What I ended up going with was creating a "/etc/udev/rules.d/60-xdma.rules" which was just:
```
SUBSYSTEM=="xdma" GROUP="xdma"
```

sujathabanoth-xlnx, 2021-01-29:
Was the above suggesions worked?

dfreese, 2021-01-29:
> Was the above suggesions worked?

We've used the solution I listed on my team without issue for the past several months.

hmaarrfk, 2021-01-30:
@dfreese can you comfirm if you are using this line of code for `qdma` or `xdma`?

dfreese, 2021-01-30:
> @dfreese can you comfirm if you are using this line of code for `qdma` or `xdma`?

xdma, so qdma may be different.