Xilinx/dma_ip_drivers #14: dmactl returns 0 even upon failure

hmaarrfk, 2019-05-14:
```
(base) mark@mark-MS-7758:~/git/dma_ip_drivers/QDMA/linux-kernel/user$ ./dmactl qdma01000 q start idx 0 dir h2c idx_ringsz 5

qdma01000-MM-0 invalid state, q_state 2.

(base) mark@mark-MS-7758:~/git/dma_ip_drivers/QDMA/linux-kernel/user$ echo $?
0
```
invalid state seems to imply failure, but the return code of `dmactl` is still 0

I thought it was due to the fact that `recv_nl_msg` ignored the return value of `recv_attrs` but that doesn't seem to be the case.

<details>

```patch
From 6d54ab0f91c49f1c8a6d01c99ec37ad8347e7afd Mon Sep 17 00:00:00 2001
From: Mark Harfouche <mark.harfouche@gmail.com>
Date: Tue, 14 May 2019 17:55:36 -0400
Subject: [PATCH] return recv_attrs if it failes.

---
 QDMA/linux-kernel/user/cli/nl_user.c | 4 +++-
 1 file changed, 3 insertions(+), 1 deletion(-)

diff --git a/QDMA/linux-kernel/user/cli/nl_user.c b/QDMA/linux-kernel/user/cli/nl_user.c
index 24746df..8335119 100644
--- a/QDMA/linux-kernel/user/cli/nl_user.c
+++ b/QDMA/linux-kernel/user/cli/nl_user.c
@@ -327,7 +327,9 @@ static int recv_nl_msg(struct xnl_hdr *hdr, struct xcmd_info *xcmd)
 	unsigned int op = hdr->g.cmd;
 	unsigned int usr_bar;
 
-	recv_attrs(hdr, xcmd);
+	int rv = recv_attrs(hdr, xcmd);
+	if (rv)
+		return rv;
 
 	switch(op) {
 	case XNL_CMD_DEV_LIST:
```

</details>

Thoughts?

sujathabanoth-xlnx, 2021-01-29:
Issue is addressed in 2019.1 Patch release, Hence closing it

closed 2021-01-29 by sujathabanoth-xlnx

hmaarrfk, 2021-01-30:
Thanks for letting me know! 

This is pretty nice.

It would be great if these user facing changes were documented in the Release notes so that we can make use of them more quickly.