Xilinx/dma_ip_drivers #17: queue might not be rewinded correctly in qdma_queue_add in case buffer too short

hmaarrfk, 2019-05-15:
In the snippet below, an error message is passed to the user.
However, if they provide a buffer that is too small, the queue doesn't get rewound correctly.

https://github.com/Xilinx/dma_ip_drivers/blob/8b8c70b697f049649d5fa99be9c6bc4302d89ac9/QDMA/linux-kernel/libqdma/libqdma_export.c#L994-L1002
https://github.com/Xilinx/dma_ip_drivers/blob/8b8c70b697f049649d5fa99be9c6bc4302d89ac9/QDMA/linux-kernel/libqdma/libqdma_export.c#L1110-L1123

sujathabanoth-xlnx, 2021-01-29:
Issue addressed in 2019.2 driver release, Hence closing it

closed 2021-01-29 by sujathabanoth-xlnx

hmaarrfk, 2021-01-30:
hmm, ok. i'm not too sure it is done correctly even in the new version.

It seems that if the queue is added there may be some cases where the following is not called

```
			descq->q_state = Q_STATE_DISABLED;

```
before an error is raised...