Xilinx/dma_ip_drivers #22: potential index out of bounds error in qdma_device_get_descq_by_id

hmaarrfk, 2019-05-15:
https://github.com/Xilinx/dma_ip_drivers/blob/f3828c8604126fa1ab149d5969d4b4c85509991f/QDMA/linux-kernel/libqdma/qdma_device.c#L462-L468

len is actually the length of the string if it would have been written with the full text. Therefore it `buf[len]` can point outside `buf`. `snprintf` is guaranteed to null terminate, so we don't need to null-terminate it ourselves. Finally because of the return statement immediately after the snprintf statement, there is no need to update `buflen`

hmaarrfk, 2019-05-15:
There is an other bug like this a few lines down too.

hmaarrfk, 2019-05-15:
ditto for `qdma_device_get_pair_descq_by_id`

sujathabanoth-xlnx, 2021-01-29:
Issue fixed in latest driver release, hence closing it

closed 2021-01-29 by sujathabanoth-xlnx