Xilinx/dma_ip_drivers #28: XDMA performance causes CPU to reboot

hinxx, 2019-06-19:
Everytime I would run `performance` tool the CPU in uTCA crate reboots.
XDMA bitfile is loaded onto AMC with Ultrascale FPGA.
 
Running `./performance -d /dev/xdma0_h2c_0 -c 1 -s 64` in one terminal, and checking `tail -f /var/log/messages` in another.

```
hinxx@obzen ~ $ ssh dev@bd-cpu06

[dev@bd-cpu06 ~]$ 
[dev@bd-cpu06 ~]$ 
[dev@bd-cpu06 ~]$ tail -f /var/log/messages

[dev@bd-cpu06 ~]$ sudo tail -f /var/log/messages
[sudo] password for dev: 
Jun 19 16:30:30 bd-cpu06 kdumpctl: kexec: loaded kdump kernel
Jun 19 16:30:30 bd-cpu06 kdumpctl: Starting kdump: [OK]
Jun 19 16:30:30 bd-cpu06 systemd: Started Crash recovery kernel arming.
Jun 19 16:30:30 bd-cpu06 systemd: Startup finished in 603ms (kernel) + 2.316s (initrd) + 11.022s (userspace) = 13.942s.
Jun 19 16:30:31 bd-cpu06 kernel: random: crng init done
Jun 19 16:30:59 bd-cpu06 systemd: Starting Stop Read-Ahead Data Collection...
Jun 19 16:30:59 bd-cpu06 systemd: Started Stop Read-Ahead Data Collection.
Jun 19 16:31:31 bd-cpu06 systemd: Created slice User Slice of dev.
Jun 19 16:31:31 bd-cpu06 systemd-logind: New session 1 of user dev.
Jun 19 16:31:31 bd-cpu06 systemd: Started Session 1 of user dev.

Jun 19 16:31:54 bd-cpu06 systemd-logind: New session 2 of user dev.
Jun 19 16:31:54 bd-cpu06 systemd: Started Session 2 of user dev.
Jun 19 16:32:13 bd-cpu06 kernel: ------------[ cut here ]------------
Jun 19 16:32:13 bd-cpu06 kernel: WARNING: CPU: 2 PID: 20 at lib/list_debug.c:62 __list_del_entry+0x82/0xd0
Jun 19 16:32:13 bd-cpu06 kernel: list_del corruption. next->prev should be ffff8db0017d1300, but was           (null)
Jun 19 16:32:13 bd-cpu06 kernel: Modules linked in: iTCO_wdt iTCO_vendor_support intel_powerclamp coretemp intel_rapl kvm_intel kvm irqbypass crc32_pclmul ghash_clmulni_intel aesni_intel lrw gf128mul glue_helper ablk_helper cryptd sg pcspkr cdc_acm mrf(OE) uio parport xdma(OE) snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep snd_seq snd_seq_device snd_pcm snd_timer snd soundcore i2c_i801 mei_me mei lpc_ich ip_tables xfs libcrc32c sd_mod crc_t10dif crct10dif_generic ata_generic pata_acpi i915 crct10dif_pclmul crct10dif_common crc32c_intel ata_piix libata serio_raw igb dca i2c_algo_bit iosf_mbi drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm e1000e video ptp pps_core drm_panel_orientation_quirks dm_mirror dm_region_hash dm_log dm_mod
Jun 19 16:32:13 bd-cpu06 kernel: CPU: 2 PID: 20 Comm: kworker/2:0 Kdump: loaded Tainted: G           OE  ------------   3.10.0-957.12.1.el7.x86_64 #1
Jun 19 16:32:13 bd-cpu06 kernel: Hardware name: Concurrent Technologies   AM900/x/AM 90X/X1X, BIOS 3.09.01 21/02/2017
Jun 19 16:32:13 bd-cpu06 kernel: Workqueue: events engine_service_work [xdma]
Jun 19 16:32:13 bd-cpu06 kernel: Call Trace:
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa7163021>] dump_stack+0x19/0x1b
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6a97688>] __warn+0xd8/0x100
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6a9770f>] warn_slowpath_fmt+0x5f/0x80
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6d950e2>] __list_del_entry+0x82/0xd0
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6d9513d>] list_del+0xd/0x30
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffc063ea73>] engine_service_final_transfer+0x93/0x240 [xdma]
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffc063ed12>] engine_service+0xf2/0x250 [xdma]
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffc063f87a>] engine_service_work+0xba/0xc0 [xdma]
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6ab9d8f>] process_one_work+0x17f/0x440
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6abae26>] worker_thread+0x126/0x3c0
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6abad00>] ? manage_workers.isra.25+0x2a0/0x2a0
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6ac1c71>] kthread+0xd1/0xe0
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6ac1ba0>] ? insert_kthread_work+0x40/0x40
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa7175bf7>] ret_from_fork_nospec_begin+0x21/0x21
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6ac1ba0>] ? insert_kthread_work+0x40/0x40
Jun 19 16:32:13 bd-cpu06 kernel: ---[ end trace 782aec7c1eaf1173 ]---
Jun 19 16:32:13 bd-cpu06 kernel: xdma:engine_service_resume: re-started 0-H2C0-MM engine with pending xfer 0xffff8db0017d1300
Jun 19 16:32:13 bd-cpu06 kernel: ------------[ cut here ]------------
Jun 19 16:32:13 bd-cpu06 kernel: WARNING: CPU: 3 PID: 62 at lib/list_debug.c:53 __list_del_entry+0x63/0xd0
Jun 19 16:32:13 bd-cpu06 kernel: list_del corruption, ffff8db0017d1300->next is LIST_POISON1 (dead000000000100)
Jun 19 16:32:13 bd-cpu06 kernel: Modules linked in: iTCO_wdt iTCO_vendor_support intel_powerclamp coretemp intel_rapl kvm_intel kvm irqbypass crc32_pclmul ghash_clmulni_intel aesni_intel lrw gf128mul glue_helper ablk_helper cryptd sg pcspkr cdc_acm mrf(OE) uio parport xdma(OE) snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep snd_seq snd_seq_device snd_pcm snd_timer snd soundcore i2c_i801 mei_me mei lpc_ich ip_tables xfs libcrc32c sd_mod crc_t10dif crct10dif_generic ata_generic pata_acpi i915 crct10dif_pclmul crct10dif_common crc32c_intel ata_piix libata serio_raw igb dca i2c_algo_bit iosf_mbi drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm e1000e video ptp pps_core drm_panel_orientation_quirks dm_mirror dm_region_hash dm_log dm_mod
Jun 19 16:32:13 bd-cpu06 kernel: CPU: 3 PID: 62 Comm: kworker/3:1 Kdump: loaded Tainted: G        W  OE  ------------   3.10.0-957.12.1.el7.x86_64 #1
Jun 19 16:32:13 bd-cpu06 kernel: Hardware name: Concurrent Technologies   AM900/x/AM 90X/X1X, BIOS 3.09.01 21/02/2017
Jun 19 16:32:13 bd-cpu06 kernel: Workqueue: events engine_service_work [xdma]
Jun 19 16:32:13 bd-cpu06 kernel: Call Trace:
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa7163021>] dump_stack+0x19/0x1b
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6a97688>] __warn+0xd8/0x100
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6a9770f>] warn_slowpath_fmt+0x5f/0x80
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6d950c3>] __list_del_entry+0x63/0xd0
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6d9513d>] list_del+0xd/0x30
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffc063ea73>] engine_service_final_transfer+0x93/0x240 [xdma]
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffc063ed12>] engine_service+0xf2/0x250 [xdma]
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffc063f87a>] engine_service_work+0xba/0xc0 [xdma]
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6ab9d8f>] process_one_work+0x17f/0x440
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6abae26>] worker_thread+0x126/0x3c0
[... 97 lines cut ...]
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffc063ed12>] engine_service+0xf2/0x250 [xdma]
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffc063f87a>] engine_service_work+0xba/0xc0 [xdma]
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6ab9d8f>] process_one_work+0x17f/0x440
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6abae26>] worker_thread+0x126/0x3c0
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6abad00>] ? manage_workers.isra.25+0x2a0/0x2a0
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6ac1c71>] kthread+0xd1/0xe0
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6ac1ba0>] ? insert_kthread_work+0x40/0x40
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa7175bf7>] ret_from_fork_nospec_begin+0x21/0x21
Jun 19 16:32:13 bd-cpu06 kernel: [<ffffffffa6ac1ba0>] ? insert_kthread_work+0x40/0x40
Jun 19 16:32:13 bd-cpu06 kernel: ---[ end trace 782aec7c1eaf1178 ]---
Jun 19 16:32:13 bd-cpu06 kernel: xdma:engine_service_resume: re-started 0-H2C0-MM engine with pending xfer 0xffff8db0017d1300
Jun 19 16:32:13 bd-cpu06 kernel: ------------[ cut here ]------------

...
```

The log gets quite long..

Reproducible every time the `performance` is called.

uname -a:

```
Linux bd-cpu06 3.10.0-957.12.1.el7.x86_64 #1 SMP Mon Apr 29 14:59:59 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

hinxx, 2019-06-20:
@hmaarrfk did you maybe see / fix this problem in your https://github.com/ramonaoptics/xilinx-dma-driver repo?
Because if I use your xdma kernel driver I do not get this problem..

karenx-xilinx, 2020-11-23:
Could you retest with TOT (current version v2020.1.8)? We are not able to reproduce this. If you still experience the issue, can you update the log? Thanks.`