Xilinx/dma_ip_drivers #34: Driver won't compile in Ubuntu 19.04

braspatta, 2019-09-20:
Driver won't compile under Ubuntu 19.04 (kernel 5.0.0-29).

<pre>root@sim-pc:/home/simulation/Documents/lfpd/dma_ip_drivers-master/XDMA/linux-kernel/xdma# sudo make install
Makefile:10: XVC_FLAGS: .
make -C /lib/modules/5.0.0-29-generic/build M=/home/simulation/Documents/lfpd/dma_ip_drivers-master/XDMA/linux-kernel/xdma modules
make[1]: Entering directory &apos;/usr/src/linux-headers-5.0.0-29-generic&apos;
/home/simulation/Documents/lfpd/dma_ip_drivers-master/XDMA/linux-kernel/xdma/Makefile:10: XVC_FLAGS: .
  CC [M]  /home/simulation/Documents/lfpd/dma_ip_drivers-master/XDMA/linux-kernel/xdma/libxdma.o
<b>/home/simulation/Documents/lfpd/dma_ip_drivers-master/XDMA/linux-kernel/xdma/libxdma.c:</b> In function ‘<b>engine_service_shutdown</b>’:
<b>/home/simulation/Documents/lfpd/dma_ip_drivers-master/XDMA/linux-kernel/xdma/libxdma.c:671:2:</b> <font color="#CC0000"><b>error: </b></font>implicit declaration of function ‘<b>swake_up</b>’; did you mean ‘<b>wake_up</b>’? [<font color="#CC0000"><b>-Werror=implicit-function-declaration</b></font>]
  <font color="#CC0000"><b>swake_up</b></font>(&amp;engine-&gt;shutdown_wq);
  <font color="#CC0000"><b>^~~~~~~~</b></font>
  <font color="#4E9A06">wake_up</font>
<b>/home/simulation/Documents/lfpd/dma_ip_drivers-master/XDMA/linux-kernel/xdma/libxdma.c:</b> In function ‘<b>xdma_xfer_submit</b>’:
<b>/home/simulation/Documents/lfpd/dma_ip_drivers-master/XDMA/linux-kernel/xdma/libxdma.c:3127:4:</b> <font color="#CC0000"><b>error: </b></font>implicit declaration of function ‘<b>swait_event_interruptible_timeout</b>’; did you mean ‘<b>wait_event_interruptible_timeout</b>’? [<font color="#CC0000"><b>-Werror=implicit-function-declaration</b></font>]
    <font color="#CC0000"><b>swait_event_interruptible_timeout</b></font>(xfer-&gt;wq,
    <font color="#CC0000"><b>^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
    <font color="#4E9A06">wait_event_interruptible_timeout</font>
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:286: /home/simulation/Documents/lfpd/dma_ip_drivers-master/XDMA/linux-kernel/xdma/libxdma.o] Error 1
make[1]: *** [Makefile:1606: _module_/home/simulation/Documents/lfpd/dma_ip_drivers-master/XDMA/linux-kernel/xdma] Error 2
make[1]: Leaving directory &apos;/usr/src/linux-headers-5.0.0-29-generic&apos;
make: *** [Makefile:27: all] Error 2
</pre>

used the latest version of the repository.

iDoka, 2019-10-01:
**TL;DR:**  type in console, it helps:
```
cd XDMA/linux-kernel/xdma
sed -i 's/swake_up/swake_up_one/' libxdma.c
sed -i 's/swait_event_interruptible_timeout/swait_event_interruptible_timeout_exclusive/' libxdma.c
sed -i 's/VERIFY_WRITE,//' cdev_ctrl.c
sed -i 's/VERIFY_READ,//' cdev_ctrl.c

```

The issue is 5.0 version of linux kernel.

I have the same in Ubuntu 18.04 LTS (kernel 5.0.0-23). Then I found solution here: 
https://forums.xilinx.com/t5/PCI-Express/PCIe-DMA-driver-compilation-issues-in-Linux-Ubuntu-19-04/td-p/1022239
and rewrite it with help of sed.

X-Ryl669, 2020-01-17:
See #44 for a clean fix that should work with any version.