Xilinx/dma_ip_drivers #40: xdma_thread.h missing

sgherbst, 2019-12-18:
Running "make install" in the linux-kernel/xdma folder doesn't work with the latest commit due to a missing file "xdma_thread.h".  Sample error:
```code
/path/to/XDMA/linux-kernel/xdma/libxdma.c:33:10: fatal error: xdma_thread.h: No such file or directory
 #include "xdma_thread.h"
```

shahbaazlokh, 2019-12-19:
same error for me

hmaarrfk, 2019-12-19:
you should post your question on xilinx's forum. and link them here. They don't monitor this. Or maybe they are told not to reply here.

sgherbst, 2019-12-19:
Thanks for letting me know -- just posted the question here: [https://forums.xilinx.com/t5/PCIe-and-CPM/xdma-thread-h-missing-in-dma-ip-drivers/td-p/1057087](https://forums.xilinx.com/t5/PCIe-and-CPM/xdma-thread-h-missing-in-dma-ip-drivers/td-p/1057087)

hmaarrfk, 2019-12-19:
Their history speaks for itself. It's like talking to a wall here.

karenx-xilinx, 2019-12-20:
The missing files are added in commit c7712d79dd251bd98c3eb70aaaf2e39f2a8c55b6. Thanks for your patience.

closed 2019-12-20 by karenx-xilinx