Xilinx/dma_ip_drivers #41: lowlatency kernel cannot use XDMA driver properly

SYJ7895, 2019-12-30:
When I load the XDMA driver with a low-latency kernel on Ubuntu1604, a segmentation fault occurs when running run_test.sh. At this time, I can see the following information through dmesg:
BUG: scheduling while atomic: dma_from_device / 3266 / 0x00000000

X-Ryl669, 2020-01-17:
This is likely an issue with the use of simple wait queues. Please try to use later kernel (at least 4.18 or better 5.0+) and report if it does not work.

SYJ7895, 2020-02-20:
The "xdma_xfer_submit" function in libxdma.c calls "spin_lock (& engine-> desc_lock)" and then calls the "wait_event_interruptible_timeout" function with sleep nature. This caused the problem of "BUG: scheduling while atomic: dma_from_device / 3266 / 0x00000000" in the low-latency kernel, but there was no problem in the general kernel.

SYJ7895, 2020-02-20:
> This is likely an issue with the use of simple wait queues. Please try to use later kernel (at least 4.18 or better 5.0+) and report if it does not work.

The "xdma_xfer_submit" function in libxdma.c calls "spin_lock (& engine-> desc_lock)" and then calls the "wait_event_interruptible_timeout" function with sleep nature. This caused the problem of "BUG: scheduling while atomic: dma_from_device / 3266 / 0x00000000" in the low-latency kernel, but there was no problem in the general kernel.

hmaarrfk, 2020-02-20:
ive seen this in QDMA as well.

chengts95, 2020-02-20:
Any suggestion to fix that?

hmaarrfk, 2020-02-20:
Honestly, it could be related to: https://github.com/Xilinx/dma_ip_drivers/pull/49

I used aligned memory, and the problems lesssned (i think)

chengts95, 2020-02-21:
On 5.3.15 x86_generic it will not cause this problem. It happens on my AARCH64 (kernel 4.9.140) board with RT patch. However, even on x86_generic the `run_test.sh` cannot work properly after I merged the 2 PRs.

hmaarrfk, 2020-02-21:
I'm not sure. You should ping xilixn on their forums.

There was a XDMA driver written by others that fixed many aarch64 issues. It may fix yours.

It was on gitlab if I remember

chengts95, 2020-02-21:
I suspect the usage of "swait.h"  in this driver causes a lot of problems since the simple wait queue is "BROKEN wait queue" in the kernel now.

bhathaway, 2020-02-24:
I have personally tested the fix on AARCH64. I would appreciate feedback on the following pull request, which I believe fixes this issue:

https://github.com/Xilinx/dma_ip_drivers/pull/50

SYJ7895, 2020-02-26:
> I have personally tested the fix on AARCH64. I would appreciate feedback on the following pull request, which I believe fixes this issue:
> 
> #50

As you said, I am currently modifying this way, and the test is currently normal.

hmaarrfk, 2020-02-26:
Maybe keep this open until xilinx acknowledge s this?

chengts95, 2020-02-26:
Still very buggy on my device, I think it is related to kernel version.