Xilinx/dma_ip_drivers #42: Current GIT XDMA tree fails to build on ubuntu 18.0.4.3

johnutz-PNSR, 2020-01-08:
The kernel rev is: 5.0.0-37-generic
following the instructions in /dma_ip_drivers/XDMA/linux-kernel/readme.txt:
Usage:
  - Change directory to the driver directory.
        cd xdma
  - Compile and install the kernel module driver.
        make install
GIT/dma_ip_drivers/XDMA/linux-kernel$ cd xdma
ryzendev@ryzen2:~/GIT/dma_ip_drivers/XDMA/linux-kernel/xdma$ make install
Makefile:10: XVC_FLAGS: .
make -C /lib/modules/5.0.0-37-generic/build M=/home/ryzendev/GIT/dma_ip_drivers/XDMA/linux-kernel/xdma modules
make[1]: Entering directory '/usr/src/linux-headers-5.0.0-37-generic'
/home/ryzendev/GIT/dma_ip_drivers/XDMA/linux-kernel/xdma/Makefile:10: XVC_FLAGS: .
  CC [M]  /home/ryzendev/GIT/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.o
/home/ryzendev/GIT/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c: In function ‘engine_service_shutdown’:
/home/ryzendev/GIT/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c:742:2: error: implicit declaration of function ‘swake_up’; did you mean ‘wake_up’? [-Werror=implicit-function-declaration]
  swake_up(&engine->shutdown_wq);
  ^~~~~~~~
  wake_up
/home/ryzendev/GIT/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c: In function ‘xdma_xfer_submit’:
/home/ryzendev/GIT/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c:3580:4: error: implicit declaration of function ‘swait_event_interruptible_timeout’; did you mean ‘wait_event_interruptible_timeout’? [-Werror=implicit-function-declaration]
    swait_event_interruptible_timeout(
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    wait_event_interruptible_timeout
cc1: some warnings being treated as errors
scripts/Makefile.build:284: recipe for target '/home/ryzendev/GIT/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.o' failed
make[2]: *** [/home/ryzendev/GIT/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.o] Error 1
Makefile:1614: recipe for target '_module_/home/ryzendev/GIT/dma_ip_drivers/XDMA/linux-kernel/xdma' failed
make[1]: *** [_module_/home/ryzendev/GIT/dma_ip_drivers/XDMA/linux-kernel/xdma] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.0.0-37-generic'
Makefile:27: recipe for target 'all' failed
make: *** [all] Error 2

johnutz-PNSR, 2020-01-08:
so this bug is a second repro of #30 and #34 and a fix is in #34. i would hope that xilinx can check in a fix soon.

X-Ryl669, 2020-01-17:
See #44 for a proposed solution.

johnutz-PNSR, 2020-01-17:
@X-Ryl669 ; good to know! thankyou very much for your effort

lesjokolat, 2020-02-16:
hi check this link:

https://github.com/Xilinx/dma_ip_drivers/pull/44#issuecomment-584447315

I think may help your issues with make.

seems a bit of a bug in actual make file