Xilinx/dma_ip_drivers #45: QDMA driver crashes Linux if MSI-X not available

unhold, 2020-01-20:
The QDMA driver creates a warning and crashes Linux if MSI-X is not available or the system or in the QDMA IP.

From Linux version 3.14, [`pci_msix_vec_count()`](https://github.com/torvalds/linux/blob/def9d2780727cec3313ed3522d0123158d87224d/drivers/pci/msi.c#L958) returns `-EINVAL` when there is no MSI-X support or (MSI-X is disabled in the QDMA IP), but the driver checks for `0`. The following `kzalloc()` tries to allocate `(size_t)-EINVAL` structs, i.e. all available memory, triggering a WARN_ON and crashing the kernel.

https://github.com/Xilinx/dma_ip_drivers/blob/a0df28038eaeffffbc4f38b6e0269dad13478b4b/QDMA/linux-kernel/libqdma/qdma_intr.c#L531-L538

hmaarrfk, 2020-01-20:
Just a quick question, are you using their latest branch on master, or the updates from 2019.1?

Is this with Vivado 2019.2 logic + latest commit from master?

unhold, 2020-01-20:
I'm using the QDMA driver from the current master branch, Vivado 2019.2 and Ubuntu Linux kernel 4.15.0-23. Other versions will also crash, as long as the kernel 3.14 or newer.

hmaarrfk, 2020-01-20:
Hmm ok, then maybe my issue is different. I would have to double check my design to see if I enabled MSI-X.

unhold, 2020-02-27:
Thank you @karenx-xilinx for merging the fix!

@hmaarrfk: I think that having only 1 MSI-X configured in the QDMA IP also triggers a bug in the driver, but I had no time analyzing it throughly.

closed 2020-02-27 by unhold