Xilinx/dma_ip_drivers #48: XDMA: Some times xdma engine crashes with irq MAGIC_STOPPED

IgorAmosov, 2020-01-29:
[dmesg_xdma.txt](https://github.com/Xilinx/dma_ip_drivers/files/4128495/dmesg_xdma.txt)
Hello!
I'am using xdma driver with XDMA IP in Memory Mapped case.
kernel:3.10.0-1062.9.1.el7.x86_64
CentOS 7.7
Some times big transaction crashes with MAGIC_STOPPED irq in dmesg (attached).
I have no idea, what is it.

bhathaway, 2020-02-03:
I have also encountered this exact issue. My only clue at this point is that somehow descriptors are being corrupted.

IgorAmosov, 2020-02-04:
I downgrade to "version: 2018.3.50" (design in vivado 2019.2) and all is ok

bhathaway, 2020-02-04:
@IgorAmosov Did you downgrade the driver source code or Vivado?

IgorAmosov, 2020-02-05:
> @IgorAmosov Did you downgrade the driver source code or Vivado?

I downgrade driver version from this repo.

bhathaway, 2020-02-18:
Please see https://github.com/Xilinx/dma_ip_drivers/pull/49

I would be curious if my bug fix resolves your issue.

IsaacJT, 2020-05-14:
Hi @bhathaway,

I encountered this exact problem and your fix resolved it for me. Thank you very much!