Xilinx/dma_ip_drivers #51: xdma_kthread_start error

imyier, 2020-03-05:
I am using vcu118 xdma IP，xdma driver can be install successfully。But，when insmod xdma.ko ，error appears as follows：
ERROR INFO:
xdma:xdma_threads_create: xdma_threads_create
BUG: unable to handle kernel NULL pointer dereference at 0000000000000009
IP: [<ffffffff811fab70>] get_partial_node.isra.62+0x10/0x1f0
 PGD 0
Oops: 0000 [#1] SMPxdma:xdma_kthread_start: kthread cmpl_status_th6, create task failed: 0x1

OS INFO:
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.6 LTS
Release:        16.04
Codename:       xenial
Linux ubuntu 4.4.0-170-generic #199-Ubuntu SMP Thu Nov 14 01:45:04 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

imyier, 2020-03-05:
i change the xdma IP design，this error disappear。

hmaarrfk, 2020-03-05:
Are you able to share your current configuration parameters?

hmaarrfk, 2020-03-05:
It might help others get to the root of the problem

imyier, 2020-03-06:
I am using the configuration patameters the same as the PG195 document describes now.Thanks for your reply.
https://www.xilinx.com/products/intellectual-property/pcie-dma.html#documentation

hmaarrfk, 2020-03-06:
I would really suggest you post on their forum if that is the case

manjuraehmad, 2020-03-26:
@imyier I have tried to test the master branch today on Host system with Ubuntu 16.04 OS and linux kernel version 4.15 and we have faced similar issue and now we are able to at least create the kthread by changing "_xdma_threads_create(8);_" to "_xdma_threads_create(4);_" in xdma/xdma_cdev.c file. Hope, this may work for you as well.

Do you have posted this query in Xilinx forum? If yes then can you please share the link to get better solution?