Xilinx/dma_ip_drivers #52: missing 'System.map' file

SamuelFarid, 2020-03-06:
Hello!

I am trying to install the driver following the procedure in the readme file but I get the following output when running **make install**:

Makefile:10: XVC_FLAGS: .
make -C /lib/modules/5.3.0-40-generic/build M=/home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-40-generic'
/home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/Makefile:10: XVC_FLAGS: .
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_events.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_sgdma.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_xvc.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_bypass.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_mod.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_thread.o
  LD [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma.o
  Building modules, stage 2.
/home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/Makefile:10: XVC_FLAGS: .
  MODPOST 1 modules
  CC      /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma.mod.o
  LD [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-40-generic'
make -C /lib/modules/5.3.0-40-generic/build M=/home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-40-generic'
  INSTALL /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma.ko
  DEPMOD  5.3.0-40-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-40-generic'

Is there any way to solve that issue? I have tried a lot of things but it doesn't work. I am running on Ubuntu 19.10

Thanks in advance!

lesjokolat, 2020-03-09:
yes there is its a tweak to makefile

https://github.com/Xilinx/dma_ip_drivers/pull/44#issuecomment-584447315

SamuelFarid, 2020-03-09:
Thanks for your reply, but I have done this and the output is still the same.

lesjokolat, 2020-03-09:
copy the output after you did it if you could. Will allow me to compare.

SamuelFarid, 2020-03-09:
root@SEAS16012:/home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma# make install
Makefile:10: XVC_FLAGS: .
make -C /lib/modules/5.3.0-40-generic/build M=/home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-40-generic'
/home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/Makefile:10: XVC_FLAGS: .
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_events.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_sgdma.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_xvc.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_bypass.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_mod.o
  CC [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_thread.o
  LD [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma.o
  Building modules, stage 2.
/home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/Makefile:10: XVC_FLAGS: .
  MODPOST 1 modules
  CC      /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma.mod.o
  LD [M]  /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-40-generic'
make -C /lib/modules/5.3.0-40-generic/build M=/home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-40-generic'
  INSTALL /home/samuel/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma.ko
  DEPMOD  5.3.0-40-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-40-generic'
depmod -a

lesjokolat, 2020-03-09:
can you add the makefile you adjusted?

SamuelFarid, 2020-03-09:
SHELL = /bin/bash
ifneq ($(xvc_bar_num),)
	XVC_FLAGS += -D__XVC_BAR_NUM__=$(xvc_bar_num)
endif

ifneq ($(xvc_bar_offset),)
	XVC_FLAGS += -D__XVC_BAR_OFFSET__=$(xvc_bar_offset)
endif

$(warning XVC_FLAGS: $(XVC_FLAGS).)

topdir := $(shell cd $(src)/.. && pwd)

TARGET_MODULE:=xdma

EXTRA_CFLAGS := -I$(topdir)/include $(XVC_FLAGS)
#EXTRA_CFLAGS += -D__LIBXDMA_DEBUG__
#EXTRA_CFLAGS += -DINTERNAL_TESTING

ifneq ($(KERNELRELEASE),)
	$(TARGET_MODULE)-objs := libxdma.o xdma_cdev.o cdev_ctrl.o cdev_events.o cdev_sgdma.o cdev_xvc.o cdev_bypass.o xdma_mod.o xdma_thread.o
	obj-m := $(TARGET_MODULE).o
else
	BUILDSYSTEM_DIR:=/lib/modules/$(shell uname -r)/build
	PWD:=$(shell pwd)
all :
	$(MAKE) -C $(BUILDSYSTEM_DIR) M=$(PWD) modules

install: all
	$(MAKE) -C $(BUILDSYSTEM_DIR) M=$(PWD) modules_install
	depmod -a
clean:
	$(MAKE) -C $(BUILDSYSTEM_DIR) M=$(PWD) clean

endif

lesjokolat, 2020-03-09:
looks ok, I see you are root the only diff I can see is a run my command "su make install" Maybe a weird issue with how root is defined. My kernel example was a bit different as well.

its also odd you didnt get any SSL errors. Typically i was getting them as well at this point. Cna you try install from a non root account then SU command when make?

SamuelFarid, 2020-03-09:
I still got the same output. I have solved the ssl errors using this [method](https://github.com/Xilinx/dma_ip_drivers/issues/39).
May I know what OS and kernel version you are running?

lesjokolat, 2020-03-09:
sure ubuntu 18.04.2
kernel
5.3.0-26

SamuelFarid, 2020-03-09:
Have you fixed the ssl error using the same method I have used? Is there anything else you have modified to solve the system.map missing file error?

lesjokolat, 2020-03-09:
you know what I just realized it may have installed ok its just a warning message. try the 2 xmod checks after install to see if its found and installed.

SamuelFarid, 2020-03-09:
I have already done this but it fails to identify DMA. In your case the make install didn't give system.map file issue?

lesjokolat, 2020-03-09:
correct I get message but this is a warning that allow install to continue.

When i run my xmod checks after I see driver is installed

lesjokolat, 2020-03-09:
ok then I have also seen where you then need to go into test folder and run load driver script. then check xmod

SamuelFarid, 2020-03-09:
I have tried loading driver but it fails. This is the script output:

Loading xdma driver...insmod: ERROR: could not insert module ../xdma/xdma.ko: Operation not permitted
Error: Kernel module did not load properly.
 FAILED

lesjokolat, 2020-03-09:
insmod seems like the polling method is tripping up. i did not change polling method when I installed mine.

To check if driver is ok i run one of these 2

- Check driver Version number
        modinfo xdma (or)
        modinfo ../xdma/xdma.ko    

Then run lspci

SamuelFarid, 2020-03-09:
modinfo xdma (or) fails to run and its output:
bash: syntax error near unexpected token `('
And modinfo ../xdma/xdma.ko output:
filename:       /home/samuel/dma_ip_drivers/XDMA/linux-kernel/tests/../xdma/xdma.ko
license:        Dual BSD/GPL
version:        2019.2.51
description:    Xilinx XDMA Reference Driver
author:         Xilinx, Inc.
srcversion:     FE994E934AE7021CE74856F
alias:          pci:v000010EEd00002808sv*sd*bc*sc*i*
alias:          pci:v000010EEd00004B28sv*sd*bc*sc*i*
alias:          pci:v000010EEd00004A28sv*sd*bc*sc*i*
alias:          pci:v000010EEd00004908sv*sd*bc*sc*i*
alias:          pci:v000010EEd00004828sv*sd*bc*sc*i*
alias:          pci:v000010EEd00004808sv*sd*bc*sc*i*
alias:          pci:v000010EEd00006D30sv*sd*bc*sc*i*
alias:          pci:v000010EEd00006A30sv*sd*bc*sc*i*
alias:          pci:v000010EEd00006A28sv*sd*bc*sc*i*
alias:          pci:v000010EEd00006930sv*sd*bc*sc*i*
alias:          pci:v000010EEd00006928sv*sd*bc*sc*i*
alias:          pci:v000010EEd00006830sv*sd*bc*sc*i*
alias:          pci:v000010EEd00006828sv*sd*bc*sc*i*
alias:          pci:v000010EEd00007038sv*sd*bc*sc*i*
alias:          pci:v000010EEd00007034sv*sd*bc*sc*i*
alias:          pci:v000010EEd00007032sv*sd*bc*sc*i*
alias:          pci:v000010EEd00007031sv*sd*bc*sc*i*
alias:          pci:v000010EEd00007028sv*sd*bc*sc*i*
alias:          pci:v000010EEd00007024sv*sd*bc*sc*i*
alias:          pci:v000010EEd00007022sv*sd*bc*sc*i*
alias:          pci:v000010EEd00007021sv*sd*bc*sc*i*
alias:          pci:v000010EEd00007018sv*sd*bc*sc*i*
alias:          pci:v000010EEd00007014sv*sd*bc*sc*i*
alias:          pci:v000010EEd00007012sv*sd*bc*sc*i*
alias:          pci:v000010EEd00007011sv*sd*bc*sc*i*
alias:          pci:v000010EEd00008038sv*sd*bc*sc*i*
alias:          pci:v000010EEd00008034sv*sd*bc*sc*i*
alias:          pci:v000010EEd00008032sv*sd*bc*sc*i*
alias:          pci:v000010EEd00008031sv*sd*bc*sc*i*
alias:          pci:v000010EEd00008028sv*sd*bc*sc*i*
alias:          pci:v000010EEd00008024sv*sd*bc*sc*i*
alias:          pci:v000010EEd00008022sv*sd*bc*sc*i*
alias:          pci:v000010EEd00008021sv*sd*bc*sc*i*
alias:          pci:v000010EEd00008018sv*sd*bc*sc*i*
alias:          pci:v000010EEd00008014sv*sd*bc*sc*i*
alias:          pci:v000010EEd00008012sv*sd*bc*sc*i*
alias:          pci:v000010EEd00008011sv*sd*bc*sc*i*
alias:          pci:v000010EEd00009011sv*sd*bc*sc*i*
alias:          pci:v000010EEd00009021sv*sd*bc*sc*i*
alias:          pci:v000010EEd00009031sv*sd*bc*sc*i*
alias:          pci:v000010EEd00009012sv*sd*bc*sc*i*
alias:          pci:v000010EEd00009022sv*sd*bc*sc*i*
alias:          pci:v000010EEd00009032sv*sd*bc*sc*i*
alias:          pci:v000010EEd00009014sv*sd*bc*sc*i*
alias:          pci:v000010EEd00009024sv*sd*bc*sc*i*
alias:          pci:v000010EEd00009034sv*sd*bc*sc*i*
alias:          pci:v000010EEd00009018sv*sd*bc*sc*i*
alias:          pci:v000010EEd00009028sv*sd*bc*sc*i*
alias:          pci:v000010EEd00009038sv*sd*bc*sc*i*
alias:          pci:v000010EEd0000903Fsv*sd*bc*sc*i*
depends:        
retpoline:      Y
name:           xdma
vermagic:       5.3.0-29-generic SMP mod_unload 
parm:           sgdma_timeout:timeout in seconds for sgdma, default is 10 sec. (uint)
parm:           poll_mode:Set 1 for hw polling, default is 0 (interrupts) (uint)
parm:           interrupt_mode:0 - MSI-x , 1 - MSI, 2 - Legacy (uint)
parm:           enable_credit_mp:Set 0 to disable credit feature, default is 1 ( credit control enabled) (uint)
parm:           desc_blen_max:per descriptor max. buffer length, default is (1 << 28) - 1 (uint)

I am able to see the pci device from lspci:
01:00.0 Memory controller: Xilinx Corporation Device 903f

lesjokolat, 2020-03-09:
what linux environement are you runing?

could it be a csh vs bash setting?

SamuelFarid, 2020-03-09:
It’s ubuntu 19 using bash.

lesjokolat, 2020-03-09:
yah that is only caveat I would worry about v19

modinfo xdma (or) fails to run and its output:
bash: syntax error near unexpected token `('

have you tried the older xdma archive 2018.3?  maybe that will work?

lesjokolat, 2020-03-09:
I am avoiding anything newer than 18.04.2 as Vivado itself doesn't support it. Otherwise you fall under excuse not a supported OS. Imagine the Vivado XDMA IPs you design  they require compatibility and as such may not work.

SamuelFarid, 2020-03-09:
I have tried to use csh instead but the modinfo xdma (or) fails to run also. The older xdma archive fails during the make process. I think I will try to run Ubuntu 18.04 on virtualbox and check. Moreover, I have also tried the windows approach before and the drivers didn't work.

SamuelFarid, 2020-03-10:
I have install Ubuntu 18.04 instead and still the same error exists. Are you using dual boot? When I ran Ubuntu on virtualbox I was able to install the drivers but the virtualmachine itself wasn't able to detect the pci device, that's very weird.

lesjokolat, 2020-03-10:
no my install is a native local host install.

SamuelFarid, 2020-03-10:
Latest kernel version.

SamuelFarid, 2020-03-10:
UPDATE: The driver is finally loaded successfully, the solution was to disable the secure boot.
The problem now is that load_driver cannot recognize any device despite the fact that the device is listed in lspci results:

lesjokolat, 2020-03-12:
if you want to try and use load driver script from the older 65444 driver set. It should make the device viewable

Mahone-h, 2020-08-15:
Error: The Kernel module installed correctly, but no devices were recognized.
Have you solved this problem？

Mahone-h, 2020-08-15:
> UPDATE: The driver is finally loaded successfully, the solution was to disable the secure boot.
> The problem now is that load_driver cannot recognize any device despite the fact that the device is listed in lspci results:
> 01:00.0 Memory controller: Xilinx Corporation Device 903f

Error: The Kernel module installed correctly, but no devices were recognized.
Have you solved this problem？

lesjokolat, 2020-08-18:
I find if i get this error after teh install try to do this.
Launch vivado
re laod bitstream
then reboot

on startup if all goes well try to load driver and see if it helps

Mahone-h, 2020-08-18:
> I find if i get this error after teh install try to do this.
> Launch vivado
> re laod bitstream
> then reboot
> 
> on startup if all goes well try to load driver and see if it helps

Ok, thank you. I have solved this problem. I used MSI interrupt and need to modify libxdma.c.
static unsigned int interrupt_ mode=1.

cr4zy0w1, 2021-01-06:
I am still getting the 'Warning: modules_install: missing 'System.map' file. Skipping depmod.' message even after the tweak in the Makefile #44  and the fix for the #39 SSL issue and secure boot in #52. The XDMA was not installed when I check with modinfo and lsmod commands.

setup as follow:
XDMA v2020.1.8
Ubuntu 18.04.5 LTS 
Linux version 5.4.0-59

lesjokolat, 2021-01-06:
Did you make clean , then install driver?
Also sometimes after you install go to test folder and run load driver script then check depmod

cr4zy0w1, 2021-01-06:
> Did you make clean , then install driver?
> Also sometimes after you install go to test folder and run load driver script then check depmod

Thank you. I forgot to make clean before make install.
Still getting the same warning message but the mod is successfully being installed.
problem solved.