Xilinx/dma_ip_drivers #53: missing 'libaio.h' file

wingsbed, 2020-03-10:
[root@localhost linux-kernel]# make
"distro=RHEL, dmajor=7 dminor=6 "
3.10.0-957.el7.x86_64: RHEL,7,6, -I/lib/modules/3.10.0-957.el7.x86_64/source/arch/x86/include/generated -DRHEL7SP6 -DRHEL7
CROSS_COMPILE_FLAG = .
ARCH = x86_64.
#######################
####  user        ####
#######################
make -C user
make[1]: 进入目录“/home/u50/dma_ip_drivers-master/QDMA/linux-kernel/user”
cc -g -c -std=c99 -o cli/cmd_parse.o cli/cmd_parse.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -g -c -std=c99 -o cli/main.o cli/main.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -g -c -std=c99 -o ../qdmautils/qdmactl.o ../qdmautils/qdmactl.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -g -c -std=c99 -o ../qdmautils/qdmactl_reg.o ../qdmautils/qdmactl_reg.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -g -c -std=c99 -o ../qdmautils/qdmautils.o ../qdmautils/qdmautils.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -g -c -std=c99 -o ../qdmautils/dmaxfer.o ../qdmautils/dmaxfer.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
../qdmautils/dmaxfer.c:34:20: 致命错误：libaio.h：没有那个文件或目录
 #include <libaio.h>
                    ^
编译中断。
make[1]: *** [../qdmautils/dmaxfer.o] 错误 1
make[1]: 离开目录“/home/u50/dma_ip_drivers-master/QDMA/linux-kernel/user”
make: *** [user] 错误 2

hmaarrfk, 2020-03-10:
You need to install libaio

wingsbed, 2020-03-11:
thanks,I got it.

closed 2020-03-11 by wingsbed