Xilinx/dma_ip_drivers #54: read file: Unknown error 512 when start a C2H transfer(stream mode) on Ubuntu 18.04.1(4.15.0-88-generic)

woshixiaohuya, 2020-03-21:
I git the code at 2020-03-17 from the respoitory. 
refer to :https://forums.xilinx.com/t5/PCIe-and-CPM/XDMA-linux-driver-from-git-C2H-transfor-1MB-fail-crash-in-Stream/m-p/1087752#M16120

The cmd is './dma_from_device -d /dev/xdma0_c2h_0 -f data/output_file.bin -s transfersize -c 1'

If I transfer 4KB(transfersize ) one time, the driver work correctly.

When I transfer 1MB(transfersize ) one time, the driver return error.

I read the libxdma.h and found  that some parameters would be related to the transfersize: XDMA_TRANSFER_MAX_DESC and CYCLIC_RX_PAGES_MAX.

I changed XDMA_TRANSFER_MAX_DESC to 4096 and CYCLIC_RX_PAGES_MAX to 1024, then I test the transfer of 1MB but the error still exist:

/dev/xdma0_c2h_0, R off 0x0, 0xffffffffffffffff != 0x100000.
read file: Unknown error 512

lesjokolat, 2020-03-21:
does the solution at bottom here help you?
https://forums.xilinx.com/t5/PCIe-and-CPM/XDMA-driver-test-scripts-error-messages/td-p/912101

woshixiaohuya, 2020-03-22:
> does the solution at bottom here help you?
> https://forums.xilinx.com/t5/PCIe-and-CPM/XDMA-driver-test-scripts-error-messages/td-p/912101

It doesn't work. I think that the offset(0x80000000) solution is mainly used for AXI-Memorymapped mode. 
But I hve tried that and same error. The offset 0x80000000 was added in read_to_buffer() and write_from_buffer() for one parameter(uint64_t base) refer to another question (https://forums.xilinx.com/t5/PCIe-and-CPM/debug-the-driver-of-IP-PCIE-with-DMA/m-p/914480#M12599).

lesjokolat, 2020-03-22:
https://forums.xilinx.com/t5/PCIe-and-CPM/C2H-Streaming-XDMA-Linux-Driver-Broken/td-p/833977

anything in this one?

woshixiaohuya, 2020-03-22:
> https://forums.xilinx.com/t5/PCIe-and-CPM/C2H-Streaming-XDMA-Linux-Driver-Broken/td-p/833977
> 
> anything in this one?
> 
> See this fix done as well.
> 
> [fd5f152](https://github.com/Xilinx/dma_ip_drivers/commit/fd5f152787f567f8237dd7471796beeb8422776e)

Thank you. I have saw that link which has no useful methods. 
I will try the fix as you mentioned. I think that it would be useful. 
I git the source code at 2020-03-17, but my code is different from the fix which is commited 26 days ago.

woshixiaohuya, 2020-03-22:
> https://forums.xilinx.com/t5/PCIe-and-CPM/C2H-Streaming-XDMA-Linux-Driver-Broken/td-p/833977
> 
> anything in this one?
> 
> See this fix done as well.
> 
> [fd5f152](https://github.com/Xilinx/dma_ip_drivers/commit/fd5f152787f567f8237dd7471796beeb8422776e)

Hello! I have tried the  [fd5f152] but it didn't work. I got the same error. 

> /dev/xdma0_c2h_0, R off 0x0, 0xffffffffffffffff != 0x100000.
read file: Unknown error 512

I have read some source code dma_from_device.c and dma_utils.c (https://github.com/Xilinx/dma_ip_drivers/tree/master/XDMA/linux-kernel/tools). It seems that in the read_to_buffer() and write_from_buffer() the read() or write() return -1(0xffffffffffffffff) cause "rc != bytes". But why the two funtions return -1 I don't know.

lesjokolat, 2020-03-22:
Im not a dev but i compared to older 65444 code of dma_from_device.c

/* file argument given? */
		if ((out_fd >= 0) & (no_write == 0)) {
			rc = write_from_buffer(ofname, out_fd, buffer,
					 size, i*size);

is it me or does this duplicate Size variable?
 size or  i*size

should it just be one or the other?

woshixiaohuya, 2020-03-23:
> Im not a dev but i compared to older 65444 code of dma_from_device.c
> 
> /* file argument given? _/ if ((out_fd >= 0) & (no_write == 0)) { rc = write_from_buffer(ofname, out_fd, buffer, size, i_size);
> 
> is it me or does this duplicate Size variable?
> size or i*size
> 
> should it just be one or the other?

I analyzer the old and the new have no problem ethier size or i*size. The difference is that memset () and posix_memalign().

woshixiaohuya, 2020-03-28:
@bhathaway would you give this issue some attention?

lesjokolat, 2020-03-28:
Hi again I am not a dev but I saw on AWS they had similar issues and made some references to 4096 buffer size.

the search results also pointed to some other code section you experts might make sense of. I tried to show first link of where this git has the memalign. Hope it helps.

From:
https://github.com/Xilinx/dma_ip_drivers/blob/8b8c70b697f049649d5fa99be9c6bc4302d89ac9/XDMA/linux-kernel/tools/dma_from_device.c

posix_memalign((void **)&allocated, 4096 /*alignment */ , size + 4096);
	if (!allocated) {
		fprintf(stderr, "OOM %lu.\n", size + 4096);
		rc = -ENOMEM;
		goto out;

From:
https://github.com/aws/aws-fpga/search?q=4096&unscoped_q=4096

//Read from specified address, specified size within a bank
    //Caller's responsibility to do sanity checks. No sanity checks done here
    int readBank(std::ofstream& aOutFile, unsigned long long aStartAddr, unsigned long long aSize) {
      char *buf = 0;
      unsigned long long blockSize = 0x20000;
      if (posix_memalign((void**)&buf, 4096, blockSize))
        return -1;
      std::memset(buf, 0, blockSize);

      if (posix_memalign(&buf, 4096, blockSize+1))//Last is for termination char
        return -1;
      if (posix_memalign(&bufPattern, 4096, blockSize+1)) {//Last is for termination char
        free(buf);
        return -1;

From:
https://github.com/aws/aws-fpga/blob/7f1e76765766f579d3a767bedf669019d50342f3/hdk/common/software/src/xdma_utils.c

  h2c_desc_table = (XDMA_DESC *)memalign(4096, 4096);  // allocate 4k aligned to a 4k boundary
    
    sv_map_host_memory(h2c_desc_table);

hmaarrfk, 2020-03-28:
I'm just going to say that @bhathaway is not affiliated with Xilinx and is one of their customers. You should probaly ping Xilinx harder and not a fellow user.

Unless we decide to create our own fork, which we could probably maintain with little effort.

woshixiaohuya, 2020-03-28:
> I'm just going to say that @bhathaway is not affiliated with Xilinx and is one of their customers. You should probaly ping Xilinx harder and not a fellow user.
> 
> Unless we decide to create our own fork, which we could probably maintain with little effort.

Sorry, I thought you were the developer of this driver. I made a wrong judgment.

hmaarrfk, 2020-03-28:
no, you can see that their PR is in limbo just like the rest of our issues ;)

XIlinx basically refuses to interact here. It is likely because their management banned them from using unauthorized channels to communicate. So just keep pinging them on their forums or through your FAE (but they likely aren't PCIe experts).

woshixiaohuya, 2020-03-29:
> no, you can see that their PR is in limbo just like the rest of our issues ;)
>
> XIlinx basically refuses to interact here. It is likely because their management banned them from using unauthorized [...]

Okay，thanks very much!

woshixiaohuya, 2020-03-29:
@karenx-xilinx I guess you are the developer of Xilinx, would you give this issue some attention? This issue has been posted on Xilinx community forum but without reply.

woshixiaohuya, 2020-04-08:
Item | Settings | result
-- | -- | --
1 | XDMA IP enable MSI-X，disable   Legacy and MSI；change load_driver.sh：insmod ../xdma/xdma.ko interrupt_mode=0 | C2H 1MB one count：read file:   Unknown error 512(dmesg log4)；ILA show that datagenerator send 128KB to XDMA   IP
2 | XDMA IP enable Legacy，disable   MSI-X and MSI；change load_driver.sh：insmod ../xdma/xdma.ko interrupt_mode=2 | C2H 1MB one count：read file:   Unknown error 512；ILA show that datagenerator send 128KB to XDMA IP
3 | XDMA IP enable MSI，disable MSI-X   and Legacy；change load_driver.sh：insmod ../xdma/xdma.ko interrupt_mode=1 | C2H 1MB one count：read file:   Unknown error 512；ILA show that datagenerator send 424KB to XDMA IP.                                  C2H 4KB   4096 count:read file: Unknown error 512(dmesg log5)；

lesjokolat, 2020-06-04:
See this one 
https://forums.xilinx.com/t5/PCIe-and-CPM/memory-mapped-XDMA-c2h-from-Stream-FIFO/m-p/1109864#M16539

512 error and xdma stream

woshixiaohuya, 2020-06-04:
> See this one
> https://forums.xilinx.com/t5/PCIe-and-CPM/memory-mapped-XDMA-c2h-from-Stream-FIFO/m-p/1109864#M16539
> 
> 512 error and xdma stream

In the reference forums it's about memory mapped mode, but this issue was happened on stream mode

hudaweidavid, 2020-07-07:
I had the same problem, If the amount of data exceeds 8K, it will time out

woshixiaohuya, 2020-07-15:
> I had the same problem, If the amount of data exceeds 8K, it will time out

This problem already exists a long time. The developers make more effort on AXI memory mapped mode.  That troubles us that using AXI Stream mode.