Xilinx/dma_ip_drivers #59: QDMA 2019.2 fails on Ubuntu 20.04

hmaarrfk, 2020-06-04:
When trying to load the kernel driver, `dmesg` reports the following:
```
09:12 $ dmesg | grep qdma
[  539.111014] qdma: module verification failed: signature and/or required key missing - tainting kernel
[  539.111871] qdma:qdma_mod_init: Xilinx QDMA PF Reference Driver v2019.2.125.213.
[  539.112579] qdma:probe_one: 0000:0a:00.0: func 0x0/0x4, p/v 0/0,0x0000000017f6a75d.
[  539.112581] qdma:probe_one: Configuring '0a:00:0' as master pf
[  539.112581] qdma:probe_one: Driver is loaded in auto(0) mode
[  539.112583] qdma:qdma_device_open: qdma_pf, 0a:00.00, pdev 0x00000000b1bc043b, 0x10ee:0x903f.
[  539.112593] qdma_pf 0000:0a:00.0: enabling device (0000 -> 0002)
[  539.314438] qdma_get_device_type: invalid device type(3), err:-1
[  539.314440] qdma_get_versal_ip_type: inv versal_ip_type(2), err:-1
[  539.516157] qdma_get_device_type: invalid device type(3), err:-1
[  539.516158] qdma_get_versal_ip_type: inv versal_ip_type(2), err:-1
[  540.323550] qdma:qdma_device_attributes_get: qdma0a000-p0000:0a:00.0: num_pfs:4, num_qs:4095, flr_present:1, st_en:1, mm_en:1, mm_cmpt_en:1, mailbox_en:1, mm_channel_max:1, qid2vec_ctx:0, cmpt_ovf_chk_dis:1, mailbox_intr:1, sw_desc_64b:1, cmpt_desc_64b:1, dynamic_bar:1, legacy_intr:1, cmpt_trig_count_timer:1
[  540.323564] qdma:qdma_device_open: Vivado version = 
[  540.727272] qdma:xdev_identify_bars: User BAR 0.
[  540.929001] qdma_dev_entry_create: Created the dev entry successfully
[  542.139486] qdma:intr_setup: Error enabling MSI-X (-12)
[  542.139492] qdma:qdma_device_online: Failed to setup interrupts, err -12
[  542.139493] qdma:qdma_device_open: Failed to set the dma device  online, err = -22
[  542.543949] qdma_pf: probe of 0000:0a:00.0 failed with error -22
```

```
09:12 $ uname -a
Linux sparrow 5.4.0-33-generic #37-Ubuntu SMP Thu May 21 12:53:59 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

When compiling from the 2019.1 sources on ubuntu 18.04 with kernel 4.X, the driver loads as expected.

hmaarrfk, 2020-06-04:
It also fails on Ubuntu 18.04 with Kernel 5.3

```
09:19 $ dmesg | grep qdma
[  205.367307] qdma:qdma_mod_init: Xilinx QDMA PF Reference Driver v2019.2.125.213.
[  205.368093] qdma:probe_one: 0000:0a:00.0: func 0x0/0x4, p/v 0/0,0x000000006085203e.
[  205.368095] qdma:probe_one: Configuring '0a:00:0' as master pf
[  205.368096] qdma:probe_one: Driver is loaded in auto(0) mode
[  205.368097] qdma:qdma_device_open: qdma_pf, 0a:00.00, pdev 0x0000000000c37130, 0x10ee:0x903f.
[  205.368106] qdma_pf 0000:0a:00.0: enabling device (0000 -> 0002)
[  205.569949] qdma_get_device_type: invalid device type(3), err:-1
[  205.569952] qdma_get_versal_ip_type: inv versal_ip_type(2), err:-1
[  205.771666] qdma_get_device_type: invalid device type(3), err:-1
[  205.771668] qdma_get_versal_ip_type: inv versal_ip_type(2), err:-1
[  206.578704] qdma:qdma_device_attributes_get: qdma0a000-p0000:0a:00.0: num_pfs:4, num_qs:4095, flr_present:1, st_en:1, mm_en:1, mm_cmpt_en:1, mailbox_en:1, mm_channel_max:1, qid2vec_ctx:0, cmpt_ovf_chk_dis:1, mailbox_intr:1, sw_desc_64b:1, cmpt_desc_64b:1, dynamic_bar:1, legacy_intr:1, cmpt_trig_count_timer:1
[  206.578716] qdma:qdma_device_open: Vivado version = 
[  206.982150] qdma:xdev_identify_bars: User BAR 0.
[  207.183866] qdma_dev_entry_create: Created the dev entry successfully
[  208.394150] qdma:intr_setup: Error enabling MSI-X (-12)
[  208.394151] qdma:qdma_device_online: Failed to setup interrupts, err -12
[  208.394152] qdma:qdma_device_open: Failed to set the dma device  online, err = -22
[  208.797883] qdma_pf: probe of 0000:0a:00.0 failed with error -22
09:19 $ uname -a
Linux sparrow 5.3.0-53-generic #47~18.04.1-Ubuntu SMP Thu May 7 13:10:50 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

hmaarrfk, 2020-06-04:
Finally, it works with Ubuntu 18.04 and Kernel 4.15

```
09:21 $ dmesg | grep qdma
[    2.440078] qdma_vf: loading out-of-tree module taints kernel.
[    2.440152] qdma_vf: module verification failed: signature and/or required key missing - tainting kernel
[    2.440571] qdma_vf:qdma_mod_init: Xilinx QDMA VF Reference Driver v2019.1.121.207.
[    2.446281] qdma:qdma_mod_init: Xilinx QDMA PF Reference Driver v2019.1.121.207.
[    2.447205] qdma:probe_one: 0000:0a:00.0: func 0x0/0x4, p/v 0/0,0x          (null).
[    2.447206] qdma:probe_one: Current device is in 0 mode
[    2.447207] qdma:probe_one: Configuring '0a:00:0' as master pf
[    2.447207] qdma:probe_one: Driver is loaded in auto mode
[    2.447208] qdma:qdma_device_open: qdma_pf, 0a:00.00, pdev 0x        (ptrval), 0x10ee:0x903f.
[    2.447216] qdma_pf 0000:0a:00.0: enabling device (0000 -> 0002)
[    2.447307] qdma:xdev_identify_bars: QDMA Config BAR passed by the user is correct
[    2.447309] qdma:xdev_identify_bars: User BAR 2.
[    2.447315] qdma:qdma_device_attributes_get: qdma0a000-p0000:0a:00.0: num_pfs:1, num_qs:64, flr_present:0, st_en:1, mm_en:1, mm_cmpt_en:0, mailbox_en:0, mm_channel_max:1
[    2.450806] qdma:qdma_device_open: 0000:0a:00.0, 0a000, pdev 0x        (ptrval), xdev 0x        (ptrval), ch 1, q 0, vf 0.
(mcam_dev) ✔ ~ 
09:21 $ dmes^C
09:21 $ uname -a
Linux sparrow 4.15.0-101-generic #102-Ubuntu SMP Mon May 11 10:07:26 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

hmaarrfk, 2020-06-04:
Finally, the last combination is:

kernel 4 + 2019.2 driver, also fails

```
10:19 $ dmesg | grep qdma
[    2.546932] qdma_vf: loading out-of-tree module taints kernel.
[    2.547112] qdma_vf: module verification failed: signature and/or required key missing - tainting kernel
[    2.547748] qdma_vf:qdma_mod_init: Xilinx QDMA VF Reference Driver v2019.2.125.213.
[    2.559099] qdma:qdma_mod_init: Xilinx QDMA PF Reference Driver v2019.2.125.213.
[    2.562029] qdma:probe_one: 0000:0a:00.0: func 0x0/0x4, p/v 0/0,0x          (null).
[    2.562031] qdma:probe_one: Configuring '0a:00:0' as master pf
[    2.562031] qdma:probe_one: Driver is loaded in auto(0) mode
[    2.562032] qdma:qdma_device_open: qdma_pf, 0a:00.00, pdev 0x        (ptrval), 0x10ee:0x903f.
[    2.562042] qdma_pf 0000:0a:00.0: enabling device (0000 -> 0002)
[    2.763908] qdma_get_device_type: invalid device type(3), err:-1
[    2.763911] qdma_get_versal_ip_type: inv versal_ip_type(2), err:-1
[    2.965626] qdma_get_device_type: invalid device type(3), err:-1
[    2.965627] qdma_get_versal_ip_type: inv versal_ip_type(2), err:-1
[    3.772481] qdma:qdma_device_attributes_get: qdma0a000-p0000:0a:00.0: num_pfs:4, num_qs:4095, flr_present:1, st_en:1, mm_en:1, mm_cmpt_en:1, mailbox_en:1, mm_channel_max:1, qid2vec_ctx:0, cmpt_ovf_chk_dis:1, mailbox_intr:1, sw_desc_64b:1, cmpt_desc_64b:1, dynamic_bar:1, legacy_intr:1, cmpt_trig_count_timer:1
[    3.772670] qdma:qdma_device_open: Vivado version = 
[    4.176092] qdma:xdev_identify_bars: User BAR 0.
[    4.377807] qdma_dev_entry_create: Created the dev entry successfully
[    5.588065] qdma:intr_setup: Error enabling MSI-X (-12)
[    5.588066] qdma:qdma_device_online: Failed to setup interrupts, err -12
[    5.588067] qdma:qdma_device_open: Failed to set the dma device  online, err = -22
[    5.991558] qdma_pf: probe of 0000:0a:00.0 failed with error -22
10:19 $ uname -a
Linux sparrow 4.15.0-101-generic #102-Ubuntu SMP Mon May 11 10:07:26 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

sujathabanoth-xlnx, 2021-01-29:
This seems to be environment issue, check at your end

closed 2021-01-29 by sujathabanoth-xlnx

hmaarrfk, 2021-01-30:
Thanks for the reply. I think it ended up being addressed in 2020.1.

We had to jump from 2019.1 to 2020.1

Ganesh93904, 2024-11-18:
When trying to load the kernel driver, dmesg reports the following:

[434533.158745] qdma_pf:qdma_mod_init: Xilinx QDMA PF Reference Driver v2023.2.0.0.
[434533.159053] qdma_pf:probe_one: 0000:01:00.0: func 0x0, p/v 0/0,0x0000000000000000.
[434533.159057] qdma_pf:probe_one: Configuring '01:00:0' as master pf
[434533.159059] qdma_pf:probe_one: Driver is loaded in auto(0) mode
[434533.159061] qdma_pf:qdma_device_open: qdma-pf, 01:00.00, pdev 0x000000009aa5ba17, 0x10ee:0x9038.
[434533.159227] entering into the qdma_is_configbar
[434533.159232] reg_address : 0
[434533.159233] reg_value is : -1
[434533.159234] field get value is :65535
[434533.159235] qdma_is_config_bar: Invalid config bar, err:-4
[434533.159236] qdma_hw_access_init: config bar passed is INVALID, err:-1
[434533.159330] qdma-pf: probe of 0000:01:00.0 failed with error -22

uname -a
Linux lab 5.15.0-124-generic #134-Ubuntu SMP Fri Sep 27 20:20:17 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

can anyone help to resolve this issue

Ganesh93904, 2024-11-18:
Thanks for the reply. I didn't do any modifications in the open/close state, but still getting the same issue
i took the driver from this link
https://github.com/Xilinx/dma_ip_drivers.git

devegali, 2024-11-18:
Hi Ganesh,

Have you enabled the Config BAR in your design?
If the Config BAR is enabled, which BAR is it connected to? Please ensure that the same Config BAR is specified in the qdma.conf file.

Please refer the following wiki page to add the proper config BAR in qdma.conf file.
https://xilinx.github.io/dma_ip_drivers/master/QDMA/linux-kernel/html/build.html#dynamic-config-bar

Ganesh93904, 2024-11-18:
Hi devegali,
Thanks for the reply 

I enabled and using BAR 0 in my design and i mentioned same in my qdma.conf file according to the document, i cross checked it and still getting the same error.

Here is my qdma.conf file
options qdma-pf mode=01:0:1
options qdma-pf config_bar=01:0:0
options qdma-pf master_pf=01:0
options qdma-vf mode=01:0:1
options qdma-vf config_bar=01:0:0

devegali, 2024-11-18:
Hi Ganesh,

Can please share the o/p of below command
**_lspci -s <B:D.F> -vvv_**

Ganesh93904, 2024-11-19:
can any one help me with this issue, I am really stuck here, it will be really helpful

devegali, 2024-11-19:
Hi Ganesh,

Which board your using for verification is it Soft IP or Hard IP?

Ganesh93904, 2024-11-19:
Hi,
It consists of both PS and PL

devegali, 2024-11-21:
Hi Ganesh,

Can please share the Vivado tool snap chats of below details.
1. Device/Board TAB
2. IP TAB 
3. PCI BAR's TAB

Ganesh93904, 2024-11-22:
Hi devendar

Here are the snap chats that you asked
[image]
[image]
[image]
[image]

devegali, 2024-11-24:
Hi Ganesh,

Could you please configure PF0_BAR0 to use the DMA type instead of memory?

Ganesh93904, 2024-11-25:
Hi devendar 
[image]

in the above image i have only memory option no DMA_memory

Ganesh93904, 2024-11-27:
Hi devendar

Here is the document link
https://www.amd.com/content/dam/xilinx/support/documents/ip_documentation/xdma/v4_1/pg195-pcie-dma.pdf