Xilinx/dma_ip_drivers #62: QDMA Driver timeout during C2H and H2C Requests

kewlcreatures, 2020-06-17:
While operating in MM Mode for both C2H and H2C 1 queue each with the ring and buffer size as 4096 we see these errors in the kernel.
[Tue Jun 16 20:58:04 2020] qdma:qdma_request_wait_for_cmpl: qdma82000-MM-1: req 0xffff9cb0da3bbdf8, R,256,0/256,0x0, done 0, err 0, tm 10000.
[Tue Jun 16 20:58:04 2020] qdma:qdma_descq_dump: qdma82000-MM-1: 0x1/0x1, desc sz 4096/4056, pidx 1783, cidx 1744_
A deeper dive into the kernel code shows that this occurs due to a timeout at this place in the code
https://github.com/Xilinx/dma_ip_drivers/blob/2018.3/QDMA/linux-kernel/libqdma/libqdma_export.c#L79
Due to this the file descriptor returns with an error -5. This issue occurs for both directions C2H and H2C above being an example of C2H.
We see this issue on 2018.3 driver code , however have not ascertained if the same issue is seen on the 2019.2 driver code.
We are using the linux driver.
Any help would be appreciated towards the resolution of this issue as its a showstopper for us.

kewlcreatures, 2020-07-07:
@karenx-xilinx  is it possible for some one to look at this issue.

sujathabanoth-xlnx, 2021-01-29:
When completions are timing out, issue is most probably with the bitstream being generated. Check once again at your end. As the issue is not seen on latest driver release, closing it

closed 2021-01-29 by sujathabanoth-xlnx

dmax621, 2021-06-30:
This look very like the [bug I've identified in qdma_request_wait_for_cmpl()](https://forums.xilinx.com/t5/PCIe-and-CPM/QDMA-Error-Timeout-for-completion-Important-and-Urgent/m-p/1254834/highlight/true#M19066) which results in "impossible" timeouts. I have a fix that seems to work -- can we reopen this issue, or shall I start a new one?