Xilinx/dma_ip_drivers #63: PCIE to thunderbolt

TengHsiang, 2020-06-22:
Hi,
I can use xdma to communicate with FPGA now.
But for our application, we need to use the external PCIE slut (Razer Core X).
Does there have any drivers can support it?
Thanks

zpillio, 2025-02-04:
I wanted to ask the same question. Have you found a solution?

martindubois, 2025-02-04:
The same driver can be use but you need to disable the DMA remapping. This is a simple line to add to the INF file.

https://learn.microsoft.com/en-us/windows-hardware/drivers/pci/enabling-dma-remapping-for-device-drivers
explain what is DMA remapping and how to disable it.

Have fun!

zpillio, 2025-03-07:
Thank you for your answer. I'm using linux and not windows. I gave a try and disabled the DMA remapping, but it didn't work for me.