Xilinx/dma_ip_drivers #64: xdma driver 512 error

shahbaazlokh, 2020-07-08:
I am using 1 stream interface with xdma configured as gen2x2, with vcu1525

i have instaled drivers and performed loopback test etc, works fine,

However, when i run my design initially it works well, but after some time it gives error 512,

my linux kernal version is, 5.4.45-050445-generic

here is the dmesg log

[Wed Jul 8 04:16:53 2020] xdma:xdma_mod_init: Xilinx XDMA Reference Driver xdma v2019.2.51
[Wed Jul 8 04:16:53 2020] xdma:xdma_mod_init: desc_blen_max: 0xfffffff/268435455, sgdma_timeout: 10 sec.
[Wed Jul 8 04:16:53 2020] xdma:xdma_threads_create: xdma_threads_create
[Wed Jul 8 04:16:53 2020] xdma:xdma_device_open: xdma device 0000:0b:00.0, 0x000000005e92819d.
[Wed Jul 8 04:16:53 2020] xdma:map_single_bar: BAR0 at 0xfce00000 mapped at 0x0000000055943a4f, length=65536(/65536)
[Wed Jul 8 04:16:53 2020] xdma:map_bars: config bar 0, pos 0.
[Wed Jul 8 04:16:53 2020] xdma:identify_bars: 1 BARs: config 0, user -1, bypass -1.
[Wed Jul 8 04:16:53 2020] xdma:probe_one: 0000:0b:00.0 xdma0, pdev 0x000000005e92819d, xdev 0x0000000075c79a4a, 0x000000008d1118f1, usr 16, ch 1,1.
[Wed Jul 8 04:23:21 2020] xdma:xdma_xfer_submit: xfer 0x00000000ae8f65d9,32, s 0x1 timed out, ep 0x20.
[Wed Jul 8 04:23:21 2020] xdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x000000003470a31b) = 0xffffffff (id).
[Wed Jul 8 04:23:21 2020] xdma:engine_reg_dump: 0-C2H0-ST: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[Wed Jul 8 04:23:21 2020] xdma:engine_status_read: Failed to dump register
[Wed Jul 8 04:23:21 2020] xdma:xdma_xfer_submit: Failed to read engine status

lukasfischer83, 2020-07-28:
Are you maybe running into a timeout by feeding data too slow into the xdma block?
There is a kernel parameter to increase the timeout, maybe it helps for testing.
Try:
`sudo modprobe xdma sgdma_timeout=1000`

dmitrym1, 2025-01-17:
#317