Xilinx/dma_ip_drivers #75: Every occurance of sprintf is a bug

hmaarrfk, 2020-09-08:
Seeing as this is a kernel driver, with the possibility of taking down an entire operating system, it seems to me that we should strive to minimize calls to `sprintf`.

The safe version of this function is `snprintf` which includes an argument for the length of the string.

Because of the unsafe nature of this call, every call to it is a bug and should be addressed.

Once can use the following command:

```
grep sprintf\( QDMA -R | wc -l
grep sprintf\( XDMA -R | wc -l
```
To find:

* 66 occurrences in QDMA
* 54 occurrences in XDMA

I've raised this issue a few times last year, and it seemed to come on def ears.

We can fix things ourselves, if our patches will be accepted. It is really annoying to have to go through a vetting process and redo all our work every 6 months due to a release dump on every Vivado release.

Beyond this, it seems that the `nl` portion of the driver tries to copy un-initialized strings to the user.
Again, I had fixed a few of these in 2019.1, **and** submitted patches but then when the 2020.1 release came along, all my work was useless due to the fact that there were major changes.

Please advise.

jdmdj, 2020-10-08:
Totally agree.

Here's one very shallow sprintf() bug I fixed (and fixed again in 2020.1):

Given this definition of xdma_engine in libxdma.h:
struct xdma_engine {
	unsigned long magic;	/* structure ID for sanity checks */
	struct xdma_dev *xdev;	/* parent device */
	char name[5];	/* name of this engine */      <------------------- name field is 5 bytes
	int version;		/* version of this engine */
        ...
};

The call to sprintf(3) in libxdma.c:engine_init() looks very suspicious:

> make install
Makefile:10: XVC_FLAGS: .
make -C /lib/modules/3.10.0-1127.19.1.el7.x86_64/build M=/tmp/dma_ip_drivers/XDMA/linux-kernel/xdma modules
make[1]: Entering directory `/usr/src/kernels/3.10.0-1127.19.1.el7.x86_64'
/tmp/dma_ip_drivers/XDMA/linux-kernel/xdma/Makefile:10: XVC_FLAGS: .
  CC [M]  /tmp/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.o
/tmp/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c: In function ‘probe_for_engine’:
/tmp/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c:3077:30: warning: ‘%d’ directive writing between 1 and 3 bytes into a region of size 0 [-Wformat-overflow=]
  sprintf(engine->name, "%d-%s%d-%s", xdev->idx,
                              ^~
/tmp/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c:3077:24: note: directive argument in the range [-22, 15]
  sprintf(engine->name, "%d-%s%d-%s", xdev->idx,
                        ^~~~~~~~~~~~
/tmp/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c:3077:2: note: ‘sprintf’ output between 10 and 22 bytes into a destination of size 5
  sprintf(engine->name, "%d-%s%d-%s", xdev->idx,
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   (dir == DMA_TO_DEVICE) ? "H2C" : "C2H", channel,
   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   engine->streaming ? "ST" : "MM");
   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

jberaud, 2020-10-09:
Just posted #80 which addresses the one outputting a warning. I will see if I can find time to fix the others in xdma.

sujathabanoth-xlnx, 2021-01-29:
Issue addressed in latest driver release, hence closing it

closed 2021-01-29 by sujathabanoth-xlnx

hmaarrfk, 2021-01-30:
Thanks for addressing the issues in QDMA @sujathabanoth-xlnx 
XDMA still seems to have some issues but I guess somebody else can open an issue regarding that one.