Xilinx/dma_ip_drivers #81: xmda  make install error

crizy, 2020-10-12:
ubuntu  16.04

libxdma.c    to many error

hmaarrfk, 2020-10-12:
It might be that ubuntu 16.04 is too old. But do share at least the first error.

hankhsu1996, 2020-10-15:
Ubuntu 18 works fine. Can't help you unless you give some hint.

jberaud, 2020-10-15:
ubuntu 16.04 works for me.

crizy, 2020-10-27:
error log

/opt/xdma_driver/driver_laster/dma_ip_drivers-master/XDMA/linux-kernel/xdma/libxdma.c:1:1: error: expected identifier or ‘(’ before ‘.’ token  ../libxdma/libxdma.c

crizy, 2020-10-27:
error log as follows,

/opt/xdma_driver/driver_laster/dma_ip_drivers-master/XDMA/linux-kernel/xdma/libxdma.c:1:1: error: expected identifier or ‘(’ before ‘.’ token ../libxdma/libxdma.c

then  I replaced linux-kernel/xdma/libxdma.c with /libxdma/libxdma.c ,   then "make install"  too many error

hmaarrfk, 2020-10-27:
Again, please upload the full output. 

Please redirect the output to a file, and upload the whole file.

```
make | tee errorlog.txt
```
Then upload the errorlog txt file

crizy, 2020-10-28:
make error  log as follows,

 
root@123:/opt/xdma_driver/driver_laster/dma_ip_drivers-master/XDMA/linux-kernel/xdma# make install
Makefile:10: XVC_FLAGS: .
make -C /lib/modules/4.15.0-122-generic/build M=/opt/xdma_driver/driver_laster/dma_ip_drivers-master/XDMA/linux-kernel/xdma modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-122-generic'
/opt/xdma_driver/driver_laster/dma_ip_drivers-master/XDMA/linux-kernel/xdma/Makefile:10: XVC_FLAGS: .
  CC [M]  /opt/xdma_driver/driver_laster/dma_ip_drivers-master/XDMA/linux-kernel/xdma/libxdma.o
/opt/xdma_driver/driver_laster/dma_ip_drivers-master/XDMA/linux-kernel/xdma/libxdma.c:1:1: error: expected identifier or ‘(’ before ‘.’ token
 ../libxdma/libxdma.c
 ^
scripts/Makefile.build:330: recipe for target '/opt/xdma_driver/driver_laster/dma_ip_drivers-master/XDMA/linux-kernel/xdma/libxdma.o' failed
make[2]: *** [/opt/xdma_driver/driver_laster/dma_ip_drivers-master/XDMA/linux-kernel/xdma/libxdma.o] Error 1
Makefile:1585: recipe for target '_module_/opt/xdma_driver/driver_laster/dma_ip_drivers-master/XDMA/linux-kernel/xdma' failed
make[1]: *** [_module_/opt/xdma_driver/driver_laster/dma_ip_drivers-master/XDMA/linux-kernel/xdma] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-122-generic'
Makefile:27: recipe for target 'all' failed
make: *** [all] Error 2
root@123:/opt/xdma_driver/driver_laster/dma_ip_drivers-master/XDMA/linux-kernel/xdma#

hmaarrfk, 2020-10-28:
Can you please format your code correctly for GitHub.

It is a really small thing, bug it makes the output much more readable.

You should use triple back ticks

crizy, 2020-10-28:
After I re-download, it's ok，thanks

hmaarrfk, 2020-10-28:
Happy to hear that you got things working.

Please f find some time to close this issues :D