Xilinx/dma_ip_drivers #82: modprobe xdma error

Vae1997, 2020-10-14:
Hello
I'm trying to compile the latest version of xdma driver on my zcu102 board.
But when I want to load xdma.ko through modprobe ,the following error occur from dmesg:
```
root@zcu102:~# modprobe xdma
[   80.864089] Unable to handle kernel write to read-only memory at virtual address ffffff8000c011d8
[   80.873012] Mem abort info:
[   80.875804]   Exception class = DABT (current EL), IL = 32 bits
[   80.881757]   SET = 0, FnV = 0
[   80.884812]   EA = 0, S1PTW = 0
[   80.887949] Data abort info:
[   80.890825]   ISV = 0, ISS = 0x0000004f
[   80.894668]   CM = 0, WnR = 1
[   80.897642] swapper pgtable: 4k pages, 39-bit VAs, pgd = ffffff80092a0000
[   80.904432] [ffffff8000c011d8] *pgd=000000087fffe003, *pud=000000087fffe003, *pmd=000000086558f003, *pte=00c0000874b97793
[   80.915398] Internal error: Oops: 9600004f [#1] SMP
[   80.920259] Modules linked in: xdma(O) zocl(O) mali(O) uio_pdrv_genirq
[   80.926782] CPU: 0 PID: 6042 Comm: modprobe Tainted: G           O    4.14.0-xilinx-v2018.3 #1
[   80.935380] Hardware name: ZynqMP ZCU102 Rev1.0 (DT)
[   80.940330] task: ffffffc87370a380 task.stack: ffffff8009548000
[   80.946239] PC is at rb_erase+0x9c/0x3c0
[   80.950150] LR is at __mod_tree_remove+0x5c/0x70
[   80.954757] pc : [<ffffff8008b717cc>] lr : [<ffffff800810c5d4>] pstate: 60000145
[   80.962135] sp : ffffff800954bca0
[   80.965434] x29: ffffff800954bca0 x28: ffffff80096a1000
[   80.970729] x27: 00000000014000c0 x26: 0000000000000124
[   80.976024] x25: ffffff800954be50 x24: 0000000000000001
[   80.981319] x23: 0000000000000001 x22: ffffffc8320dac00
[   80.986615] x21: ffffff8008f86c00 x20: ffffff8000c074e8
[   80.991910] x19: ffffff8008f86000 x18: 0000000000000000
[   80.997205] x17: 0000007fbe55d958 x16: 000000558f418ab0
[   81.002501] x15: ffffffffffffffff x14: ffffffc82869491c
[   81.007805] x13: 0000000000000040 x12: 0000000000000000
[   81.013100] x11: 0101010101010101 x10: 7f7f7f7f7f7f7f7f
[   81.018395] x9 : ffffff8008f71000 x8 : ffffffc82bc8b680
[   81.023691] x7 : 0000000000000000 x6 : 0000000000012000
[   81.028986] x5 : 0000000000000000 x4 : ffffff8000c011d8
[   81.034281] x3 : ffffff8000bf1620 x2 : ffffff8000bf1638
[   81.039577] x1 : ffffff8008f86c10 x0 : ffffff8000c07508
[   81.044874] Process modprobe (pid: 6042, stack limit = 0xffffff8009548000)
[   81.051730] Call trace:
[   81.054163] Exception stack(0xffffff800954bb60 to 0xffffff800954bca0)
[   81.060587] bb60: ffffff8000c07508 ffffff8008f86c10 ffffff8000bf1638 ffffff8000bf1620
[   81.068408] bb80: ffffff8000c011d8 0000000000000000 0000000000012000 0000000000000000
[   81.076221] bba0: ffffffc82bc8b680 ffffff8008f71000 7f7f7f7f7f7f7f7f 0101010101010101
[   81.084034] bbc0: 0000000000000000 0000000000000040 ffffffc82869491c ffffffffffffffff
[   81.091846] bbe0: 000000558f418ab0 0000007fbe55d958 0000000000000000 ffffff8008f86000
[   81.099659] bc00: ffffff8000c074e8 ffffff8008f86c00 ffffffc8320dac00 0000000000000001
[   81.107472] bc20: 0000000000000001 ffffff800954be50 0000000000000124 00000000014000c0
[   81.115285] bc40: ffffff80096a1000 ffffff800954bca0 ffffff800810c5d4 ffffff800954bca0
[   81.123097] bc60: ffffff8008b717cc 0000000060000145 0000000000000001 0000000000000001
[   81.130910] bc80: 0000008000000000 ffffff800810c13c ffffff800954bca0 ffffff8008b717cc
[   81.138726] [<ffffff8008b717cc>] rb_erase+0x9c/0x3c0
[   81.143681] [<ffffff8008110f78>] do_init_module+0x108/0x1b0
[   81.149244] [<ffffff8008110018>] load_module+0x1d20/0x2298
[   81.154713] [<ffffff80081107b0>] SyS_finit_module+0xa8/0xc0
[   81.160268] Exception stack(0xffffff800954bec0 to 0xffffff800954c000)
[   81.166693] bec0: 0000000000000003 0000005560eee7b8 0000000000000000 0000000000000003
[   81.174505] bee0: 0000000000000000 0000000000000c61 0000000000000000 0000000000000000
[   81.182319] bf00: 0000000000000111 0000007f97dbe2a0 0101010101010101 0000000000000000
[   81.190131] bf20: 0000000000000001 000000000000270f 0000007f97c2b308 0000007f97c1de08
[   81.197944] bf40: 0000007f97ce68d0 0000005560f012b0 000000000000036d 0000005583c559c0
[   81.205757] bf60: 0000000000000000 0000005560eee7b8 0000005583c5d740 0000000000040000
[   81.213569] bf80: 0000000000000000 0000005583c55440 0000000000000000 0000000000040000
[   81.221382] bfa0: 0000000000000000 0000007fe35ad810 0000005560ee5b90 0000007fe35ad810
[   81.229195] bfc0: 0000007f97ce68f4 0000000080000000 0000000000000003 0000000000000111
[   81.237007] bfe0: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[   81.244823] [<ffffff80080837b0>] el0_svc_naked+0x24/0x28
[   81.250127] Code: b2400042 f90000e2 d65f03c0 f9400002 (f9000082)
[   81.256211] ---[ end trace 1c9a839217d7be36 ]---
[   81.260819] Kernel panic - not syncing: Fatal exception
[   81.266028] SMP: stopping secondary CPUs
[   81.269934] Kernel Offset: disabled
[   81.273406] CPU features: 0x002004
[   81.276790] Memory Limit: none
[   81.279831] Rebooting in 10 seconds..
```
Every time when I boot my sys,I can find the following message about my memory from series port:
```
[    0.000000] Memory: 2987756K/4193280K available (11324K kernel code, 782K rwdata, 3504K rodata, 512K init, 2404K bss, 156948K reserved, 1048576K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000]       .text : 0xffffff8008080000 - 0xffffff8008b90000   ( 11328 KB)
[    0.000000]     .rodata : 0xffffff8008b90000 - 0xffffff8008f00000   (  3520 KB)
[    0.000000]       .init : 0xffffff8008f00000 - 0xffffff8008f80000   (   512 KB)
[    0.000000]       .data : 0xffffff8008f80000 - 0xffffff8009043a00   (   783 KB)
[    0.000000]        .bss : 0xffffff8009043a00 - 0xffffff800929cae0   (  2405 KB)
[    0.000000]     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbf00000000 - 0xffffffbf22000000   (   544 MB actual)
[    0.000000]     memory  : 0xffffffc000000000 - 0xffffffc880000000   ( 34816 MB)

```

uname -a :
Linux zcu102 4.14.0-xilinx-v2018.3 #1 SMP Thu Jun 25 12:38:16 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
 
By the way,the sys on zcu102 is made by petalinux2018.3

So what should I do? Thank you~

jberaud, 2020-10-15:
Why would you want to build the xdma driver on a zynqmp platform ? Do you want to connect to a pci xdma controller ?

Vae1997, 2020-10-15:
> Why would you want to build the xdma driver on a zynqmp platform ? Do you want to connect to a pci xdma controller ?

Yes，I just want to try to connect other xilinx fpga device through pcie on my 102 board.

Vae1997, 2020-10-15:
> Why would you want to build the xdma driver on a zynqmp platform ? Do you want to connect to a pci xdma controller ?

I just referred "Xilinx_Answer_71493_ZCU102_ZC706.pdf",when I lspci：
```
root@zcu102:~# lspci -vv
00:00.0 PCI bridge: Xilinx Corporation Device d021 (prog-if 00 [Normal decode])
        Control: I/O- Mem- BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 54
        Bus: primary=00, secondary=01, subordinate=0c, sec-latency=0
        I/O behind bridge: 00000000-00000fff
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity- SERR- NoISA- VGA- MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot+,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [60] Express (v2) Root Port (Slot-), MSI 00
                DevCap: MaxPayload 256 bytes, PhantFunc 0
                        ExtTag- RBE+
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend+
                LnkCap: Port #0, Speed 5GT/s, Width x1, ASPM not supported, Exit Latency L0s unlimited, L1 unlimited
                        ClockPM- Surprise- LLActRep+ BwNot+ ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna+ CRSVisible+
                RootCap: CRSVisible+
                RootSta: PME ReqID 0000, PMEStatus- PMEPending-
                DevCap2: Completion Timeout: Range B, TimeoutDis+, LTR-, OBFF Not Supported ARIFwd-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled ARIFwd-
                LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
                         EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
        Capabilities: [100 v1] Device Serial Number 00-00-00-00-00-00-00-00
        Capabilities: [10c v1] Virtual Channel
                Caps:   LPEVC=0 RefClk=100ns PATEntryBits=1
                Arb:    Fixed- WRR32- WRR64- WRR128-
                Ctrl:   ArbSelect=Fixed
                Status: InProgress-
                VC0:    Caps:   PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
                        Arb:    Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
                        Ctrl:   Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
                        Status: NegoPending- InProgress-
        Capabilities: [128 v1] Vendor Specific Information: ID=1234 Rev=1 Len=018 <?>
        Kernel driver in use: pcieport
        Kernel modules: shpchp

root@zcu102:~#
```
It seems that the config of zcu102 is fine. But I'm still so confused about the flow of pcie config on zynqmp platform.

jberaud, 2020-10-15:
>Yes，I just want to try to connect other xilinx fpga device through pcie on my 102 board.

My ignorance, I didn't know that this was even possible, though it makes perfect sense. Sorry I don't have an answer to your crash.

hmaarrfk, 2020-10-15:
It is my understanding that xilinx didn't consider the arm endienness in their XDMA driver.

I had found a driver for XDMA that did alow this. 

It was from a small team and hosted on gitlab. They had claimed to improve on performance too. So hopefully you can find it.

The Intel x86 and AMD64 / x86-64 series of processors use the little-endian format.
While the ARM is big-endian.

jberaud, 2020-10-15:
>While the ARM is big-endian.

Arm is little endian most of the time.

hmaarrfk, 2020-10-15:
Ah you are right. Thats why i'm always confused.

I think the project was called `XDMA_v2`
https://github.com/bernardocarvalho/XDMA_v2
https://gitlab.com/WZab/Artix-DMA1

https://forums.xilinx.com/t5/PCIe-and-CPM/C2H-Streaming-XDMA-Linux-Driver-Broken/td-p/833977

Vae1997, 2020-10-29:
Hello，I'm back after verifying.
The first thing what I just did is that install nwl_pcie driver on my zcu102 board.
After do that，every time when I boot my sys，“nwl_pcie:Link is Down” can be found from series port. 
And when I insert k7 board in the pcie slot，it shown “Link is Up”.
On the other hand,  device ID and vender also can be found through "lspci".
Finally, I tried to recompile xdma src on zcu102 from xlnx-linux2018 kernel src code.
I try to fix some “implicit-function-declaration” error in order to finish process.(Just copy the func declaration)
When I try to exec insmod xdma.ko again, memory Oops error appear.
==================================================
I change my mind，just try to install other module which located in /lib/module/xxx.ko
To my surprise, none of the ko modules can be installed correctly.
One Oops is NULL pointer and the other Oops is read-only memory.
I realized that I cannot directly compile the module locally, but must go through Petalinux, because the previous zocl driver was also compiled and loaded through petalinux.
So I found UG1144 and found how to add custom modules through petalinux in Ch7.
In order to verify my guess, I found the simplest module code helloworld on the Internet.
I first compiled and generated hello.ko locally. When insmod was executed, Oops read-only appeared.
Then, I refer to the instructions of 1144, create a custom module, and modify the generated hello.c, and finally generate hello.ko in the /lib/module directory of rootfs
Finally, when I copy hello.ko and execute insmod, the module is loaded correctly!
===================================================
I think I should try to compile the xdma driver in the petalinux environment by the above method, but I don’t know if you have tried this before?
In addition, I don't know if I need to modify the dts file when compiling xdma driver in petalinux.
Finally, I noticed that the xdma driver is composed of multiple files. Therefore, in the petalinux environment, I should modify the default Makefile to ensure that the xdma can be compiled successfully.
If all goes well, the new xdma driver should be successfully installed on the zcu102, and the pcie-dma has been successfully configured for the k7 board, so I should be able to execute the test script for simple DMA data transfer.

lesjokolat, 2020-10-29:
Try kernel at least 4.19. Even better 5.3.x Other than that you will need to use older version of driver.

Vae1997, 2020-10-30:
> @jberaud
> Hello，I'm back after verifying.
> The first thing what I just did is that install nwl_pcie driver on my zcu102 board.
> After do that，every time when I boot my sys，“nwl_pcie:Link is Down” can be found from series port.
> And when I insert k7 board in the pcie slot，it shown “Link is Up”.
> On the other hand, device ID and vender also can be found through "lspci".
> Finally, I tried to recompile xdma src on zcu102 from xlnx-linux2018 kernel src code.
> I try to fix some “implicit-function-declaration” error in order to finish process.(Just copy the func declaration)
> When I try to exec insmod xdma.ko again, memory Oops error appear.
> 
> I change my mind，just try to install other module which located in /lib/module/xxx.ko
> To my surprise, none of the ko modules can be installed correctly.
> One Oops is NULL pointer and the other Oops is read-only memory.
> I realized that I cannot directly compile the module locally, but must go through Petalinux, because the previous zocl driver was also compiled and loaded through petalinux.
> So I found UG1144 and found how to add custom modules through petalinux in Ch7.
> In order to verify my guess, I found the simplest module code helloworld on the Internet.
> I first compiled and generated hello.ko locally. When insmod was executed, Oops read-only appeared.
> Then, I refer to the instructions of 1144, create a custom module, and modify the generated hello.c, and finally generate hello.ko in the /lib/module directory of rootfs
> Finally, when I copy hello.ko and execute insmod, the module is loaded correctly!
> 
> I think I should try to compile the xdma driver in the petalinux environment by the above method, but I don’t know if you have tried this before?
> In addition, I don't know if I need to modify the dts file when compiling xdma driver in petalinux.
> Finally, I noticed that the xdma driver is composed of multiple files. Therefore, in the petalinux environment, I should modify the default Makefile to ensure that the xdma can be compiled successfully.
> If all goes well, the new xdma driver should be successfully installed on the zcu102, and the pcie-dma has been successfully configured for the k7 board, so I should be able to execute the test script for simple DMA data transfer.

closed 2020-10-30 by Vae1997

xin-tubs, 2021-11-09:
> @jberaud
> Hello，I'm back after verifying.
> The first thing what I just did is that install nwl_pcie driver on my zcu102 board.
> After do that，every time when I boot my sys，“nwl_pcie:Link is Down” can [...]

Hi, 
Did you successed? I have tried to add driver to kernel of petalinux-system and it's work, but wenn I transfer data to Endpoint. It's failed and shows some message like this:
[image]
Did you meet the same problem?