Xilinx/dma_ip_drivers #86: lspci command has detected the device. But where I run load_driver.sh, report there is no xdma in /proc/devices dir

MarchSakura, 2020-10-28:
- The system is Ubuntu 16.04.
- The fpga is kintex ultrascale+ KU115.
- Using xshell remote link to the workstation.

when using lspci, I can detect the device.

```shell
$ lspci | grep Xi
04:00.0 Serial controller: Xilinx Corporation Device 4b88
04:00.1 Serial controller: Xilinx Corporation Device 4b87
```

But I sudo execute load_driver.sh, It reported did not recoginise the device.

```shell
xdma                   69632  0
Loading driver...
Error: The Kernel module installed correctly, but no devices were recognized.
 FAILED
```

It caused by there are xdma in /proc/devices dir.

I have reboot the machine, but it did not work. 

Some post says that is caused by bios setting. But until I get what to do to set the bios well, I will not rush in the lab.

MarchSakura, 2020-10-28:
The ubuntu kernel version is 4.15.0-107-generic.

lesjokolat, 2020-10-28:
Try newer Kernel.
 4.19.x
or 5.x(I ve used up to 5.3.x)

If you still want to use this kernel then may have to use older XDMA driver.

closed 2020-10-29 by MarchSakura

luckymaji, 2021-01-13:
maybe your FPGA version didn't load successfully