Xilinx/dma_ip_drivers #88: XDMA: Removal of cyclic transfer routines in libxdma

jberaud, 2020-11-09:
The cyclic transfer routines that used to be called by the read functions in the XDMA driver have been broken for a while and have been removed recently.

Those routines are not used by the standard XDMA driver but they can be very useful for some customers (like us).
In the case where the XDMA peripheral is used as a network-like interface, it is necessary to do some parsing inside the driver when receiving data on the DMA link, in order to distribute data to potentially multiple users.
In the current way the driver works, userland pages are mapped in the kernel when doing a C2H transfer, which can be more efficient than doing cyclic transfers when used in a single-user context. In the case of multiple users expecting different data, it is different. The cyclic transfer mode is the most efficient and makes it possible to continuously receive data while distributing it.

I can send a PR restoring the cyclic mode and potentially add some use cases and tests for that if people are interested.

BeauGesteMark, 2020-11-09:
Julien has explained why we use cyclic inbound mode.

I do not know our longevity with Xilinx’s XDMA driver. Their QDMA driver seems to have mostly replaced it. Will need SoC configurations have different PCIe that require a different driver? I do not know their direction.

I would suggest putting out a message to see the general XDMA-community interest in the cyclic reading – with some discussion of why we use it to avoid the “just give it up” comments. With support, we probably should offer a PR.

Mark H Bramhall, Director Host Software Engineering, SAS Fermat
+1 781 956 9598 or +33 7 87 10 48 81
mark@sasfermat.com

mpb27, 2020-11-21:
Hi @jberaud,

I need some sort of cyclic buffer to store multiple S2MM transactions of varying length.  Do the routines that you have account for AXI-Stream End-of-Packet (TLAST) creating a shorter than requested transfer?

The newest driver version (2c01de2211bcf452e90cb65b2ee49fcaafda375c) added the EOP, but it's still buggy.  I also tried using the new async IO mode to queue multiple async reads from user space, essentially creating a pool of reads, but that also does not work with the EOP.

My final option is to ditch XDMA, use the AXI-MM-PCIe core, wire in a AXI-DMA core in direct register mode, and reuse my driver from 4 years ago (`xilinx-dma-dr`) with a CMA pool after modifying it to find the PCIe device rather than using the device-tree.

jberaud, 2020-11-23:
>I need some sort of cyclic buffer to store multiple S2MM transactions of varying length. Do the routines that you have account for AXI-Stream End-of-Packet (TLAST) creating a shorter than requested transfer?

I only took the 2019 version and fixed the different things that had broken cyclic transfers. So I guess that the answer is no.

mpb27, 2020-11-24:
Right now I'm leaning toward a rewrite that would borrow heavily from libxdma only while simplifying it:

1. No polling mode.
2. No performance stuff. 
3. Remove splitting descriptors that have a transfer size larger than 256MB.  This can be enforced at a higher level.
4. Remove schedule_work and do most of the work in the interrupt or a tasklet.
5. I'm torn about cyclic mode, because in my application I can create a pool of transfers rather than using a cyclic mode.  The pool of transfers can be backed by user space memory using either ioctl or async reads.

jberaud, 2020-11-27:
1. We are not using polling mode.
2. Performance counters could be added using another form later.
3. I am not aware of this, I'll take a look at how it's handled.
4. Tasklets are going to be removed from the kernel at some point so I don't think it makes sense to use them. Doing work in the interrupt routine makes sense provided it is threaded, which I think is the default on recent kernels but not on older ones. I have to dig a bit more on what the benefits vs. drawbacks are as compared to using a workqueue.
5. I need cyclic mode to do some in-kernel parsing of incoming packets. It is the case of any type of multi-user application.

I think it would also make sense to have a test suite that is a bit more complete, plus complying to the kernel's checkpatch.pl and possibly adding coverity (or other tools) scan. In general, I think it only makes sense to rewrite the driver if we manage to build a community around it, else it will just be another fork that will die at some point.

Note: I am only interested in xdma, and I think that the drivers should be split anyway.

mpb27, 2020-11-27:
Any chance you can provide more details on how you use the cyclic mode?  I understand the requirement for in-kernel parsing, but don't quite understand why cyclic mode would be the only way to accomplish it.  Thanks.

jberaud, 2020-11-27:
>Any chance you can provide more details on how you use the cyclic mode? I understand the requirement for in-kernel parsing, but don't quite understand why cyclic mode would be the only way to accomplish it. Thanks.

When you want to receive data, with the non-cyclic mode that is currently in place, the user allocates memory, then passes the pages to the driver and the driver maps those pages in the DMA space so that the PCIe device can write data to those pages. It works well when you have a single user because it avoids having to do anything in the kernel.
It means that whatever data is received next is going to be written in those pages.

On the other hand, if you have what the cyclic mode used to do, you have a ring buffer of allocated kernel pages that are mapped in the DMA space, so that the PCIe device writes data to them. When new data is received in those pages, the driver can read the data, and depending on the content, determine that the said transfer is destined to one user or another, then copy the data from the kernel pages to the user pages.

mpb27, 2020-11-27:
I agree you have to use non-userspace memory, probably CMA (`dma_alloc_coherent`), and then inspect and copy to user space for each user.  But, at first glance, `xdma_xfer_submit` supports submitting a premapped `sg_table` which could be backed by CMA (just pass `dma_mapped=1`).  Ofcourse `cdev_sgdma` would have to be heavily modified to do all this, or a new driver would need to be written using libxdma.

I did something similar in a DMA driver for Zynq-7000 series and backed it by a circular buffer of transactions rather than a circular buffer of descriptors:
https://github.com/mpb27/xilinx-dma/blob/master/axis-reader/axis_reader.c

The DMA and the character device driver were split into two seperate modules.

jberaud, 2020-11-27:
>probably CMA

It doesn't have to be physically contiguous. It just needs to be mapped in the kernel space at some point in order to read the data and parse it.

>I did something similar in a DMA driver for Zynq-7000 series and backed it by a circular buffer of transactions rather than a circular buffer of descriptors:
https://github.com/mpb27/xilinx-dma/blob/master/axis-reader/axis_reader.c

I'll take some time to read the driver but using CMA (dma_alloc_coherent) for buffers. Even for descriptors themselves, I am not 100% sure they really need to be contiguous as long as adjacent descriptors are. It simplifies some computations if they are though.

captainliuy, 2021-08-14:
> The cyclic transfer routines that used to be called by the read functions in the XDMA driver have been broken for a while and have been removed recently.
>
> Those routines are not used by the standard [...]

captainliuy, 2021-08-14:
Hi @jberaud, 
Can you send the PR to restore the cyclic mode and potentially add some use cases and tests. That's usefull for me, thank you.

jberaud, 2021-08-16:
@captainliuy Hi. I have gone for the option of rewriting the whole driver. I plan to release it publicly soon. Even more if someone is interested. I still have the old code restoring the cyclic but it would take some time to put together a PR and I am afraid I don't have much right now.