Xilinx/dma_ip_drivers #95: BUG: DebugFS locking mechanism

mclayton1000, 2020-11-30:
Hi @ ,

enabling the **mutex debugging** mechanism on Linux we found an **issue in libqdma
DebugFS usage**.

In particular the problem is that the function **dbgfs_queue_init()**
(file qdma_debugfs_queue.c) **takes a spinlock** (descq->xdev->qidx_lock)
as soon as it is called and **then calls several DebugFS APIs**.

**That APIs internally take an RW lock: it is forbidden to call these APIs
with a spinlock taken. Really bad things could happen.**

**For example: debugfs_create_dir() internally takes an RW semaphore.
It is forbidden to take a possible sleeping lock in atomic context.**

The **locking mechanism** of the DebugFS should be **reviewed**.

Thank you, best regards.

sujathabanoth-xlnx, 2021-01-29:
verifying issue at our end. Will get back with details soon

mclayton1000, 2021-01-29:
Hi @sujathabanoth-xlnx ,

thank you for the help and support. Looking forward to hearing from you.

Best regards.