Xilinx/dma_ip_drivers #96: xdma0_user lseek fails

KOVI89alipes, 2020-11-30:
Hi,

I have the following setup

- Dell workstation
- Alveo U50 card
- XDMA IP Core 4.1 (reference design, configured through JTAG cable)
- Ubuntu 18.04
- Xilinx DMA IP drivers from 1fe2bd0235cf97d326222e8d8abcae1a5f2552f3

The problem - can't change the offset for  the `xdma0_user` device. I tried to open this device in 3 different modes from C++ - as a stream, file and char device, all 3 ways are failing. All of 3 ways work fine for `xdma0_[h2c][c2]_[0..3]` devices.

See the full error reproducing code here
https://paste.ofcode.org/Q5iP3cVfzH9mevswKLXUXm

Errors from stdout for `xdma0_user`.
```
STREAM:	Seekg Before -1, after  -1
CHAR:  	Leek return code  0xFFFFFFFFFFFFFFFF (d-1) ERROR 
POSIX: 	Fseek return code  -1 
```

Two aspects of this problem

1. When I call it for 
```
    FILE *pFile = nullptr;
    pFile = fopen( dev.c_str(), "w+");

    if (pFile){
        fprintf(stdout, "POSIX: \t");
        auto rc  =  fseek ( pFile , offset , SEEK_CUR );
        fprintf(stdout, "Fseek return code  %d \n",rc);   
```

 it leads to the following error in kernel log

```
[112106.318541] xdma:char_ctrl_ioctl: cmd 0x5401, xdev 0x00000000753b09a1, pdev 0x00000000ad1dd365.
[112106.318544] xdma:char_ctrl_ioctl: cmd 21505, bad magic 0x54/0x78
```
From 
https://github.com/Xilinx/dma_ip_drivers/blob/1fe2bd0235cf97d326222e8d8abcae1a5f2552f3/XDMA/linux-kernel/xdma/cdev_ctrl.c#L146

2. When I open `xdma0_user` device as a stream, I can't read `.tellg()` value
```
        std::fstream is ( dev );
        if (is) {
            fprintf(stdout, "STREAM:\t");
            int length_b = is.tellg(); // -1 here for xdma0_user
```

http://www.cplusplus.com/reference/istream/istream/tellg/
So something is wrong even before offset changing

JamieLuo, 2022-04-22:
The xdma0_user device does not implement the lseek, it was using the mmap to access the bar space.

Prandr, 2024-11-08:
I've had the same problem and I created https://github.com/Xilinx/dma_ip_drivers/pull/299, which adds lseek to all interfaces. Please have a look.

I would like to add that fstream with `fseek` shouldn't be used with xdma, only `lseek` system function.

Prandr, 2024-11-11:
The answer above are not quite correct, because it is not the only and IMO not the best way to access  xdmaN_user.  Upstream XDMA already implements read/write functions, which can be utilised with Linux POSIX [pread/pwrite system functions](https://github.com/mwrnd/notes/tree/main/XDMA_Communication#m_axi_lite).  

I've had the same problem with lseek and I created https://github.com/Xilinx/dma_ip_drivers/pull/299, which adds lseek to all interfaces. Please have a look and check.

I would like to add that `<stdio.h>` with `fseek` shouldn't be used with XDMA as it has those side effects with `ioctl`. Only pure Linux POSIX  file operation functions (including `lseek`) must be used.