Xilinx/dma_ip_drivers #98: git clone this repo failed on Linux platfom

byhongda, 2020-12-22:
On ubuntu18.04 (or any other linux systems supporting symbol link, I think), do a plain git clone from github:
```bash
$ git clone git@github.com:Xilinx/dma_ip_drivers.git
Cloning into 'dma_ip_drivers'...
remote: Enumerating objects: 327, done.
remote: Counting objects: 100% (327/327), done.
remote: Compressing objects: 100% (294/294), done.
remote: Total 2730 (delta 138), reused 91 (delta 26), pack-reused 2403
Receiving objects: 100% (2730/2730), 17.50 MiB | 319.00 KiB/s, done.
Resolving deltas: 100% (1495/1495), done.
error: unable to create symlink QDMA/linux-kernel/README: No such file or directory
fatal: unable to checkout working tree
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry the checkout with 'git checkout -f HEAD'
```

The file `QDMA/linux-kernel/README` is a abnorml link, which is not compatible with linux filesystem(ext4, etc.).
By the way, a workaround is as follows:
```
$ cd dma_ip_drivers
$ git config core.symlinks false
$ git checkout -f HEAD
```

hmaarrfk, 2020-12-22:
This is because the link is broken.

https://github.com/Xilinx/dma_ip_drivers/pull/74

sujathabanoth-xlnx, 2021-01-29:
This is environment issue and fixed

closed 2021-01-29 by sujathabanoth-xlnx