Xilinx/dma_ip_drivers #102: xdma:engine_status_dump: SG engine 0-H2C0-ST status: 0x00000001: BUSY

luckymaji, 2021-01-12:
hi，i have a problem with nvidia driver and xdma driver using together，has anyone ever had this problem？
when i only use xdma driver, every thins is ok，after i instsalled the nvidia GPU and its dirver, the GPU uses successfully, but the xdma can only send about 100 frame msg, and then stopped, the syslog tells the timeout and busy info

env：
Linux caros-Default-string 5.10.0-051000-generic #202012132330 SMP Sun Dec 13 23:33:36 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

 lspci -v -s 86:00.0
86:00.0 VGA compatible controller: NVIDIA Corporation GM107GL [Quadro K1200] (rev a2) (prog-if 00 [VGA controller])
	Subsystem: NVIDIA Corporation GM107GL [Quadro K1200]
	Flags: bus master, fast devsel, latency 0, IRQ 69, NUMA node 1
	Memory at df000000 (32-bit, non-prefetchable) [size=16M]
	Memory at 105fe0000000 (64-bit, prefetchable) [size=256M]
	Memory at 105ff0000000 (64-bit, prefetchable) [size=32M]
	I/O ports at d000 [size=128]
	[virtual] Expansion ROM at 000c0000 [disabled] [size=128K]
	Capabilities: <access denied>
	Kernel driver in use: nvidia
	Kernel modules: nvidiafb, nouveau, nvidia_drm, nvidia

lspci -v -s 5e:00.0
5e:00.0 Serial controller: Xilinx Corporation Device 9038 (prog-if 01 [16450])
	Subsystem: Xilinx Corporation Device 0007
	Flags: bus master, fast devsel, latency 0, IRQ 63, NUMA node 0
	Memory at c5d00000 (32-bit, non-prefetchable) [size=1M]
	Memory at c5e00000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: xdma
	Kernel modules: xdma

/dhclient" pid=955 comm="apparmor_parser"
[   22.073941] igb 0000:04:00.0 enp4s0: igb: enp4s0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   22.181655] IPv6: ADDRCONF(NETDEV_CHANGE): enp4s0: link becomes ready
[   26.855014] resource sanity check: requesting [mem 0x000c0000-0x000fffff], which spans more than PCI Bus 0000:00 [mem 0x000c4000-0x000c7fff window]
[   26.855248] caller os_map_kernel_space.part.14+0x78/0xa0 [nvidia] mapping multiple BARs
[  101.323775] **xdma:xdma_xfer_submit: xfer 0x00000000b2482d87,1304, s 0x1 timed out, ep 0x518.**
[  101.323782] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x00000000d7d99944) = 0x1fc08006 (id).
[  101.323786] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x0000000077090af4) = 0x00000001 (status).
[  101.323790] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x0000000029cf1a5c) = 0x00f83e1f (control)
[  101.323794] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x00000000ba8ed98f) = 0x6ced0ce0 (first_desc_lo)
[  101.323797] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x00000000886a2894) = 0x00000000 (first_desc_hi)
[  101.323801] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x00000000c1dba387) = 0x00000000 (first_desc_adjacent).
[  101.323804] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x00000000ad72189b) = 0x00000000 (completed_desc_count).
[  101.323807] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x00000000c2b642e2) = 0x00f83e1e (interrupt_enable_mask)
[  101.323811] **xdma:engine_status_dump: SG engine 0-H2C0-ST status: 0x00000001: BUSY**
[  101.323815] xdma:transfer_abort: abort transfer 0x00000000b2482d87, desc 1, engine desc queued 0.
[  111.562085] xdma:xdma_xfer_submit: xfer 0x00000000b2482d87,1304, s 0x1 timed out, ep 0x518.
[  111.562091] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x00000000d7d99944) = 0x1fc08006 (id).
[  111.562095] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x0000000077090af4) = 0x00000001 (status).
[  111.562098] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x0000000029cf1a5c) = 0x00f83e1f (control)
[  111.562102] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x00000000ba8ed98f) = 0x6ced0d00 (first_desc_lo)
[  111.562105] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x00000000886a2894) = 0x00000000 (first_desc_hi)
[  111.562109] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x00000000c1dba387) = 0x00000000 (first_desc_adjacent).
[  111.562112] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x00000000ad72189b) = 0x00000000 (completed_desc_count).
[  111.562115] xdma:engine_reg_dump: 0-H2C0-ST: ioread32(0x00000000c2b642e2) = 0x00f83e1e (interrupt_enable_mask)
[  111.562119] xdma:engine_status_dump: SG engine 0-H2C0-ST status: 0x00000001: BUSY

luckymaji, 2021-01-13:
i'v fixed this problem

closed 2021-01-13 by luckymaji

Ringhu, 2021-03-07:
hi, @luckymaji, I've got the same issue: SG engine 0-H2C0-ST status: 0x00000001: BUSY.  Could you please share how you handle it?

makslevental, 2022-02-04:
@luckymaji @Ringhu any advice on how to fix this?

MADMechaniculus, 2025-02-13:
> i'v fixed this problem

How? I have the same error. And most interesting that, than on another motherboard (with the same OS, FPGA and firmware on it) that's work normally. Kernel module of XDMA driver same yet.