Xilinx/dma_ip_drivers #108: Does QDMA in DPDK have multi-process support ?

ssc1010, 2021-01-23:
I active process A with `--proc-type=primary`. Then active process B with `--proc-type=secondary` and do dma write in process B, but run into segmentation fault as executing `rte_eth_tx_burst`.  Apart from launching the process with different arguments, is there anything else I need to pay attention to ? Or does it actually support multi-process at all ?

hmaarrfk, 2022-08-22:
Hello,

My name is Mark Harfouche. I am not affiliated with Xilinx in any way. Over the
years of using QDMA, I've been wanted better community organization.

I've created a fork of dma_ip_drivers which I intend to maintain and work with the
community at large to improve.

The fork can be found https://github.com/hmaarrfk/dma_ip_drivers

For now, I am stating the main goals of the repository in
https://github.com/hmaarrfk/dma_ip_drivers/issues/2

If you are interested in working together, feel free to open an issue or PR to
my fork.

Best,

Mark

digaru19, 2023-10-29:
@ssc1010 Did you have any luck with this issue: running multiple processes using QDMA and DPDK? I'm also running into a segmentation fault while executing the `rte_eth_rx_burst` call.

shuihan0555, 2023-12-10:
> @ssc1010 Did you have any luck with this issue: running multiple processes using QDMA and DPDK? I'm also running into a segmentation fault while executing the `rte_eth_rx_burst` call. Thanks!

@digaru19 Did you have done? I have the same segmentation fault while execting the ``rte_eth_rx_burst` call. Some pointer is 0x00

digaru19, 2023-12-19:
@shuihan0555  I was able to resolve this using this fix in [qdma_devops.c](https://github.com/digaru19/dma_ip_drivers/commit/90483bb2b6626775420cd4ec800731ef7028aea9). I am able to receive packets in both primary and secondary processes now. Although, this is an experimental fix and I'm not sure if this negatively affects any other part of the driver.