Xilinx/dma_ip_drivers #111: ERRORs when install xdma driver on Ubuntu

Lumpyzc, 2021-02-04:
Hi,everyone!
I have got a problem when I attempt to install xdma driver on my Ubuntu server.
Please give me some advice!!!!
My linux system is Ubuntu 18.04
kernel is 4.15.0-132-generic

firstly I download the .zip file from here and then install the driver like what the usage says;
unfortunately ,when I “cd xdma” and “make install” I meet the first problem as below:

_At main.c:160:

SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
DEPMOD 4.15.0-132-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-132-generic'_
then I use the method #39 gives
cd /usr/src/linux-headers-$(uname -r)/certs
sudo openssl req -new -x509 -newkey rsa:2048 -keyout signing_key.pem -outform DER -out signing_key.x509 -nodes -subj "/CN=Owner/"

then I got that like below shows:

Can't load /root/.rnd into RNG
140138494010560:error:2406F079:random number generator:RAND_load_file:Cannot open file:../crypto/rand/randfile.c:88:Filename=/root/.rnd
Generating a RSA private key
.......................................+++++
..................+++++
writing new private key to 'signing_key.pem'
I DONT know what that means and then make install again,the SSL problem gone missing !

then I do "cd tools" and "make" the result is :

cc -c -std=c99 -o reg_rw.o reg_rw.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -o reg_rw reg_rw.o
cc -c -std=c99 -o dma_to_device.o dma_to_device.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
In file included from /usr/include/assert.h:35,
from dma_to_device.c:13:
/usr/include/features.h:184:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
184 | # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
| ^~~~~~~
cc -lrt -o dma_to_device dma_to_device.o -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -c -std=c99 -o dma_from_device.o dma_from_device.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
In file included from /usr/include/assert.h:35,
from dma_from_device.c:13:
/usr/include/features.h:184:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
184 | # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
| ^~~~~~~
cc -lrt -o dma_from_device dma_from_device.o -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -c -std=c99 -o performance.o performance.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -o performance performance.o -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE

then I began to "modprobe xdma" another problem appeared as shown below:

modprobe: FATAL: Module xdma not found in directory /lib/modules/4.15.0-132-generic

I also tried to "cd tests" and "./load_driver.sh" BUT it went wrong:

Loading xdma driver...insmod: ERROR: could not insert module ../xdma/xdma.ko: Invalid module format
Error: Kernel module did not load properly.
FAILED

Please give me some advice !!!

BEST REGARDS!!

Lumpyzc, 2021-02-04:
finally ,I GET THE cause！
MY ubuntu version mismatch my GCC?
I tried lowed my GCC to 7, then complie the .ko file,
then modprobe xdma
it works!

closed 2021-02-04 by Lumpyzc

Gothel17, 2023-04-18:
Hello,
I get the same error. Can you eplain how you lowered your gcc version ? And after that did you uninstall and install again the kernel driver ? Because in the README it's not specified to run the .ko file as you did.
Thank you in advance!