Xilinx/dma_ip_drivers #113: Kernel Bug when using async read

yaxn, 2021-02-26:
Hello, when i'm try to read from Memory-Mapped XDMA device using Boost::ASIO

```c++
#include <boost/asio.hpp>
#include <boost/asio/posix/stream_descriptor.hpp>

void main()
{
    int dev = ::open("/dev/xdma0_c2h_0", O_RDONLY);
    boost::asio::io_service io_svc;
    boost::asio::posix::stream_descriptor iodevice(io_svc, dev);
 
    char buffer[1024];
    size_t r = iodevice.read_some(boost::asio::buffer(buffer)); // error here
}
```
I got kernel bug, see two dmesg log with enabled debug in xdma module:
- [dmesg_log_3_dbg.txt](https://github.com/Xilinx/dma_ip_drivers/files/6050046/dmesg_log_3_dbg.txt)
- [dmesg_log_4_dbg.txt](https://github.com/Xilinx/dma_ip_drivers/files/6050047/dmesg_log_4_dbg.txt)

Last function in call trace:
```log
[  592.680375]  async_io_handler+0x139/0x180 [xdma]
```
Kernel Version 5.4.0
Driver commit 747d13f

codypiersall, 2022-03-10:
I ran into the same issue with a minimal C program that just calls `readv()`.

```c
#include <assert.h>
#include <fcntl.h>
#include <unistd.h>
#include <stdlib.h>
#include <stdio.h>
#include <sys/uio.h>

int main() {
    int f = open("/dev/xdma0_c2h_0", O_RDONLY);
    assert(f > 0);
    int n_iovec = 2;
    int sz_iovec = 1024 * 1024;
    struct iovec vs[n_iovec];
    for (int i = 0; i < n_iovec; i++) {
        vs[i].iov_base = aligned_alloc(4096, sz_iovec);
        vs[i].iov_len = sz_iovec;
    }
    while (1) {
        int ret = readv(f, vs, n_iovec);
        printf("%d ", ret);
    }

    close(f);
}
```

hmaarrfk, 2022-03-11:
I've noticed this too. 

I've reported many bugs to xilinx regarding their driver. They haven't been super responsive on GitHub. 

They have a mandate to only respond to the forums. You need to post there.

codypiersall, 2022-03-23:
Thanks for the info.  I opened a ticket: https://support.xilinx.com/s/question/0D52E0000757EyXSAU/kernel-bug-in-xdma-driver-when-readv-system-call-is-used

eniv, 2024-08-22:
If anyone is interested, I posted more information about this issue [here](https://support.xilinx.com/s/question/0D52E0000757EyXSAU/kernel-bug-in-xdma-driver-when-readv-system-call-is-used?language=en_US)