Xilinx/dma_ip_drivers #115: Error: Kernel module did not load properly.

yuxisjtu, 2021-04-01:
Hello!
i want to install the driver  under ubuntu 18.04.5 and kernel 5.4.0-42-generic but failed
i have try [https://github.com/Xilinx/dma_ip_drivers/issues/39](url) and [https://github.com/Xilinx/dma_ip_drivers/pull/44](url) but still not work,it reports below.
(im sorry their may be a little language problem, i change it into english)

yux@yux-HP-Zhan-86-Pro-G2-MT-Business-PC:~/Desktop/dma_ip_drivers-master/XDMA/linuxernel$ cd xdma
yux@yux-HP-Zhan-86-Pro-G2-MT-Business-PC:~/Desktop/dma_ip_drivers-master/XDMA/linuxernel/xdma$ sudo make clean
[sudo] yux password： 
Makefile:10: XVC_FLAGS: .
make -C /lib/modules/5.4.0-42-generic/build M=/home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma clean
make[1]: Entering directory“/usr/src/linux-headers-5.4.0-42-generic”
/home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/Makefile:10: XVC_FLAGS: .
  CLEAN   /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/Module.symvers
make[1]:Leaving directory“/usr/src/linux-headers-5.4.0-42-generic”
yux@yux-HP-Zhan-86-Pro-G2-MT-Business-PC:~/Desktop/dma_ip_drivers-master/XDMA/linuxernel/xdma$ sudo make install
Makefile:10: XVC_FLAGS: .
make -C /lib/modules/5.4.0-42-generic/build M=/home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma modules
make[1]: Entering directory“/usr/src/linux-headers-5.4.0-42-generic”
/home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/Makefile:10: XVC_FLAGS: .
  CC [M]  /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/libxdma.o
  CC [M]  /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/xdma_cdev.o
  CC [M]  /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/cdev_ctrl.o
  CC [M]  /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/cdev_events.o
  CC [M]  /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/cdev_sgdma.o
  CC [M]  /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/cdev_xvc.o
  CC [M]  /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/cdev_bypass.o
  CC [M]  /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/xdma_mod.o
  CC [M]  /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/xdma_thread.o
  LD [M]  /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/xdma.o
/home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/Makefile:10: XVC_FLAGS: .
  Building modules, stage 2.
  MODPOST 1 modules
  CC [M]  /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/xdma.mod.o
  LD [M]  /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/xdma.ko
make[1]: Leaving directory“/usr/src/linux-headers-5.4.0-42-generic”
make -C /lib/modules/5.4.0-42-generic/build M=/home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma modules_install
make[1]: Entering directory“/usr/src/linux-headers-5.4.0-42-generic”
  INSTALL /home/yux/Desktop/dma_ip_drivers-master/XDMA/linux-kernel/xdma/xdma.ko
  DEPMOD  5.4.0-42-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory“/usr/src/linux-headers-5.4.0-42-generic”
depmod -a
yux@yux-HP-Zhan-86-Pro-G2-MT-Business-PC:~/Desktop/dma_ip_drivers-master/XDMA/linuxernel/xdma$ cd ../tools
yux@yux-HP-Zhan-86-Pro-G2-MT-Business-PC:~/Desktop/dma_ip_drivers-master/XDMA/linuxernel/tools$ make clean
rm -rf reg_rw *.o *.bin dma_to_device dma_from_device performance
yux@yux-HP-Zhan-86-Pro-G2-MT-Business-PC:~/Desktop/dma_ip_drivers-master/XDMA/linuxernel/tools$ make
cc -c -std=c99 -o reg_rw.o reg_rw.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -o reg_rw reg_rw.o
cc -c -std=c99 -o dma_to_device.o dma_to_device.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
In file included from /usr/include/assert.h:35:0,
                 from dma_to_device.c:13:
/usr/include/features.h:184:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
 # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
   ^~~~~~~
cc -lrt -o dma_to_device dma_to_device.o -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -c -std=c99 -o dma_from_device.o dma_from_device.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
In file included from /usr/include/assert.h:35:0,
                 from dma_from_device.c:13:
/usr/include/features.h:184:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
 # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
   ^~~~~~~
cc -lrt -o dma_from_device dma_from_device.o -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -c -std=c99 -o performance.o performance.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -o performance performance.o -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
yux@yux-HP-Zhan-86-Pro-G2-MT-Business-PC:~/Desktop/dma_ip_drivers-master/XDMA/linuxernel/tools$ cd ../tests
yux@yux-HP-Zhan-86-Pro-G2-MT-Business-PC:~/Desktop/dma_ip_drivers-master/XDMA/linuxernel/tests$ sudo ./load_driver.sh
Loading xdma driver...insmod: ERROR: could not insert module ../xdma/xdma.ko: Operation not permitted
Error: Kernel module did not load properly.
 FAILED

do you have any advise?thanks!

jberaud, 2021-04-01:
Can you please send the content of the dmesg ?

yuxisjtu, 2021-04-01:
> Can you please send the content of the dmesg ?

oh thank you, do you mean this?
yux@yux-HP-Zhan-86-Pro-G2-MT-Business-PC:~/Desktop/dma_ip_drivers-master/XDMA/linuxernel$ dmesg
[    0.000000] microcode: microcode updated early to revision 0xd6, date = 2020-04-27
[    0.000000] Linux version 5.4.0-42-generic (buildd@lgw01-amd64-023) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)) #46~18.04.1-Ubuntu SMP Fri Jul 10 07:21:24 UTC 2020 (Ubuntu 5.4.0-42.46~18.04.1-generic 5.4.44)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.4.0-42-generic root=UUID=b1686b24-9cfc-4dba-8de4-082fb0fd5e4b ro quiet splash vt.handoff=1
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000048fddfff] usable
[    0.000000] BIOS-e820: [mem 0x0000000048fde000-0x000000004943dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000004943e000-0x00000000494bffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000494c0000-0x000000004955dfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000004955e000-0x000000004bc0dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000004bc0e000-0x000000004bc0efff] usable
[    0.000000] BIOS-e820: [mem 0x000000004bc0f000-0x000000004f7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000002ae7fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x41539018-0x41549057] usable ==> usable
[    0.000000] e820: update [mem 0x41539018-0x41549057] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000041539017] usable
[    0.000000] reserve setup_data: [mem 0x0000000041539018-0x0000000041549057] usable
[    0.000000] reserve setup_data: [mem 0x0000000041549058-0x0000000048fddfff] usable
[    0.000000] reserve setup_data: [mem 0x0000000048fde000-0x000000004943dfff] reserved
[    0.000000] reserve setup_data: [mem 0x000000004943e000-0x00000000494bffff] ACPI data
[    0.000000] reserve setup_data: [mem 0x00000000494c0000-0x000000004955dfff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000004955e000-0x000000004bc0dfff] reserved
[    0.000000] reserve setup_data: [mem 0x000000004bc0e000-0x000000004bc0efff] usable
[    0.000000] reserve setup_data: [mem 0x000000004bc0f000-0x000000004f7fffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100000000-0x00000002ae7fffff] usable
[    0.000000] efi: EFI v2.60 by American Megatrends
[    0.000000] efi:  ACPI 2.0=0x4945c000  ACPI=0x4945c000  SMBIOS=0x4b9ae000  SMBIOS 3.0=0x4b9ad000  ESRT=0x473b1f98  MEMATTR=0x473b0018 
[    0.000000] secureboot: Secure boot enabled
[    0.000000] Kernel is locked down from EFI Secure Boot mode; see man kernel_lockdown.7
[    0.000000] SMBIOS 3.1.1 present.
[    0.000000] DMI: HP HP Zhan 86 Pro G2 MT Business PC/843C, BIOS F.13 10/11/2018
[    0.000000] tsc: Detected 3200.000 MHz processor
[    0.000431] tsc: Detected 3199.980 MHz TSC
[    0.000431] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[... 852 lines cut ...]
[   20.120651] audit: type=1400 audit(1617276218.645:7): apparmor="STATUS" operation="profile_load" profile="unconfined" name="man_filter" pid=764 comm="apparmor_parser"
[... cut ...]
[  897.011922] Lockdown: insmod: unsigned module loading is restricted; see man kernel_lockdown.7
[ 1008.230084] Lockdown: insmod: unsigned module loading is restricted; see man kernel_lockdown.7
[ 1127.052516] Lockdown: insmod: unsigned module loading is restricted; see man kernel_lockdown.7
[ 1477.740666] Lockdown: insmod: unsigned module loading is restricted; see man kernel_lockdown.7
[ 1615.231065] Lockdown: insmod: unsigned module loading is restricted; see man kernel_lockdown.7
[ 2394.200196] Lockdown: insmod: unsigned module loading is restricted; see man kernel_lockdown.7

jberaud, 2021-04-01:
>oh thank you, do you mean this?

Yes exactly. Thanks. So the problem is that you have secureboot and thus kernel lockdown enabled. It means that the kernel modules you load on your machine need to be signed. For upstream kernel modules, they are signed as part of the kernel generation. For out of tree drivers, you need to sign them using your own key.

You can find information about how to do that there:
https://ubuntu.com/blog/how-to-sign-things-for-secure-boot

yuxisjtu, 2021-04-01:
> Yes exactly. Thanks. So the problem is that you have secureboot and thus kernel lockdown enabled. It means that the kernel modules you load on your machine need to be signed. For upstream kernel [...]

you help me a lot! i will try and then leave a message if it succeeds or not.

yuxisjtu, 2021-04-01:
> Yes exactly. Thanks. So the problem is that you have secureboot and thus kernel lockdown enabled. It means that the kernel modules you load on your machine need to be signed. For upstream kernel [...]

ohhh！ you successfully solved my problem，thank you very much! Wish you have a good day！