Xilinx/dma_ip_drivers #117: Multithreaded access to different channel and different memory area fails

X-Ryl669, 2021-04-02:
## Description of the issue:
1. I've 2 threads (R & S).
2. R has opened `xdma0_c2h_0`
3. S has opened `xdma0_h2c_0`
3. R does `char buf[1024*1024]; pread(c2h_fd, buf, sizeof(buf), SOME_ADDR1);` in a loop
4. S does `char buf[20*1024]; pwrite(h2c_fd, buf, sizeof(buf), SOME_ADDR2);` in a loop
5. Most transfer will work but some will fail (timeout) with this in dmesg:
```
Event logged [IO_PAGE_FAULT device=0e:00.0 domain=0x001a address=0xfe809400 flags=0x0020]
xdma_xfer_submit: xfer 0x00000000352a938a,1048832, s 0x1 timed out, ep 0x300100.
engine_reg_dump: 0-C2H0-MM: ioread32(0x00000000389bf350) = 0x1fc10006 (id).
engine_reg_dump: 0-C2H0-MM: ioread32(0x00000000c5bcecb2) = 0x00000001 (status).
engine_reg_dump: 0-C2H0-MM: ioread32(0x000000002fc32109) = 0x00f83e1f (control)
engine_reg_dump: 0-C2H0-MM: ioread32(0x00000000fc4073fa) = 0xffeb0000 (first_desc_lo)
engine_reg_dump: 0-C2H0-MM: ioread32(0x000000007b883500) = 0x00000000 (first_desc_hi)
engine_reg_dump: 0-C2H0-MM: ioread32(0x00000000190cefd2) = 0x0000003f (first_desc_adjacent).
engine_reg_dump: 0-C2H0-MM: ioread32(0x0000000017def8cc) = 0x00000004 (completed_desc_count).
engine_reg_dump: 0-C2H0-MM: ioread32(0x000000005cc4976c) = 0x00f83e1e (interrupt_enable_mask)
engine_status_dump: SG engine 0-C2H0-MM status: 0x00000001: BUSY
transfer_abort: abort transfer 0x00000000352a938a, desc 257, engine desc queued 0.
```

## Remarks
If I add a mutex around these calls to prevent both thread to access the XDMA driver at the same time, it does not happen (but obviously with a much lower performance)

I'm not using the same memory buffer, I'm not using the same file descriptor, and I'm not using the same destination/src address in the FPGA here.

For me, it's seems that both h2c and c2h shares the same work queue in the driver, and servicing the work queue happens at a later step, so if the 2 transfers happens simultaneously, it's possible that the later interrupt happens before the work queue is serviced and the registers are read in `engine_service` (mixing the interrupt state processing and breaking the descriptor updating). 
Shouldn't all required registers be read in the `xdma_isr` function and saved in the `engine` object, instead of:

1. Delaying locking in `engine_service_work` (it seems it can have a race condition here, since another irq can trigger before this lock is taken)
2.  Thus, register reads in `engine_service` are also racy even if the lock is taken above

jberaud, 2021-04-02:
h2c and c2h share the same workqueue, which is the global kernel workqueue. That said, the global kernel workqueue is not a single thread, so it is possible that h2c and c2h engines which have different works (on the global workqueue), run in different threads.
I am not sure I understand what you mean about "the later interrupt happen before the workqueue is serviced". The workqueue is the global kernel workqueue, but different engines (c2h and h2c) have different works. Reading the interrupt status in the isr, disabling the irqs that have been read as requested, for them to be re-enabled later by the corresponding works does work in theory. If a c2h irq happens between and isr that has read a h2c request and scheduled the h2c work, it will trigger another xdma_isr, and vice-versa. I have quickly checked and writes to enable/disable interrupts are done using w1s and w1c which means that there cannot be race condition in reading/writing the interrupt enable register.

I am not saying that this actually works in practice, because it obviously doesn't on your side, and I am not familiar with MM transfers but it seems that this should work in theory, so I would look for the cause elsewhere or add more debug prints or traces to catch what is actually happening.

X-Ryl669, 2021-04-02:
In engine_service, there's a : `read_register(&engine->regs->completed_desc_count)` line. 
Let's consider this timeline:
1. H2C's is done processing some descriptor and triggers `xdma_isr`. Here `engine->regs->completed_desc_count` = 3 (for example)
2. work is posted on the work queue. 
3. ISR returns
4. C2H is also done processing and triggers `xdma_isr`. Here `engine->regs->completed_desc_count` = 6 (for example)
5. `engine_service` runs and starts processing the H2C's work item and read 6 in the register and not 3 as expected.
6. `engine_service` will run for the next work item (C2H) and read 6 as expected.

jberaud, 2021-04-02:
>engine_service runs and starts processing the H2C's work item and read 6 in the register and not 3 as expected.

The "engine" object is not a global variable. There is 1 engine for h2c and 1 for c2h.
Typically, if you have 1 h2c channel and 1 c2h channel, `engine->regs->completed-desc_count` is at a different address for each.
Please check function `engine_init` which initializes engine->regs for each engine, which is called by `probe_for_engine` which is called for each c2h channel and each h2c channel in `probe_engines`.

>I'm just trying to find out how the XDMA IP could get wrong descriptors (since it's obvious it's processing wrong descriptors: that causes the IO MMU's fault we see above, I have plenty of those when the bug happens)

Since you get an IO_PAGE_FAULT, there is likely something wrong with the descriptors indeed. Can you enable full debugging and try to see if you can still reproduce the issue. It is possible that it makes the issue disappear if it is a race condition but it is worth trying.

jberaud, 2021-04-02:
Other question, I have seen you have opened the same issue in aws-fpga. What version of the xdma driver are you using ?
The 2 versions differ a lot between here and aws-fpga.

X-Ryl669, 2021-04-02:
I've tried on both. Same results. I'm not using any feature that's not on available on aws-fpga (and their code is a lot lighter so it's easier to follow).

X-Ryl669, 2021-04-02:
> Can you enable full debugging and try to see if you can still reproduce the issue.
It's very verbose when I do this. The issue is reproducible but it requires **many** transfers (around 1000 to 3000) to trigger so enabling the log here is a pain. I'll try to do that on Tuesday and keep you informed.

X-Ryl669, 2021-04-13:
I'm not able to reproduce this with full debugging enabled (it's **WAY** too slow to happen). I had some failures:
<details>
<summary>Logs</summary>
```
[  378.390053] xdma:engine_start: ioread32(0x00000000a9f2a7e5) (dummy read flushes writes).
[  378.390055] xdma:engine_start_mode_config: iowrite32(0x00f83e1f to 0x00000000c7428b56) (control)
[  378.390056] xdma:__write_register: engine_start_mode_config: w reg 0x1bdd34a02604(0x00000000c7428b56), 0xf83e1f.
[  378.390059] xdma:engine_start_mode_config: ioread32(0x00000000a9f2a7e5) = 0x00000001 (dummy read flushes writes).
[  378.390062] xdma:engine_start: 0-C2H0-MM engine 0x00000000f5d9d73d now running
[  378.390063] xdma:transfer_queue: transfer=0x000000006aa24fce started 0-C2H0-MM engine with transfer 0x000000006aa24fce.
[  378.390064] xdma:transfer_queue: engine->running = 1
[  378.390844] xdma:xdma_isr: (irq=119, dev 0x000000000b49adfd) <<<< ISR.
[  378.390847] xdma:xdma_isr: ch_irq = 0x00000010
[  378.390848] xdma:__write_register: channel_interrupts_disable: w reg 0x2000(0x000000005476bcd8), 0x10.
[  378.390852] xdma:xdma_isr: user_irq = 0x00000000
[  378.390853] xdma:xdma_isr: schedule_work, 0-C2H0-MM.
[  378.390855] xdma:engine_service_work: engine_service() for 0-C2H0-MM engine 00000000f5d9d73d
[  378.390858] xdma:engine_service_shutdown: engine just went idle, resetting RUN_STOP.
[  378.390859] xdma:xdma_engine_stop: xdma_engine_stop(engine=00000000f5d9d73d)
[  378.390860] xdma:xdma_engine_stop: Stopping SG DMA 0-C2H0-MM engine; writing 0x00f83e1e to 0x00000000c7428b56.
[  378.390861] xdma:__write_register: xdma_engine_stop: w reg 0x1bdd34a02604(0x00000000c7428b56), 0xf83e1e.
[  378.390862] xdma:xdma_engine_stop: xdma_engine_stop(0-C2H0-MM) done
[  378.390865] xdma:engine_service: 0-C2H0-MM wb 0x0, desc_count 177, err 0, dequeued 0.
[  378.390866] xdma:engine_service: head of queue transfer 0x000000006aa24fce has 177 descriptors
[  378.390867] xdma:engine_service: Engine completed 177 desc, 177 not yet dequeued
[  378.390868] xdma:engine_service_final_transfer: engine 0-C2H0-MM completed transfer
[  378.390869] xdma:engine_service_final_transfer: Completed transfer ID = 0x000000006aa24fce
[  378.390870] xdma:engine_service_final_transfer: *pdesc_completed=177, transfer->desc_num=177
[  378.390871] xdma:engine_service_resume: no pending transfers, 0-C2H0-MM engine stays idle.
[  378.390874] xdma:__write_register: channel_interrupts_enable: w reg 0x2000(0x0000000073ec3465), 0x10.
[  378.390876] xdma:xdma_xfer_submit: transfer 000000006aa24fce, 724992, ep 0x200000 compl, +0.
[  378.390877] xdma:transfer_destroy: done freeing desc 0 1
[  378.390878] xdma:transfer_init: xfer= 000000006aa24fce transfer->desc_bus = 0xfff30000.
[  378.390879] xdma:transfer_build: sw desc 177/257: 0xfe8b1000, 0x1000, ep 0x2b1000.
[  378.390881] xdma:transfer_build: sw desc 178/257: 0xfe8b2000, 0x1000, ep 0x2b2000.
[  378.390883] xdma:transfer_build: sw desc 179/257: 0xfe8b3000, 0x1000, ep 0x2b3000.
[  378.390884] xdma:transfer_build: sw desc 180/257: 0xfe8b4000, 0x1000, ep 0x2b4000.
[  378.390885] xdma:transfer_build: sw desc 181/257: 0xfe8b5000, 0x1000, ep 0x2b5000.
[  378.390886] xdma:transfer_build: sw desc 182/257: 0xfe8b6000, 0x1000, ep 0x2b6000.
[  378.390886] xdma:transfer_build: sw desc 183/257: 0xfe8b7000, 0x1000, ep 0x2b7000.
[  378.390887] xdma:transfer_build: sw desc 184/257: 0xfe8b8000, 0x1000, ep 0x2b8000.
[  378.390888] xdma:transfer_build: sw desc 185/257: 0xfe8b9000, 0x1000, ep 0x2b9000.
[  378.390889] xdma:transfer_build: sw desc 186/257: 0xfe8ba000, 0x1000, ep 0x2ba000.
[  378.390890] xdma:transfer_build: sw desc 187/257: 0xfe8bb000, 0x1000, ep 0x2bb000.
[  378.390892] xdma:transfer_build: sw desc 188/257: 0xfe8bc000, 0x1000, ep 0x2bc000.
[  378.390893] xdma:transfer_build: sw desc 189/257: 0xfe8bd000, 0x1000, ep 0x2bd000.
[  378.390894] xdma:transfer_build: sw desc 190/257: 0xfe8be000, 0x1000, ep 0x2be000.
[  378.390895] xdma:transfer_build: sw desc 191/257: 0xfe8bf000, 0x1000, ep 0x2bf000.
[  378.390896] xdma:transfer_build: sw desc 192/257: 0xfe8c0000, 0x1000, ep 0x2c0000.
[  378.390897] xdma:transfer_build: sw desc 193/257: 0xfe8c1000, 0x1000, ep 0x2c1000.
[  378.390898] xdma:transfer_build: sw desc 194/257: 0xfe8c2000, 0x1000, ep 0x2c2000.
[  378.390899] xdma:transfer_build: sw desc 195/257: 0xfe8c3000, 0x1000, ep 0x2c3000.
[  378.390900] xdma:transfer_build: sw desc 196/257: 0xfe8c4000, 0x1000, ep 0x2c4000.
[  378.390902] xdma:transfer_build: sw desc 197/257: 0xfe8c5000, 0x1000, ep 0x2c5000.
[  378.390903] xdma:transfer_build: sw desc 198/257: 0xfe8c6000, 0x1000, ep 0x2c6000.
[  378.390904] xdma:transfer_build: sw desc 199/257: 0xfe8c7000, 0x1000, ep 0x2c7000.
[  378.390905] xdma:transfer_build: sw desc 200/257: 0xfe8c8000, 0x1000, ep 0x2c8000.
[  378.390906] xdma:transfer_build: sw desc 201/257: 0xfe8c9000, 0x1000, ep 0x2c9000.
[  378.390907] xdma:transfer_build: sw desc 202/257: 0xfe8ca000, 0x1000, ep 0x2ca000.
[  378.390908] xdma:transfer_build: sw desc 203/257: 0xfe8cb000, 0x1000, ep 0x2cb000.
[  378.390909] xdma:transfer_build: sw desc 204/257: 0xfe8cc000, 0x1000, ep 0x2cc000.
[  378.390911] xdma:transfer_build: sw desc 205/257: 0xfe8cd000, 0x1000, ep 0x2cd000.
[  378.390912] xdma:transfer_build: sw desc 206/257: 0xfe8ce000, 0x1000, ep 0x2ce000.
[  378.390913] xdma:transfer_build: sw desc 207/257: 0xfe8cf000, 0x1000, ep 0x2cf000.
[  378.390914] xdma:transfer_build: sw desc 208/257: 0xfe8d0000, 0x1000, ep 0x2d0000.
[... cut ...]
[  378.391411] xdma:engine_status_dump: SG engine 0-C2H0-MM status: 0x00000001: BUSY
[  378.391411] xdma:transfer_abort: abort transfer 0x000000006aa24fce, desc 80, engine desc queued 0.
[  378.391412] xdma:xdma_engine_stop: xdma_engine_stop(engine=00000000f5d9d73d)
[  378.391413] xdma:xdma_engine_stop: Stopping SG DMA 0-C2H0-MM engine; writing 0x00f83e1e to 0x00000000c7428b56.
[  378.391414] xdma:__write_register: xdma_engine_stop: w reg 0x1bdd34a02604(0x00000000c7428b56), 0xf83e1e.
[  378.391415] xdma:xdma_engine_stop: xdma_engine_stop(0-C2H0-MM) done
```
</details>

X-Ryl669, 2021-04-13:
I've disabled __LIBXDMA_DEBUG__ and only enabled dumping in the case of error (TRANSFER_FAILED or timeout) (in `xdma_xfer_completion` and `xdma_xfer_submit` and it does not reproduce either. As soon as I remove my change, it reproduces almost instantly.
<details>
<summary>Patch</summary>

```diff
diff --git a/XDMA/linux-kernel/xdma/Makefile b/XDMA/linux-kernel/xdma/Makefile
index 7c22681..e21cdfe 100644
--- a/XDMA/linux-kernel/xdma/Makefile
+++ b/XDMA/linux-kernel/xdma/Makefile
@@ -15,6 +15,7 @@ TARGET_MODULE:=opu
 
 EXTRA_CFLAGS := -I$(topdir)/include $(XVC_FLAGS)
 #EXTRA_CFLAGS += -D__LIBXDMA_DEBUG__
+EXTRA_CFLAGS += -D__LIBXDMA_SMALLDBG__
 #EXTRA_CFLAGS += -DINTERNAL_TESTING
 
 ifneq ($(KERNELRELEASE),)

diff --git a/XDMA/linux-kernel/xdma/libxdma.c b/XDMA/linux-kernel/xdma/libxdma.c
index e163bc5..2c41058 100644
--- a/XDMA/linux-kernel/xdma/libxdma.c
+++ b/XDMA/linux-kernel/xdma/libxdma.c
@@ -181,7 +185,7 @@ static inline int debug_check_dev_hndl(const char *fname, struct pci_dev *pdev,
 	return 0;
 }
 
-#ifdef __LIBXDMA_DEBUG__
+#if defined(__LIBXDMA_DEBUG__)
 /* SECTION: Function definitions */
 inline void __write_register(const char *fn, u32 value, void *iomem,
 			     unsigned long off)
@@ -2198,7 +2202,7 @@ static int irq_setup(struct xdma_dev *xdev, struct pci_dev *pdev)
 	return irq_legacy_setup(xdev, pdev);
 }
 
-#ifdef __LIBXDMA_DEBUG__
+#if defined(__LIBXDMA_DEBUG__) || defined(__LIBXDMA_SMALLDBG__)
 static void dump_desc(struct xdma_desc *desc_virt)
 {
 	int j;
@@ -3002,7 +3007,6 @@ static int transfer_init(struct xdma_engine *engine,
 	transfer_build(engine, req, xfer, desc_max);
 
 	xfer->desc_adjacent = desc_max;
-	 */
 
 	/* terminate last descriptor */
 	last = desc_max - 1;
@@ -3038,7 +3042,7 @@ static int transfer_init(struct xdma_engine *engine,
 	return 0;
 }
 
-#ifdef __LIBXDMA_DEBUG__
+#if defined(__LIBXDMA_DEBUG__) || defined(__LIBXDMA_SMALLDBG__)
 static void sgt_dump(struct sg_table *sgt)
 {
 	int i;
@@ -3265,7 +3269,7 @@ ssize_t xdma_xfer_submit(void *dev_hndl, int channel, bool write, u64 ep_addr,
 			req->ep_addr, done, req->sw_desc_idx, req->sw_desc_cnt);
 
 #ifdef __LIBXDMA_DEBUG__
-//		transfer_dump(xfer);
+		transfer_dump(xfer);
 #endif
 
 		rv = transfer_queue(engine, xfer);
@@ -3318,7 +3322,7 @@ ssize_t xdma_xfer_submit(void *dev_hndl, int channel, bool write, u64 ep_addr,
 				xfer->len, req->ep_addr - xfer->len);
 			spin_unlock_irqrestore(&engine->lock, flags);
 
-#ifdef __LIBXDMA_DEBUG__
+#if defined(__LIBXDMA_DEBUG__) || defined(__LIBXDMA_SMALLDBG__)
 			transfer_dump(xfer);
 			sgt_dump(sgt);
 #endif
@@ -3344,7 +3348,7 @@ ssize_t xdma_xfer_submit(void *dev_hndl, int channel, bool write, u64 ep_addr,
 			}
 			spin_unlock_irqrestore(&engine->lock, flags);
 
-#ifdef __LIBXDMA_DEBUG__
+#if defined(__LIBXDMA_DEBUG__) || defined(__LIBXDMA_SMALLDBG__)
 			transfer_dump(xfer);
 			sgt_dump(sgt);
 #endif
@@ -3455,7 +3459,7 @@ ssize_t xdma_xfer_completion(void *cb_hndl, void *dev_hndl, int channel,
 			pr_info("xfer 0x%p,%u, failed, ep 0x%llx.\n",
 				xfer, xfer->len, req->ep_addr - xfer->len);
 
-#ifdef __LIBXDMA_DEBUG__
+#if defined(__LIBXDMA_DEBUG__) || defined(__LIBXDMA_SMALLDBG__)
 			transfer_dump(xfer);
 			sgt_dump(sgt);
 #endif
@@ -3471,7 +3475,7 @@ ssize_t xdma_xfer_completion(void *cb_hndl, void *dev_hndl, int channel,
 
 			xdma_engine_stop(engine);
 
-#ifdef __LIBXDMA_DEBUG__
+#if defined(__LIBXDMA_DEBUG__) || defined(__LIBXDMA_SMALLDBG__)
 			transfer_dump(xfer);
 			sgt_dump(sgt);
 #endif
```
</details>

jberaud, 2021-04-13:
Ok it was worth trying. The errors you are seeing seem to be different.
It is interesting that adding debug code only in case of failures seem to make the io faults disappear. I agree with the possible explanation you proposed on aws-fpga. In case of a timeout, the run bit is cleared but there is no waiting for the engine to actually stop before unmapping the user pages from dma so it is totally possible that it is what happens. There is the question of why the timeout occurs in the first place though, but the reason is possibly hardware.

X-Ryl669, 2021-04-13:
For such error:
```
[ 7269.622085] amd_iommu_report_page_fault: 173 callbacks suppressed
[ 7269.622086] AMD-Vi: Event logged [IO_PAGE_FAULT device=0e:00.0 domain=0x001a address=0xfec09400 flags=0x0020]
[ 7269.622093] AMD-Vi: Event logged [IO_PAGE_FAULT device=0e:00.0 domain=0x001a address=0xfec09600 flags=0x0020]
[ 7269.622100] AMD-Vi: Event logged [IO_PAGE_FAULT device=0e:00.0 domain=0x001a address=0xfec09800 flags=0x0020]
[ 7269.622106] AMD-Vi: Event logged [IO_PAGE_FAULT device=0e:00.0 domain=0x001a address=0xfec09a00 flags=0x0020]
[ 7269.622113] AMD-Vi: Event logged [IO_PAGE_FAULT device=0e:00.0 domain=0x001a address=0xfec09c00 flags=0x0020]
[ 7269.622120] AMD-Vi: Event logged [IO_PAGE_FAULT device=0e:00.0 domain=0x001a address=0xfec09e00 flags=0x0020]
[ 7269.727902] xdma:xdma_xfer_submit: xfer 0x00000000f87a0510,1048832, s 0x1 timed out, ep 0x300100.
[ 7269.727905] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x000000005c758ea8) = 0x1fc10006 (id).
[ 7269.727908] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x00000000512d09a5) = 0x00000001 (status).
[ 7269.727911] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x000000002b363f27) = 0x00f83e1f (control)
[ 7269.727914] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x000000008241e18b) = 0xffbc97e0 (first_desc_lo)
[ 7269.727917] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x000000001a9f4222) = 0x00000000 (first_desc_hi)
[ 7269.727920] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x00000000901bcee4) = 0x00000000 (first_desc_adjacent).
[ 7269.727923] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x00000000cf103922) = 0x00000005 (completed_desc_count).
[ 7269.727926] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x00000000551ee86f) = 0x00f83e1e (interrupt_enable_mask)
[ 7269.727929] xdma:engine_status_dump: SG engine 0-C2H0-MM status: 0x00000001: BUSY
[ 7269.727930] xdma:transfer_abort: abort transfer 0x00000000f87a0510, desc 257, engine desc queued 0.
[ 7269.727943] xdma:transfer_dump: xfer 0x00000000f87a0510, state 0x4, f 0x1, dir 2, len 1048832, last 1.
```
I have this io mapping:
```
$ sudo cat /proc/iomem
[...]
    feb80000-febfffff : amd_iommu
    fec00000-fec003ff : IOAPIC 0
    fec01000-fec013ff : IOAPIC 1
  fec10000-fec10fff : pnp 00:04
[...]
```

So the address *0xfec09400* is indeed not mapped.

My current hypothesis is that the driver "times out" and start deregistering descriptors/sgl but the engine is still running. It would then tries to read from (now) unknown address causing the IO_PAGE_FAULT error I'm experiencing. Adding the descriptor dumping in that case adds more time for the engine to actually finish the transfer before the descriptor are unregistered/unmapped and that would explain why it does not happen in that case.
I've moved the dumping code after the transfer_destroy function call and I do get the error (but the descriptors are gone so I'm dumping all zero in that case), with this code:
```cpp
		default:
			/* transfer can still be in-flight */
			pr_info("xfer 0x%p,%u, s 0x%x timed out, ep 0x%llx.\n",
				xfer, xfer->len, xfer->state, req->ep_addr);
			rv = engine_status_read(engine, 0, 1);
			if (rv < 0) {
				pr_err("Failed to read engine status\n");
			} else if (rv == 0) {
				//engine_status_dump(engine);
				rv = transfer_abort(engine, xfer);
				if (rv < 0) {
					pr_err("Failed to stop engine\n");
				} else if (rv == 0) {
					rv = xdma_engine_stop(engine);
					if (rv < 0)
						pr_err("Failed to stop engine\n");
				}
			}
			spin_unlock_irqrestore(&engine->lock, flags);

#if defined(__LIBXDMA_DEBUG__)
			transfer_dump(xfer);
			sgt_dump(sgt);
#endif
#if defined(__LIBXDMA_SMALLDBG__)
			dump = 1; /* Don't delay the destroy of the transfer by dumping here, do it later */
#endif
			rv = -ERESTARTSYS;
			break;
		}

		engine->desc_used -= xfer->desc_num;
		transfer_destroy(xdev, xfer);
#if defined(__LIBXDMA_SMALLDBG__)
		if (dump) {
			transfer_dump(xfer);
			sgt_dump(sgt);
		}
#endif
```

X-Ryl669, 2021-04-13:
> but there is no waiting for the engine to actually stop

I'd also like to understand why it can time out (it does not seems possible to me from a HW point of view). Yet I would have expected that the call to `xdma_engine_stop` above would have actually stopped the engine...

From a software point of view, in my initial code, when I get such error, I'm calling XDMA_IOCOFFLINE/XDMA_IOCONLINE to recover and retry the transfer and it usually succeed (until the some later failure for another transfer). But it's not stable enough IMHO.

jberaud, 2021-04-13:
I think your explanation is correct. So you have 2 issues. First the timeout which shouldn't happen, but it possibly comes from the hardware limitations. You could try to increase the timeout params at module load (h2c_timeout and c2h_timeout).
As for how to shutdown properly, I think the right way to handle that would be that in case of timeout, after calling xdma_engine_stop, the xfer_submit routine waits for the engine to stop being busy, checking for the busy bit in the status register and sleeping.

dmax621, 2021-06-30:
This may or may not be related, but I think there's a [bug in qdma_request_wait_for_cmpl()](https://forums.xilinx.com/t5/PCIe-and-CPM/QDMA-Error-Timeout-for-completion-Important-and-Urgent/m-p/1254834/highlight/true#M19066) which results in "impossible" timeouts. The XDMA code may be similarly vulnerable to pending signals in multi-threaded programs -- there's at least one place where it ignores the return value from some flavour of wait_event.