Xilinx/dma_ip_drivers #118: QDMA latency measurment (dmalat) for MM Engine

insoulity, 2021-04-08:
Hi there, 

I have a question abut latency measurement for QDMA system.

https://github.com/Xilinx/dma_ip_drivers/blob/11747e09faf287c264ad5ccaf13ed26fb9e4593a/QDMA/linux-kernel/apps/dma-latency/dmalat.c#L158

The line marked in the link above seems to indicate this application (dmalat) may intend to measure the latency of only Stream Engine (and NOT MM Engine) although there are some cases that consider when q_mode = Q_MODE_MM.

Thus, I'd like to ask whether the application is designed to measure the latency of both Q_MODE_MM and Q_MODE_ST or not.

Thanks,
Yongkee

hmaarrfk, 2021-04-08:
Xilinx has a policy not to respond to people using this website. You should ask your question on their forum.

insoulity, 2021-04-09:
Thanks, I will post there then.

closed 2021-04-09 by insoulity