Xilinx/dma_ip_drivers #120: Second C2H DMA doesn't assert `tready` signal

faku99, 2021-04-23:
First of all, here is the design I'm using:

[image]

The first DMA channel (`C2H_0`) initialization is done correctly. After allocating memory, descriptors and requesting a transfer, the `tready` signal is asserted and I can run multiple transfers without any issues.

But when the second DMA channel (`C2H_1`) initialization is (successfully) done, the `tready` signal simply stays at `0`. It never takes the value `1`. I already checked if there could be some non-wanted `tlast` or `tvalid` before initialization but that's not the case.

Below, find a dump of both DMA channels after initialization:

```
[  310.820955] selftest_qrng:sgt_alloc_with_pages: 0-C2H0-ST: Successfully allocated PCI coherent memory - phys: 0x3C000000, virt: 0xAC000000, size: 0x20000
[  310.840149] selftest_qrng:engine_reg_dump: 0-C2H0-ST: ioread32(0xe6281406) = 0x1fc18006 (id).
[  310.848698] selftest_qrng:engine_reg_dump: 0-C2H0-ST: ioread32(0x93a33b61) = 0x00000001 (status).
[  310.857574] selftest_qrng:engine_reg_dump: 0-C2H0-ST: ioread32(0x159b0e0e) = 0x04f83e19 (control)
[  310.866467] selftest_qrng:engine_reg_dump: 0-C2H0-ST: ioread32(0x0d4507f3) = 0x3c080000 (first_desc_lo)
[  310.875874] selftest_qrng:engine_reg_dump: 0-C2H0-ST: ioread32(0x735f40c1) = 0x00000000 (first_desc_hi)
[  310.885291] selftest_qrng:engine_reg_dump: 0-C2H0-ST: ioread32(0x28744a12) = 0x0000000f (first_desc_adjacent).
[  310.895304] selftest_qrng:engine_reg_dump: 0-C2H0-ST: ioread32(0x1fe7ae6e) = 0x00000000 (completed_desc_count).
[  310.905413] selftest_qrng:engine_reg_dump: 0-C2H0-ST: ioread32(0x3730dec6) = 0x00f83e18 (interrupt_enable_mask)
[  310.915873] selftest_qrng:sgt_alloc_with_pages: 0-C2H1-ST: Successfully allocated PCI coherent memory - phys: 0x3C020000, virt: 0xAC020000, size: 0x20000
[  310.938338] selftest_qrng:engine_reg_dump: 0-C2H1-ST: ioread32(0x7e61b4f9) = 0x1fc18106 (id).
[  310.946865] selftest_qrng:engine_reg_dump: 0-C2H1-ST: ioread32(0xe4974543) = 0x00000001 (status).
[  310.955760] selftest_qrng:engine_reg_dump: 0-C2H1-ST: ioread32(0x49369d06) = 0x04f83e19 (control)
[  310.964648] selftest_qrng:engine_reg_dump: 0-C2H1-ST: ioread32(0xee13876c) = 0x3c0a0000 (first_desc_lo)
[  310.974061] selftest_qrng:engine_reg_dump: 0-C2H1-ST: ioread32(0xe2796730) = 0x00000000 (first_desc_hi)
[  310.983474] selftest_qrng:engine_reg_dump: 0-C2H1-ST: ioread32(0x393b21b4) = 0x0000000f (first_desc_adjacent).
[  310.993496] selftest_qrng:engine_reg_dump: 0-C2H1-ST: ioread32(0xf06c42d3) = 0x00000000 (completed_desc_count).
[  311.003595] selftest_qrng:engine_reg_dump: 0-C2H1-ST: ioread32(0xe15fe714) = 0x00f83e18 (interrupt_enable_mask)
```

The registers dump looks fine to me.

For the sake of curiosity, I swapped both channels and surprisingly, the channel `C2H_0` works but `C2H_1` doesn't.

Thus, the problem comes from the driver, and I suspect an issue with some index somewhere.

Don't hesitate if you need some more information.

Lucas

hmaarrfk, 2021-04-23:
Xilinx has a policy not to monitor this forum. Better to use their official one.

MischaBaars, 2024-01-13:
Hello Lucas,

Are you still working on this or similar projects? Looks like a very good start.

Best regards,
Mischa.