Xilinx/dma_ip_drivers #123: QDMA insmod error with QDMA_OFFSET_IND_CTXT_CMD

wonsikleee, 2021-07-01:
Hi,

I found some error with hw_monitor_reg function call.

In driver/libqdma/qdma_device.c file, there are some code trying to initialize the context memory.

```
258     if (xdev->conf.master_pf) {
259         rv = xdev->hw.qdma_init_ctxt_memory(xdev);
260         if (rv < 0) {
261             pr_err("init ctxt write failed, err %d\n", rv);
262             return xdev->hw.qdma_get_error_code(rv);
263         }
264     }
```

However, when xdev->hw.qdma_init_ctxt_memory(xdev) calls qdma_indirect_reg_clear function, it returns error -3.
The address offset the driver trying to read is 0x844.

driver/libqdma/qdma_access/qdma_soft_access/qdma_soft_reg.h
 93 #define QDMA_OFFSET_IND_CTXT_CMD                            0x844

If I comment the code lines from 258 to 264 in qdma_device.c file, it works.

wonsikleee, 2021-07-01:
I found the reference link for this problem.
https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/QDMA-driver-fails-to-initialize-eqdma-indirect-reg-clear/td-p/1135310

closed 2021-07-01 by wonsikleee

hmaarrfk, 2021-07-01:
ah yes. you have to hold certain signals high yourself. this bit us too for a long time.

best of luck!