Xilinx/dma_ip_drivers #125: installing and loading module failed

mashemat, 2021-08-04:
Hi,

I am trying to use the driver. I successfully make the module however I receive the following error when try to run _**make install**_

Makefile:10: XVC_FLAGS: .
make -C /lib/modules/4.15.0-151-generic/build M=/home/mashemat/Downloads/dma_ip_drivers-master/XDMA/linux-kernel/xdma modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-151-generic'
/home/mashemat/Downloads/dma_ip_drivers-master/XDMA/linux-kernel/xdma/Makefile:10: XVC_FLAGS: .
  Building modules, stage 2.
/home/mashemat/Downloads/dma_ip_drivers-master/XDMA/linux-kernel/xdma/Makefile:10: XVC_FLAGS: .
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-151-generic'
make -C /lib/modules/4.15.0-151-generic/build M=/home/mashemat/Downloads/dma_ip_drivers-master/XDMA/linux-kernel/xdma modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-151-generic'
  INSTALL /home/mashemat/Downloads/dma_ip_drivers-master/XDMA/linux-kernel/xdma/xdma.ko
At main.c:160:
**- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory**
  DEPMOD  4.15.0-151-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-151-generic'

if I ignore this step and try to run a test (e.g., load_driver.sh) it says module is installed successfully but it could not detect the device. 

What would be problem?

Thanks,

ksielewicz, 2021-08-05:
Hi @mashemat,

1) What OS are you using?
2) is your device visible in the system? lspcie | grep Xilinx ?
3) try running
sudo depmod
sudo modprobe xdma
4) did you run ./load_driver.sh with sudo?

Regards

mashemat, 2021-08-05:
Hi @ksielewicz,

Thanks for your reply!

Actually, I have an NVIDIA Melanox Innova 2 flex board that has Xilinx Kintex over it. It came with a design example that support xdma PCIe and an IMG in the design with vivado 2018. Here it is the page:

https://docs.mellanox.com/display/Innova2Flex/NVIDIA+INNOVA-2+FLEX+OPEN+ADAPTER+CARD

I programmed the FPGA with example. I hav Ubuntu 18.04.5 LTS. This is the output of  _**lspci**_:

c1:00.0 PCI bridge: Mellanox Technologies MT28800 Family [ConnectX-5 PCIe Bridge]
c2:08.0 PCI bridge: Mellanox Technologies MT28800 Family [ConnectX-5 PCIe Bridge]
c2:10.0 PCI bridge: Mellanox Technologies MT28800 Family [ConnectX-5 PCIe Bridge]
c4:00.0 Ethernet controller: Mellanox Technologies MT27800 Family [ConnectX-5]
c4:00.1 Ethernet controller: Mellanox Technologies MT27800 Family [ConnectX-5]

It can detect the board in this way. But when I run **_./load_driver_** again it fails:

xdma                   81920  0
Loading xdma driver...
Error: The Kernel module installed correctly, but no devices were recognized.
 FAILED

Thanks,

ksielewicz, 2021-08-05:
Yes, it fails because lspci does not see the Xilinx FPGA. From the documentation it looks like there is a switch between the host and the FPGA.

Regards

mashemat, 2021-08-05:
Yes, there is a switch in between. But in the documentation they didn’t mention it causes any problem. Even, they redirect using xdma as the host driver to test. What would be solution now? 

Thanks,

mashemat, 2021-08-10:
Hello again,

Actually, I tried to add my PCIe device ID to the list  in the xdma_mod.c and compile again but it can not detect my board again:

diff -auT xdma_mod.c.ORIG xdma_mod.c
--- xdma_mod.c.ORIG	2021-08-09 14:38:47.613949122 +0200
+++ xdma_mod.c	2021-08-09 15:50:47.318399338 +0200
@@ -104,6 +104,7 @@
		{ PCI_DEVICE(0x10ee, 0x4B28), },
	
		{ PCI_DEVICE(0x10ee, 0x2808), },
+		{ PCI_DEVICE(0x15b3, 0x1974), },  // Mellanox Innova-2 15b3:1974
	
	#ifdef INTERNAL_TESTING
		{ PCI_DEVICE(0x1d0f, 0x1042), 0},

Do you think this driver is useful for my case?

Thanks,

mashemat, 2021-08-11:
My problem solved. The problem was the MLNX_OFED version not the switch since my board (Innova-2 flex) requires specific version of OFED to work. Now the xdma driver is working fine!

closed 2021-08-11 by mashemat

mwrnd, 2022-12-07:
@mashemat - What are the versions of `MLNX_OFED`, `dma_ip_drivers`, and Linux Kernel that work for you?

When you run `ls /dev`, do you have `xdma0_xvc`?