Xilinx/dma_ip_drivers #133: xdma error : Failed to dump register

KcMeterCEC, 2021-11-10:
Hi,
I use IMX8MM to communicate with zynq7035 through PCIE2.0x1 interface.

This is block design of zynq7035:
[image]
And the configuration of xdma is:
[image]

These are the processes how i test PCIE：
1. Write a constant 32bits value to AXI-Lite register,and readback to check if they are equal.
2. Write 10MBytes block data to AXI-Stream,and readback to check if they are equal.

But the AXI-Stream sometimes become failed with error messages:
```shell
[10788.112808] xdma:xdma_xfer_submit: xfer 0x00000000ecec6f5c,8388608, s 0x1 timed out, ep 0x900000.
[10788.172892] xdma:engine_reg_dump: 0-H2C0-MM: ioread32(0x00000000f7add3ea) = 0xffffffff (id).
[10788.181329] xdma:engine_reg_dump: 0-H2C0-MM: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[10788.190891] xdma:engine_status_read: Failed to dump register
[10788.196547] xdma:xdma_xfer_submit: Failed to read engine status
```
The error messages may happen in10 hours, 3 hours,30 minutes after start the test.

I used the newest version of `dma_ip_drivers` in `master` branch.
I found the default value of  `h2c_timeout` and `c2h_timeout`  are 10s.Which means it's long enough to get data from C2H.

It looks like that xdma is crashed.
How can i resolve the problem?

Thanks in advance!

makslevental, 2022-02-07:
@KcMeterCEC did you manage to figure this out?

KcMeterCEC, 2022-02-07:
@makslevental 
Hi,
I have tried:
1. Update linux kernel to newleast version(5.4.160).
2. Use riffa driver instead of xdma driver.

But didn't solve this problem.
So i think this is hardware problem.

makslevental, 2022-02-07:
@KcMeterCEC oh interesting. I'm curious: were you able to get riffa to work?

dmitrym1, 2025-01-17:
#317