Xilinx/dma_ip_drivers #152: About dma subsystem for pci express(3.0)

nalamalav, 2022-03-24:
hi all,
Currently i a m working on dma subsystem for pci express version 3.0 in tool of vivado 2016.3.
i was taken a 1 MB of bar size,when i was trying to accessing a entire memory location from driver half of memory location was getting and i try to written next memory location(512kb onwards) the message going to be killed it was not written to the entire memory location what is the issue.
i am written a driver logic in linux.

hmaarrfk, 2022-03-30:
I'm not sure I understand.

hmaarrfk, 2022-03-30:
Also, you will get more help from xilinx if you post on their forum.

nalamalav, 2022-03-31:
> Also, you will get more help from xilinx if you post on their forum.

okay sure.