Xilinx/dma_ip_drivers #154: Not receiving interrupt on Rx side when using QDMA linux driver 2019.2

aditya-venu, 2022-03-25:
Hi,
I developed a wrapper around the QDMA linux driver for the FPGA to be detected as an ethernet interface like QEP drivers(https://xilinx.github.io/qep-drivers/master/linux-kernel/html/index.html)

I created 2 tx and 2 rx queues for QDMA. I made an external SFP loopback connection. When I send tx in queue number 1, I am able to receive in queue 0 of Rx. This happens only when qmax is set to 2048 in the driver.

But when qmax is 1024 or any other value, I am not able to receive on queue number 0 of Rx.

I configured QDMA in direct interrupt mode. In both the cases(qmax=1024 and qmax=2048), Tx interrupt is getting triggered but Rx interrupt is not getting triggered when qmax is 1024. 

In the dmesg logs, I find qbase is 0 when qmax is 2048, and qbase is 256 when qmax is 1024. 

In Tx path, I observed TDEST as 0x101(257 in decimal) when I send from queue number 1 and qmax set to 1024. I inserted a HLS IP to route the packet to ethernet IP even if the TDEST is 0x101(In my design QDMA is connected to Ethernet IP).

But on Rx path, I observe everytime that TDEST value as 0 at the input of QDMA IP, even if Qmax is 2048 or 1024. But only if qmax is 2048, I am able to receive interrupt on Rx. I am not getting receive interrupt when qmax is 1024. 

I want Rx to work even if qmax is 1024. 

How to resolve this issue?

How to resolve this?

hmaarrfk, 2022-03-26:
Did you restart your computer between reprogramming? I found strangeness when I didn't.

Ps. I'm just an other QDMA user.

aditya-venu, 2022-03-26:
> _Did you restart your computer between reprogramming? I found strangeness when I didn't._
 Yes I did. 
> Ps. I'm just an other QDMA user.