Xilinx/dma_ip_drivers #169: QDMA linux driver not compiling

aditya-venu, 2022-06-30:
Hi,
I followed the following steps:
1. git clone https://github.com/Xilinx/dma_ip_drivers.git
2. cd dma_ip_drivers
3. git checkout 2019.2
4. cd QDMA/linux-kernel/
5. make clean && make

Here is the output of the make:

 make clean && make
CROSS_COMPILE_FLAG = .
ARCH = .
#######################
####  qdmautils    ####
#######################
make -C qdmautils clean;
make[1]: Entering directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/qdmautils'
rm -rf *.o
make[1]: Leaving directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/qdmautils'
#######################
####  user         ####
#######################
make -C user clean;
make[1]: Entering directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/user'
make[1]: Leaving directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/user'
#######################
####  tools        ####
#######################
make -C tools clean;
make[1]: Entering directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/tools'
rm -rf *.o *.bin dma_to_device dma_from_device dmaperf
make[1]: Leaving directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/tools'
#######################
####  drv          ####
#######################
srcdir = /home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel.
KSRC = .
VF = .
CROSS_COMPILE_FLAG = .
ARCH = .
EXTRA_FLAGS = .
ccflags-y = -D__GENL_REG_FAMILY_OPS_FUNC__ -D__READ_ONCE_DEFINED__.
make[1]: Entering directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/drv'
make[1]: Leaving directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/drv'
"distro=, dmajor= dminor= "
5.4.170-rt68: GIT,,, -DGITSP -DGIT
CROSS_COMPILE_FLAG = .
ARCH = x86_64.
#######################
####  user        ####
#######################
make -C user
make[1]: Entering directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/user'
cc -g -c -std=c99 -o cli/cmd_parse.o cli/cmd_parse.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -g -c -std=c99 -o cli/main.o cli/main.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -g -c -std=c99 -o ../qdmautils/qdmactl.o ../qdmautils/qdmactl.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -g -c -std=c99 -o ../qdmautils/qdmactl_reg.o ../qdmautils/qdmactl_reg.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -g -c -std=c99 -o ../qdmautils/qdmautils.o ../qdmautils/qdmautils.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -g -c -std=c99 -o ../qdmautils/dmaxfer.o ../qdmautils/dmaxfer.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -pthread -lrt cli/cmd_parse.o cli/main.o ../qdmautils/qdmactl.o ../qdmautils/qdmactl_reg.o ../qdmautils/qdmautils.o ../qdmautils/dmaxfer.o -o cli/dmactl -laio -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -g -c -std=c99 -o qdma_xfer_app/qdma_xfer_app.o qdma_xfer_app/qdma_xfer_app.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -pthread -lrt qdma_xfer_app/qdma_xfer_app.o ../qdmautils/qdmactl.o ../qdmautils/qdmactl_reg.o ../qdmautils/qdmautils.o ../qdmautils/dmaxfer.o -o qdma_xfer_app/qdma_xfer_app -laio -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
make[1]: Leaving directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/user'
#######################
####  tools        ####
#######################
make -C tools
make[1]: Entering directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/tools'
cc -g -c -std=c99 -o dma_to_device.o dma_to_device.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -lrt -o dma_to_device dma_to_device.o -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -g -c -std=c99 -o dma_from_device.o dma_from_device.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -lrt -o dma_from_device dma_from_device.o -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
cc -g -c -std=c99 -o dmaperf.o dmaperf.c -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE -D_AIO_AIX_SOURCE
cc -pthread -lrt -o dmaperf dmaperf.o -laio -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGE_FILE_SOURCE
make[1]: Leaving directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/tools'
#######################
#### PF drv          ####
#######################
srcdir = /home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel.
KSRC = /lib/modules/5.4.170-rt68/source.
VF = 0.
CROSS_COMPILE_FLAG = .
ARCH = x86_64.
EXTRA_FLAGS = .
ccflags-y = -D__READ_ONCE_DEFINED__.
make[1]: Entering directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/drv'
make[2]: Entering directory `/kernel/linux-5.4.170'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  DESCEND  objtool
  CHK     include/generated/compile.h
  SKIPPED include/generated/compile.h
  Building modules, stage 2.
  MODPOST 3802 modules
make[2]: Leaving directory `/kernel/linux-5.4.170'
cp: cannot stat ‘qdma.ko’: No such file or directory
make[1]: *** [build] Error 1
make[1]: Leaving directory `/home/iota/aditya/dma_ip_drivers/QDMA/linux-kernel/drv'
make: *** [mod_pf] Error 2

**Here are my os details**

 uname -arn
Linux gay 5.4.170-rt68 #1 SMP PREEMPT_RT Tue Jan 18 13:29:21 IST 2022 x86_64 x86_64 x86_64 GNU/Linux

It is compiling fine without any errors in 
uname -arn
Linux localhost.localdomain 3.10.0-1160.el7.x86_64 #1 SMP Mon Oct 19 16:18:59 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Can you please resolve this? 
Please let me know if you want further details.

hmaarrfk, 2022-06-30:
I've also had trouble compiling 2019.2 with "newer" kernels. However, yours seems "old enough" that said, I think I would suggest moving toward the newer drivers. They seem to work with the older FPGA IP.

PS. I'm not affiliated with xilinx. I just know that they seem to have a mandate to not answer any questions here.

PPS. It really helps others if you format your code in triple back ticks. https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code

aditya-venu, 2022-06-30:
Will it be of any use if I post it in their forums?

hmaarrfk, 2022-08-22:
Hello,

My name is Mark Harfouche. I am not affiliated with Xilinx in any way. Over the
years of using QDMA, I've been wanted better community organization.

I've created a fork of dma_ip_drivers which I intend to maintain and work with the
community at large to improve.

The fork can be found https://github.com/hmaarrfk/dma_ip_drivers

For now, I am stating the main goals of the repository in
https://github.com/hmaarrfk/dma_ip_drivers/issues/2

If you are interested in working together, feel free to open an issue or PR to
my fork.

Best,

Mark

PS. Xilinx over the years, has shown that it does not intent to reply to this
repository. I've found that they tend reply if you post on their own forum.

aditya-venu, 2022-08-23:
> hmaarrfk#2 <https://github.com/hmaarrfk/dma_ip_drivers/issues/2>

Hi,

Thank you for mailing me. I agree with you that Xilinx doesn't reply on the
issues on their github QDMA repo or even in their forums.

With that being said, I just used their qdma driver to develop a network
driver around it for one of their hardware. Other than that I don't have
extensive experience with drivers and QDMA. For now I am using it in kernel
version 3.10. and QDMA version 2019.2

However I want to make it work with kernel version 5.4.170-rt68

I appreciate your effort in addressing the issues which are not answered by
them. I'd love to contribute but don't know how to.

Thanks,
Aditya