Xilinx/dma_ip_drivers #186: error: impossible constraint in ‘asm’   asm volatile("rdtscp" : "=a" (low), "=d" (high)) for ARM64

jumpmanhou, 2022-12-22:
I try to build the QDMA driver for the ARM64 arch, but there is an error, here is the error log:
#######################
####  Driver       ####
#######################
make  -C driver modulesymfile=/home/m1_sw/chunpeng/dma_ip_drivers/QDMA/linux-kernel/driver/src/Module.symvers
make[1]: Entering directory '/home/m1_sw/chunpeng/dma_ip_drivers/QDMA/linux-kernel/driver'
"distro=, dmajor= dminor= "
5.10.134-15852-gf17a528bd1c2-dirty: GIT,,, -DGITSP -DGIT
CROSS_COMPILE_FLAG = .
ARCH = aarch64.
#######################
#### PF src          ####
#######################
make[2]: Entering directory '/home/m1_sw/chunpeng/dma_ip_drivers/QDMA/linux-kernel/driver/src'
srcdir = /home/m1_sw/chunpeng/dma_ip_drivers/QDMA/linux-kernel/driver/src.
KSRC = /home/m1_sw/chunpeng/kernel-5.10.
VF = 0.
CROSS_COMPILE_FLAG = .
ARCH = aarch64.
EXTRA_FLAGS = .
ccflags-y = .
make[3]: Entering directory '/home/m1_sw/chunpeng/kernel-5.10'
srcdir = /home/m1_sw/chunpeng/dma_ip_drivers/QDMA/linux-kernel/driver/src.
KSRC = /home/m1_sw/chunpeng/kernel-5.10.
VF = 0.
CROSS_COMPILE_FLAG = .
ARCH = arm64.
EXTRA_FLAGS = .
ccflags-y = .
  CC [M]  /home/m1_sw/chunpeng/dma_ip_drivers/QDMA/linux-kernel/driver/src/libqdma/qdma_descq.o
/home/m1_sw/chunpeng/dma_ip_drivers/QDMA/linux-kernel/driver/src/libqdma/qdma_descq.c: In function ‘rdtsc_gettime’:
/home/m1_sw/chunpeng/dma_ip_drivers/QDMA/linux-kernel/driver/src/libqdma/qdma_descq.c:68:2: error: impossible constraint in ‘asm’
  asm volatile("rdtscp" : "=a" (low), "=d" (high));
  ^~~
make[4]: *** [scripts/Makefile.build:286: /home/m1_sw/chunpeng/dma_ip_drivers/QDMA/linux-kernel/driver/src/libqdma/qdma_descq.o] Error 1
make[3]: *** [Makefile:1831: /home/m1_sw/chunpeng/dma_ip_drivers/QDMA/linux-kernel/driver/src] Error 2
make[3]: Leaving directory '/home/m1_sw/chunpeng/kernel-5.10'
make[2]: *** [Makefile:135: bin] Error 2
make[2]: Leaving directory '/home/m1_sw/chunpeng/dma_ip_drivers/QDMA/linux-kernel/driver/src'
make[1]: *** [Makefile:206: mod_pf] Error 2
make[1]: Leaving directory '/home/m1_sw/chunpeng/dma_ip_drivers/QDMA/linux-kernel/driver'
make: *** [Makefile:94: driver] Error 2

Does this driver support on ARM64 platform?

hmaarrfk, 2022-12-22:
No it does not to my knowledge.

Xilinx doesn't like to answer on github. if you want to go through the upgrade collaboratively, I'll help review and move you along when i can:

https://github.com/hmaarrfk/dma_ip_drivers/

jumpmanhou, 2022-12-23:
Thanks, mark. The "rdtscp" is X86 instruction, I modified the function according to this link:https://lore.kernel.org/lkml/20200914115311.2201-3-leo.yan@linaro.org/ and it worked.

closed 2022-12-23 by jumpmanhou

dreamweevah, 2025-02-12:
My problem is a follow on to this problem.  I am trying to get a PCIe connection working between a Jetson AGX Orin Development Kit and a ZCU106.  I started here: 
https://www.youtube.com/watch?v=TzzzM97L4HI
and ended up here, finding this thread when I had the x86 vs ARM function issue described above.  Once I fixed the issue presented here, another popped up during the make.  Apparently, since the introduction of the driver from Answer 65444 (https://adaptivesupport.amd.com/s/article/65444?language=en_US), a function of the same name was introduced in Linux. FIELD_GET from the DMA drivers was giving me an error because FIELD_GET was already defined in the Linux distribution that I am running on the Jetson.  Ubuntu 20.04.6 is the version.  I fixed that by redefining FIELD_GET in the driver to FIELD_GET2 and it built.  I was then able to detect the ZCU106 with lspci.  However, when I add the DMA driver with modprobe and do a dmesg immediately afterward to see how it went, I am getting 'qdma_is_config_bar: Invalid config bar, err -4' and 'qdma_hw_access_init:config bar passed is invalid, err -1'.  I tracked the error to the code in QDMA/linux-kernel/driver/src/libqdma/qdma_access/qdma_access_common.c from the drivers at https://github.com/Xilinx/dma_ip_drivers.  I added the code that you see in the snippet that I have attached to show that reg_val comes up zero for whatever value that I put in the qdma.conf file for the config bar value.  In that file, the comments say 
"/** TODO: Version register for VFs is 0x5014 for EQDMA and
	 *  0x1014 for QDMA/Versal. First time reading 0x5014 for
	 *  all the device and based on the upper 16 bits value
	 *  (i.e. 0x1fd3), finding out whether its EQDMA or QDMA/Versal
	 *  for EQDMA VFs.
	 *  Need to modify this logic once the hardware team
	 *  comes up with a common register for VFs
	 */"

Is it reading from the wrong register?  Any help from anyone in the know would be greatly appreciated.  I am a longtime embedded systems engineer, but a Linux noob, so I wouldn't want to hazard a guess on what to change moving forward.

Thank you for your brain energy.  Linux takes alot of mine.

[config_bar problems.txt](https://github.com/user-attachments/files/18772901/config_bar.problems.txt)