Xilinx/dma_ip_drivers #187: Why reserving extra 4 bytes in mmap_bar()?

yu-zou, 2022-12-26:
In reg_read_mmap function defined in linux-kernel/apps/dma-utils/dmactl_regs.c
`cpp
mmap_bar(fname, xcmd->req.reg.reg+4, PROT_READ)
`
reserves extra 4 bytes in addition to the register length.
What's the point of this reservation?

I tried removing this extra reservation, and the code does not work correctly.
Originally, I thought it was used for address alignment, but it seems this extra 4 bytes have to be there.

Thank you.

hmaarrfk, 2022-12-26:
The integer before the +4 likely refers to its address. You thus need to map not only the address, but also the size of the address you want to read.

yu-zou, 2022-12-26:
@hmaarrfk Thanks for the reply. I see how it works.
If I want to access a register at 4K address, then I should map 0~(4K+4), otherwise when I access the register at 4K, it
will exceed boundary.

closed 2022-12-26 by yu-zou