Xilinx/dma_ip_drivers #191: Support for Ubuntu 22.04

juan-escobedo, 2023-02-21:
Hello,

I am trying to build the driver on a computer running Ubuntu 22.04 but when trying to run "make install" from the xdma folder, I get the following errors related to undefined functions:

/home/user1/repos/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c: In function ‘transfer_destroy’:
/home/user1/repos/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c:2911:25: error: implicit declaration of function ‘pci_unmap_sg’; did you mean ‘pci_unmap_rom’? [-Werror=implicit-function-declaration]
 2911 |                         pci_unmap_sg(xdev->pdev, sgt->sgl, sgt->nents,
      |                         ^~~~~~~~~~~~
      |                         pci_unmap_rom
/home/user1/repos/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c: In function ‘xdma_xfer_aperture’:
/home/user1/repos/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c:3195:30: error: implicit declaration of function ‘pci_map_sg’; did you mean ‘pci_map_rom’? [-Werror=implicit-function-declaration]
 3195 |                 sgt->nents = pci_map_sg(xdev->pdev, sgt->sgl, sgt->orig_nents,
      |                              ^~~~~~~~~~
      |                              pci_map_rom
/home/user1/repos/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c: In function ‘set_dma_mask’:
/home/user1/repos/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c:4194:14: error: implicit declaration of function ‘pci_set_dma_mask’; did you mean ‘set_dma_mask’? [-Werror=implicit-function-declaration]
 4194 |         if (!pci_set_dma_mask(pdev, DMA_BIT_MASK(64))) {
      |              ^~~~~~~~~~~~~~~~
      |              set_dma_mask
/home/user1/repos/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.c:4200:17: error: implicit declaration of function ‘pci_set_consistent_dma_mask’ [-Werror=implicit-function-declaration]
 4200 |                 pci_set_consistent_dma_mask(pdev, DMA_BIT_MASK(64));
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:257: /home/user1/repos/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.o] Error 1
make[1]: *** [Makefile:1850: /home/user1/repos/dma_ip_drivers/XDMA/linux-kernel/xdma] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-32-generic'
make: *** [Makefile:39: all] Error 2

The contents of /proc/version:

Linux version 5.19.0-32-generic (buildd@lcy02-amd64-026) (x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #33~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Jan 30 17:03:34 UTC 2

Thanks beforehand,

Juan

gnodipac886, 2023-02-23:
Same here! Any help would be great!

closed 2023-02-23 by juan-escobedo

juan-escobedo, 2023-02-23:
[This](https://github.com/Xilinx/dma_ip_drivers/pull/179) pull request this repo aims to solve this issue but is still not merged as of writing this.