Xilinx/dma_ip_drivers #198: Unknown error 512

zchliu, 2023-04-05:
I am compiling xdma driver on Ubuntu 22.04 and successfully install the driver according to the method described in the article https://github.com/Xilinx/dma_ip_drivers/issues/191. However, when I ran the read-write program, it gave me an error.
```$sudo ./dma_from_device -d /dev/xdma0_c2h_0 -f receive.bin -s 4 -a 0x0 -c 1 
/dev/xdma0_c2h_0, read 0x4 @ 0x0 failed -1.
read file: Unknown error 512
```
I used the `sudo dmesg `command to view the messages, and it printed the following information
```
[ 1349.164578] xdma: loading out-of-tree module taints kernel.
[ 1349.164625] xdma: module verification failed: signature and/or required key missing - tainting kernel
[ 1349.165265] xdma:xdma_mod_init: Xilinx XDMA Reference Driver xdma v2020.2.2
[ 1349.165266] xdma:xdma_mod_init: desc_blen_max: 0xfffffff/268435455, timeout: h2c 10 c2h 10 sec.
[ 1349.165303] xdma:xdma_device_open: xdma device 0000:01:00.0, 0x0000000086afbf74.
[ 1349.165310] xdma 0000:01:00.0: enabling device (0000 -> 0002)
[ 1349.165447] xdma:map_single_bar: BAR0 at 0xb1200000 mapped at 0x00000000b878dea7, length=65536(/65536)
[ 1349.165452] xdma:map_bars: config bar 0, pos 0.
[ 1349.165453] xdma:identify_bars: 1 BARs: config 0, user -1, bypass -1.
[ 1349.165505] xdma:probe_one: 0000:01:00.0 xdma0, pdev 0x0000000086afbf74, xdev 0x00000000b02dd059, 0x00000000f7701d90, usr 16, ch 1,1.
[ 1474.075399] xdma:xdma_xfer_submit: xfer 0x00000000f4f545ee,4, s 0x1 timed out, ep 0x4.
[ 1474.075407] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x00000000708578f0) = 0x1fc10006 (id).
[ 1474.075410] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x000000000e88c804) = 0x00000006 (status).
[ 1474.075412] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x000000003220d0fe) = 0x00f83e1f (control)
[ 1474.075415] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x00000000004d8ab3) = 0x275d0000 (first_desc_lo)
[ 1474.075417] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x000000003ecfa7d8) = 0x00000002 (first_desc_hi)
[ 1474.075420] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x00000000ae30d472) = 0x00000000 (first_desc_adjacent).
[ 1474.075422] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x00000000e12eddd0) = 0x00000001 (completed_desc_count).
[ 1474.075424] xdma:engine_reg_dump: 0-C2H0-MM: ioread32(0x00000000dd5f2229) = 0x00f83e1e (interrupt_enable_mask)
[ 1474.075427] xdma:engine_status_dump: SG engine 0-C2H0-MM status: 0x00000006: DESC_STOPPED,DESC_COMPL
[ 1474.075428] xdma:transfer_abort: abort transfer 0x00000000f4f545ee, desc 1, engine desc queued 0.
```
Does anyone know the cause of this problem and give some suggestions?

Gothel17, 2023-04-20:
I have the same problem. Do you have any solution ?

ProbsNotNova, 2024-05-24:
I am having the same 512 problem coming up. No obvious code issues that I can find that wouldn't require going down to some deep level of abstraction or having to change the method of my code that utilizes the dma_to_device and dma_from_device programs in some specialized form or another.

What might I want to try and do? The base programs and the specialized forms both work, but implementing them within further specialization that involves a mix of CPP and C seems to be where the issue comes up. Even though all the values I am using within read in read_to_buffer should be valid as they are no different from what I used in the base ones

Beandot4, 2024-12-06:
I met the error, too, and got the same messages with `sudo dmesg | grep xdma` on Ubuntu 22.04. Is there any solution?

ZhangLinYiQAQ, 2025-01-04:
> 我也遇到了这个错误，`sudo dmesg | grep xdma`在 Ubuntu 22.04 上也收到了同样的消息。有什么解决办法吗？
请问解决了吗

dmitrym1, 2025-01-17:
#311
@Beandot4 try this

tjy-zhu, 2025-10-07:
> [#311](https://github.com/Xilinx/dma_ip_drivers/issues/311) [@Beandot4](https://github.com/Beandot4) try this

I try it,  but still have this:
Info: The PCIe DMA core is memory mapped.
Info: Running PCIe DMA memory mapped write read test
	transfer size:  1024, count: 1
Info: Writing to h2c channel 0 at address offset 0.
Info: Wait for current transactions to complete.
/dev/xdma0_h2c_0, write 0x400 @ 0x0 failed -1.
write file: Unknown error 512
Info: Writing to h2c channel 0 at address offset 1024.
Info: Wait for current transactions to complete.
/dev/xdma0_h2c_0, write 0x400 @ 0x400 failed -1.
write file: Unknown error 512
Info: Writing to h2c channel 0 at address offset 2048.
Info: Wait for current transactions to complete.