Xilinx/dma_ip_drivers #206: Error: The Kernel module installed correctly, but no devices were recognized.  FAILED

Ranyang-Zhou, 2023-05-17:
Hi everyone,

I met the problem as the title shown when I try to connect FPGA to host PC.
The environment is:

FPGA: Alveo U200
Vivado 2020.2
Ubuntu 20.02
error report:
sudo ./load_driver.sh 
[sudo] password for acad: 
interrupt_selection .
xdma                   98304  0
Loading driver...insmod xdma.ko interrupt_mode=2 ...
insmod: ERROR: could not insert module ../xdma/xdma.ko: Invalid module format

Error: The Kernel module installed correctly, but no devices were recognized.
 FAILED

manwu1994, 2023-06-08:
Hello, did you solved the mentioned issue? I met the same issue.

hinxx, 2023-09-15:
```
insmod: ERROR: could not insert module ../xdma/xdma.ko: Invalid module format
```

This might be the root of the problem. Might mean that the xdma.ko module was compiled for another kernel version. This woild happen if you compiled the module at some point, running kernel version a.b.c, then kernel got updated some time later. After rebooting into new kernel previously out of tree compiled modules would break.

Try to recompile the xdma module..

unmanedsystem, 2024-12-06:
Hello, did you solved the mentioned issue? I met the same issue.