Xilinx/dma_ip_drivers #235: QDMA AIO Read

DevDebugger, 2023-09-21:
using the qdma driver, when I try to read asynchronously, my system causes a freeze then restarts.

the restart is happening from a result of the line below:
rv = xcdev->fp_aiorw(xcdev->xcb->xpdev->dev_hndl, qhndl, caio->req_count, caio->reqv);

My overall goal is to read asynchronously, while also performing zero copy DMA. I turn off no_memcpy, however i believe this is only for performance testing and actually doesnt do anything besides that?

hmaarrfk, 2023-09-23:
i ran into this a while back too. i never had the patience to debug it. I just don't use the AIO path.

sofiachebotareva, 2023-09-26:
Hello @hmaarrfk 

My name is Sofia, I'm an Account Manager at Jungo Connectivity.

I noticed you've faced some challenges during your driver development project. I’d like to introduce you to WinDriver, our comprehensive driver development toolkit designed specifically for PCI/PCIe and USB devices. WinDriver is equipped with an array of debugging tools that could potentially assist you in resolving the issue at hand.

I would like to note that we provide special support for Xilinx cards, including QDMA samples.

We offer a 30-day free trial for our solution, allowing you to evaluate its functionalities and benefits without any obligations.

Would you be interested in receiving more detailed information and a link to access the free evaluation?