Xilinx/dma_ip_drivers #237: Increase QDMA Buffer Size

DevDebugger, 2023-09-27:
Is it possible to increase the buffer size (count) on the QDMA Driver?

The read function in cdev.c is reading buffers at a size of 4096. I would like to increase this to at least 8k.

I want to DMA around 10 GB/s, however i am only able to achieve around 6 GB/s when i write to file. Using the DMAPERF tool I can get the 10 GB/s, however that is not realistic since it is not copying the data. Any suggestions on how I can improve this QDMA driver performance?

hmaarrfk, 2023-09-27:
I think qdma should be using a userland buffer directly instead of instantiating a memory copy from kernel to userland.

When the user requests, create a ring there, then transfer the data to that buffer.

hmaarrfk, 2023-09-27:
you can increase qringsz to get to 8k i htink. yoj might even be able to get to 2^15-1 or 2^16-1.

DevDebugger, 2023-09-28:
Hi Mark, thank you for the suggestion. I believe this solution would work. I am only reading from the device and not writing. Do you suggest only modifying the read function in the cdev.c file? or where is the best place to create the ring and transfer the data to the buffer? maybe the descq_st_c2h_read() function in the qdma_st_c2h.c file?

hmaarrfk, 2023-09-29:
i'm not too sure. i never dug into it with success.

i do have a fork with a few fixes.

https://github.com/hmaarrfk/dma_ip_drivers/issues/2

excited to see a PR when you have some finidngs.