Xilinx/dma_ip_drivers #253: XDMA: driver build on kernel 6.6.11-1 (Debian 12, Bookworm)

MADMechaniculus, 2024-01-17:
The build of kernel module results in the following errors due to changing the function prototype `class_create`:

```bash
user@mirage:~/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma$ make
Makefile:17: XVC_FLAGS: .
make -C /lib/modules/6.6.11-amd64/build M=/home/user/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma modules
make[1]: Entering directory '/usr/src/linux-headers-6.6.11-amd64'
/home/user/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma/Makefile:17: XVC_FLAGS: .
  CC [M]  /home/user/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.o
  CC [M]  /home/user/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.o
In file included from /usr/src/linux-headers-6.6.11-common/include/linux/linkage.h:7,
                 from /usr/src/linux-headers-6.6.11-common/include/linux/kernel.h:17,
                 from /home/user/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.h:23,
                 from /home/user/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:22:
/home/user/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c: In function ‘xdma_cdev_init’:
/usr/src/linux-headers-6.6.11-common/include/linux/export.h:29:22: error: passing argument 1 of ‘class_create’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   29 | #define THIS_MODULE (&__this_module)
      |                     ~^~~~~~~~~~~~~~~
      |                      |
      |                      struct module *
/home/user/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:606:37: note: in expansion of macro ‘THIS_MODULE’
  606 |         g_xdma_class = class_create(THIS_MODULE, XDMA_NODE_NAME);
      |                                     ^~~~~~~~~~~
In file included from /usr/src/linux-headers-6.6.11-common/include/linux/device.h:31,
                 from /usr/src/linux-headers-6.6.11-common/include/linux/cdev.h:8,
                 from /home/user/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_mod.h:25,
                 from /home/user/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.h:27:
/usr/src/linux-headers-6.6.11-common/include/linux/device/class.h:230:54: note: expected ‘const char *’ but argument is of type ‘struct module *’
  230 | struct class * __must_check class_create(const char *name);
      |                                          ~~~~~~~~~~~~^~~~
/home/user/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:606:24: error: too many arguments to function ‘class_create’
  606 |         g_xdma_class = class_create(THIS_MODULE, XDMA_NODE_NAME);
      |                        ^~~~~~~~~~~~
/usr/src/linux-headers-6.6.11-common/include/linux/device/class.h:230:29: note: declared here
  230 | struct class * __must_check class_create(const char *name);
      |                             ^~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.6.11-common/scripts/Makefile.build:248: /home/user/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.11-common/Makefile:1938: /home/user/cxx/dma_ip_drivers/XDMA/linux-kernel/xdma] Error 2
make[1]: *** [/usr/src/linux-headers-6.6.11-common/Makefile:246: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.11-amd64'
make: *** [Makefile:39: all] Error 2
```

Kernel changes insides too quickly.

hinxx, 2024-01-17:
Yeah..

Change:

    class_create(THIS_MODULE, XDMA_NODE_NAME);

To:

    class_create(XDMA_NODE_NAME);

MADMechaniculus, 2024-01-17:
And here is complete guide for repaire the issue:
1. File cdev_sgdma.c, function bodies for `cdev_write_iter` and `cdev_read_iter`, need to change `io->iov` to `io->__iov`;
2. File xdma_cdev.c, line 606, change call of `class_create(THIS_MODULE, XDMA_NODE_NAME)` to `class_create(XDMA_NODE_NAME)`;
3. File cdev_ctrl.c, line 236. Need to add conversion of `vma->vm_flags` to `vm_area_struct *`. For example, you could write a function to write a flag into `vm_flags` field of vma (now is const):

```C
// Source cpied from another devforum 
void vm_flags_set(struct vm_area_struct *vma, vm_flags_t flags) {
    vma->vm_flags |= flags;
}
```

Usage:
```C
// ... bridge mmap function body
vm_flags_set(vma, VMEM_FLAGS)
// ... 
```

With this changes, driver completly builds, again :)

closed 2024-01-17 by MADMechaniculus